clash-lang/clash-compiler #1100: Function with blackbox does not compile to HDL when using point-free style

JvWesterveld, 2020-02-20:
This code compiles to VHDL fine:
```
module BlackboxTest where

import Clash.Prelude
import Clash.Annotations.Primitive (hasBlackBox)

{-# NOINLINE myId #-}
{-# ANN myId hasBlackBox #-}
myId = id

topEntity :: Bool -> Bool -> Bool
topEntity a b = myId $ seq a b
--topEntity = myId . seq
```

```
[{ "BlackBox" :
    { "name"      : "BlackboxTest.myId"
    , "workInfo"  : "Never"
    , "kind"      : "Declaration"
    , "type"      :
"myId
  :: x      -- ARG [0]
  -> x      -- RESULT"
    , "template" :
"~RESULT <= ~ARG[0];"
    }
  }
]
```
But, when using point-free style for `topEntity` (`topEntity = myId . seq`), Clash spits out this with git master from not so long ago:
```
*** Exception: Clash.Netlist.BlackBox(212): Forced to evaluate untranslatable type: (λ(v[4755801206503243778] :: GHC.Types.Bool[3674937295934324744]) ->
v[4755801206503243778][LocalId] :: GHC.Types.Bool[3674937295934324744]
-> GHC.Types.Bool[3674937295934324744])
CallStack (from HasCallStack):
  error, called at src/Clash/Netlist/BlackBox.hs:212:21 in clash-lib-1.1.0-inplace:Clash.Netlist.BlackBox
```
Is this a bug or is my understanding of how to write blackboxes wrong?

christiaanb, 2020-02-20:
You’re misunderstanding point-free style. You’d want to write: `(myId .) . seq` because `seq` is a binary function.

martijnbastiaan, 2020-02-20:
Sorry @JvWesterveld , should have spotted that. The problem is that Clash now tries to pass a function to you `myId` blackbox. In order to do that, it should be able to translate it to HDL (which is can't).

In effect, you're writing:

```haskell
topEntity :: Bool -> Bool -> Bool
topEntity a b = (myId (seq a)) b
```

JvWesterveld, 2020-02-20:
Whoops, of course, you guys are right!

closed 2020-02-20 by JvWesterveld