clash-lang/clash-compiler #1115: clash cannot synthesise large scale circuits

IchiroKawashima, 2020-02-24:
Hello.

I'm using clash for designing a large scale circuit. Then, I need to give the compiler a large type-level number. However, clash compiler does not finish the compilation. For example, I designed a summation circuit as follows:

```haskell
{-# LANGUAGE PartialTypeSignatures #-}

{-# OPTIONS_GHC -Wno-partial-type-signatures#-}

module Sum where

import           Clash.Prelude           hiding ( sum )
import           Data.Proxy
import           Data.Kind
import           Data.Singletons

data Sum (n :: Nat) (f :: TyFun Nat Type) :: Type
type instance Apply (Sum n) l = Signed (l + n)

sum :: forall k n . KnownNat k => KnownNat n => Vec (2 ^ k) (Signed n) -> Signed (k + n)
sum = dtfold (Proxy @(Sum n)) id (\SNat -> add)

{-# ANN sumTop
    (Synthesize
        { t_name = "sumTop"
        , t_inputs =
            [ PortName "clk"
            , PortName "rst"
            , PortName "ena"
            , PortName "idata"
            ]
        , t_output = PortName "odata"
        }) #-}
sumTop :: Clock System
       -> Reset System
       -> Enable System
       -> Signal System (Vec 4096 (Signed 32))
       -> Signal System (Signed _)
sumTop clk rst en xs = withClockResetEnable clk rst en $ register def $ sum <$> xs
{-# NOINLINE sumTop #-}
```

The compiler worked for the circuit whose input vector's length is `1024` but the compiler does not work in the case the length is more than `2048`. I did not see the compiler having generated HDL files even I waited a day.

Thank you.

christiaanb, 2020-02-25:
Apologies, this is somewhat of a known issue with the current version of the Clash compiler. Its core compilation strategy was not designed with unrolling recursive definitions in mind, and becomes quite slow when it is forced to do so (such as the case with `dtfold`).

There is a high-level plan specified in https://github.com/clash-lang/clash-compiler/issues/950 that hopefully speeds up the translation of these recursive definitions. It is currently being implemented at https://github.com/clash-lang/clash-compiler/tree/partial-evaluator-2 but is still at the very early stages.

christiaanb, 2020-02-25:
To elaborate further: not all recursive higher-order functions lead to this big slow-down. That's because in many cases, Clash is able to translate such functions to generate-for loops in VHDL/Verilog. However, because the types change in every recursive call of `dtfold`, it is not possible to translate this to a generate-for loop, and Clash must unroll all recursive calls.

IchiroKawashima, 2020-02-25:
Sure. I'll consider using `fold` instead of `dtfold` to workaround the problem until the problem is fixed.