clash-lang/clash-compiler #1166: clash-prelude 1.2 doesn't build out-of-the-box with GHC 8.6.5

gergoerdi, 2020-03-06:
I tried installing Clash 1.2 with the latest Stackage LTS version that doesn't use GHC 8.8 yet (because `stack` warns me against using GHC 8.8... *sigh*), which is [14.27](https://www.stackage.org/lts-14.27). It all goes well until we get to `clash-prelude`:

```
clash-prelude > [46 of 96] Compiling Clash.Sized.Fixed
clash-prelude > 
clash-prelude > /tmp/stack28763/clash-prelude-1.2.0/src/Clash/Sized/Fixed.hs:1:1: error:
clash-prelude >     solveSimpleWanteds: too many iterations (limit = 4)
clash-prelude >       Set limit with -fconstraint-solver-iterations=n; n=0 for no limit
clash-prelude >       Simples = {[WD] $dNum_adqD2 {0}:: Num a0 (CNonCanonical),
clash-prelude >                  [WD] $dKnownNat_adqD5 {0}:: KnownNat n0 (CNonCanonical),
clash-prelude >                  [WD] $dExtendingNum_adqDe {0}:: ExtendingNum a0 b0 (CNonCanonical),
clash-prelude >                  [WD] hole{co_adqDf} {0}:: rep (int + frac) ~ a0 (CNonCanonical),
clash-prelude >                  [WD] hole{co_adqDg} {0}:: rep (int + frac) ~ b0 (CNonCanonical),
clash-prelude >                  [WD] $dBits_adqDm {0}:: Bits a0 (CNonCanonical),
clash-prelude >                  [WD] hole{co_adqDn} {0}:: MResult a0 b0 ~ a1 (CNonCanonical),
clash-prelude >                  [WD] $dResize_adqDu {0}:: Resize f0 (CNonCanonical),
clash-prelude >                  [WD] $dKnownNat_adqDx {0}:: KnownNat a0 (CNonCanonical),
clash-prelude >                  [WD] $dKnownNat_adqDy {0}:: KnownNat b0 (CNonCanonical),
clash-prelude >                  [WD] hole{co_adqDz} {0}:: b0 ~ (int0 + frac0) (CNonCanonical),
clash-prelude >                  [WD] $dNum_adqDD {0}:: Num a0 (CNonCanonical),
clash-prelude >                  [WD] $dKnownNat_adqDG {0}:: KnownNat n0 (CNonCanonical),
clash-prelude >                  [WD] $dExtendingNum_adqDP {0}:: ExtendingNum a0 b0 (CNonCanonical),
clash-prelude >                  [WD] hole{co_adqDQ} {0}:: rep (int + frac) ~ a0 (CNonCanonical),
clash-prelude >                  [WD] hole{co_adqDR} {0}:: rep (int + frac) ~ b0 (CNonCanonical),
clash-prelude >                  [WD] $dBitPack_adqE8 {0}:: BitPack a0 (CNonCanonical),
clash-prelude >                  [WD] $d~_adqE9 {0}:: BitSize a0 ~ (m0 + n0) (CNonCanonical),
clash-prelude >                  [WD] $dKnownNat_adqEa {0}:: KnownNat n0 (CNonCanonical),
clash-prelude >                  [WD] hole{co_adqEb} {0}:: MResult a0 b0 ~ a1 (CNonCanonical),
clash-prelude >                  [WD] hole{co_adqEc} {0}:: n0
clash-prelude >                                            ~ ((int + frac) + frac) (CNonCanonical),
clash-prelude >                  [WD] hole{co_adqEd} {0}:: BitVector ((int + frac) + frac)
clash-prelude >                                            ~ b0 (CNonCanonical),
clash-prelude >                  [WD] $dBits_adqEg {0}:: Bits a0 (CNonCanonical),
clash-prelude >                  [WD] hole{co_adqEh} {0}:: rep (int + frac) ~ a0 (CNonCanonical),
clash-prelude >                  [WD] $dBits_adqEm {0}:: Bits a0 (CNonCanonical),
clash-prelude >                  [WD] $dBits_adqEo {0}:: Bits a0 (CNonCanonical),
clash-prelude >                  [WD] $dBitPack_adqEq {0}:: BitPack a0 (CNonCanonical),
clash-prelude >                  [WD] $dKnownNat_adqEt {0}:: KnownNat m0 (CNonCanonical),
clash-prelude >                  [WD] $dBitPack_adqEv {0}:: BitPack a0 (CNonCanonical),
clash-prelude >                  [WD] $dBitPack_adqEx {0}:: BitPack a0 (CNonCanonical),
clash-prelude >                  [WD] hole{co_adqEy} {0}:: BitSize
clash-prelude >                                              Clash.Sized.Internal.BitVector.Bit
clash-prelude >                                            ~ n0 (CNonCanonical),
clash-prelude >                  [WD] $dBitPack_adqEA {0}:: BitPack a0 (CNonCanonical),
clash-prelude >                  [WD] hole{co_adqEB} {0}:: BitSize (BitVector int)
clash-prelude >                                            ~ m0 (CNonCanonical),
clash-prelude >                  [WD] hole{co_adqEC} {0}:: BitVector (n0 + m0) ~ a0 (CNonCanonical),
clash-prelude >                  [WD] $dBitPack_adqEE {0}:: BitPack a0 (CNonCanonical),
clash-prelude >                  [WD] $dKnownNat_adqEH {0}:: KnownNat m0 (CNonCanonical),
clash-prelude >                  [WD] $dBitPack_adqEJ {0}:: BitPack a0 (CNonCanonical),
clash-prelude >                  [WD] $dBitPack_adqEL {0}:: BitPack a0 (CNonCanonical),
clash-prelude >                  [WD] hole{co_adqEM} {0}:: BitSize
clash-prelude >                                              Clash.Sized.Internal.BitVector.Bit
clash-prelude >                                            ~ n0 (CNonCanonical),
clash-prelude >                  [WD] $dBitPack_adqEO {0}:: BitPack a0 (CNonCanonical),
clash-prelude >                  [WD] hole{co_adqEP} {0}:: BitSize (BitVector int)
clash-prelude >                                            ~ m0 (CNonCanonical),
clash-prelude >                  [WD] hole{co_adqEQ} {0}:: BitVector (n0 + m0) ~ a0 (CNonCanonical),
clash-prelude >                  [WD] $dEq_adqEW {0}:: Eq a0 (CNonCanonical),
clash-prelude >                  [WD] $dNum_adqEY {0}:: Num a0 (CNonCanonical),
clash-prelude >                  [WD] $dBitPack_adqF0 {0}:: BitPack a0 (CNonCanonical),
clash-prelude >                  [WD] $dResize_adqF2 {0}:: Resize f0 (CNonCanonical),
clash-prelude >                  [WD] $dKnownNat_adqF5 {0}:: KnownNat a0 (CNonCanonical),
clash-prelude >                  [WD] $dKnownNat_adqF6 {0}:: KnownNat b0 (CNonCanonical),
clash-prelude >                  [WD] $dBits_adqF8 {0}:: Bits a0 (CNonCanonical),
clash-prelude >                  [WD] hole{co_adqF9} {0}:: b0 ~ BitSize a0 (CNonCanonical),
clash-prelude >                  [WD] $dBitPack_adqFc {0}:: BitPack a0 (CNonCanonical),
clash-prelude >                  [WD] $dEq_adqFf {0}:: Eq a0 (CNonCanonical),
clash-prelude >                  [WD] $dNum_adqFh {0}:: Num a0 (CNonCanonical),
clash-prelude >                  [WD] $dBounded_adqFj {0}:: Bounded a0 (CNonCanonical),
[... 117 lines cut ...]
clash-prelude >                  [WD] $dBounded_adqJj {0}:: Bounded a0 (CNonCanonical),
clash-prelude >                  [WD] $dEnum_adqJl {0}:: Enum a0 (CNonCanonical),
clash-prelude >                  [WD] $dBounded_adqJn {0}:: Bounded a0 (CNonCanonical),
clash-prelude >                  [WD] $dEq_adqJr {0}:: Eq a0 (CNonCanonical),
clash-prelude >                  [WD] $dNum_adqJt {0}:: Num a0 (CNonCanonical),
clash-prelude >                  [WD] $dBitPack_adqJv {0}:: BitPack a0 (CNonCanonical),
clash-prelude >                  [WD] $dResize_adqJx {0}:: Resize f0 (CNonCanonical),
clash-prelude >                  [WD] $dKnownNat_adqJA {0}:: KnownNat a0 (CNonCanonical),
clash-prelude >                  [WD] $dKnownNat_adqJB {0}:: KnownNat b0 (CNonCanonical),
clash-prelude >                  [WD] $dBits_adqJD {0}:: Bits a0 (CNonCanonical),
clash-prelude >                  [WD] hole{co_adqJE} {0}:: b0 ~ BitSize a0 (CNonCanonical),
clash-prelude >                  [WD] $dBounded_adqJG {0}:: Bounded a0 (CNonCanonical)}
clash-prelude >       WC = WC {wc_simple =
clash-prelude >                  [WD] $dKnownNat_adrlC {0}:: KnownNat fsk0 (CNonCanonical)
clash-prelude >                  [WD] $d~_adrlD {0}:: fsk0
clash-prelude >                                       GHC.TypeNats.<= (int + (fsk0 + frac)) (CNonCanonical)}
clash-prelude >   |
clash-prelude > 1 | {-|
clash-prelude >   | ^
clash-prelude > 
            
--  While building package clash-prelude-1.2.0 using:
      /home/cactus/sdk/stack/setup-exe-cache/x86_64-linux/Cabal-simple_mPHDZzAJ_2.4.0.1_ghc-8.6.5 --builddir=.stack-work/dist/x86_64-linux/Cabal-2.4.0.1 build --ghc-options " -fdiagnostics-color=always"
    Process exited with code: ExitFailure 1
```

gergoerdi, 2020-03-06:
It seems to work with Stackage LTS-15.2 and GHC 8.8.2.

christiaanb, 2020-03-06:
I don't understand how you got that far with 1.4.27, because it doesn't have the versions of the plugins required by clash-prelude-1.2

leonschoorl, 2020-03-06:
How did you get stack to compile clash-prelude-1.2.0 with stackage LTS-14.27?
If I try that I get:
```
In the dependencies for clash-prelude-1.2.0:
    ghc-typelits-extra-0.3.2 from stack configuration does not
                             match >=0.3.3 && <0.4  (latest matching version
                             is 0.3.3)
    ghc-typelits-knownnat-0.6 from stack configuration does not
                              match >=0.7.2 && <0.8  (latest matching version
                              is 0.7.2)
    ghc-typelits-natnormalise-0.6.2 from stack configuration does not
                                    match >=0.7.1 && <0.8  (latest matching
                                    version is 0.7.1)
needed since clash-prelude is a build target.
[...]
  * Recommended action: try adding the following to your extra-deps
    in /home/leon/qbaylogic/clash-1.2/stack.yaml:

ghc-typelits-extra-0.3.3@sha256:cc0ca56e47d6ba997414a39a5be0fbce4c1cdd251f37c2575c5778194e213b8e
ghc-typelits-knownnat-0.7.2@sha256:63054c8108f21a4bc5ace477227476b72a4e3792f35f37f2d406eef262ae4346
ghc-typelits-natnormalise-0.7.1@sha256:dfeb54a04020081604cfe26546d6c7e42ae70ce88df6f8deb3b9b79caaeb883b
```

And after added those extra-deps, it just worked for me.
(tested using stack-1.9.3)

gergoerdi, 2020-03-06:
I don't think I did anything special, I used a custom `stack` resolver that contained

```
resolver: lts-14.27
name: clash-1.2

packages:
  - clash-ghc-1.2.0
  - clash-prelude-1.2.0
  - clash-lib-1.2.0

  - concurrent-supply-0.1.7
```

leonschoorl, 2020-03-06:
Do you happen to have `allow-newer: true` in `$HOME/.stack/config.yaml`?

Because this causes stack to "to ignore all version constraints and build anyway".
Which contrary to what the name of the options suggest, this causes it to build with older versions of the `ghc-typelits-*` packages then required by the cabal files.
And this gets me the same build failures as you got.

gergoerdi, 2020-03-11:
As a principle, I don't use `allow-newer: true`.

Here is the full setup to reproduce this:

`clash-1.2.yaml` file:

```
resolver: lts-14.27
name: clash-1.2

packages:
  - clash-ghc-1.2.0
  - clash-prelude-1.2.0
  - clash-lib-1.2.0
```

`stack.yaml` file:

```
resolver: clash-1.2.yaml
```

`package.yaml` file:

```
name: retroclash
version: 0.1.0
category: acme

dependencies:
  - base
  - clash-prelude
  - clash-lib
  - clash-ghc

source-dirs:
  - src
  
executables:
  empty:
    main: Main.hs
```

leonschoorl, 2020-03-12:
I think the problem is with your `clash-1.2.yaml`. 
It takes `lts-14.27` and adds to that the three clash packages.
But you're not adding the proper `ghc-typelits-...` packages, and stack warns about this:
```
WARNING: Ignoring out of range dependency (trusting snapshot over Hackage revisions): ghc-typelits-extra-0.3.2. clash-prelude requires: >=0.3.3 && <0.4
WARNING: Ignoring out of range dependency (trusting snapshot over Hackage revisions): ghc-typelits-knownnat-0.6. clash-prelude requires: >=0.7.2 && <0.8
WARNING: Ignoring out of range dependency (trusting snapshot over Hackage revisions): ghc-typelits-natnormalise-0.6.2. clash-prelude requires: >=0.7.1 && <0.8
```

If you add them to `clash-1.2.yaml` the warnings go away and everything compiles as it should
```yaml
resolver: lts-14.27
name: clash-1.2

packages:
  - ghc-typelits-extra-0.3.3
  - ghc-typelits-knownnat-0.7.2
  - ghc-typelits-natnormalise-0.7.2
  - clash-ghc-1.2.0
  - clash-prelude-1.2.0
  - clash-lib-1.2.0
```

gergoerdi, 2020-04-17:
So I guess this was PEBKAC: Problem Exists Between Keyboard And Chair.

closed 2020-04-17 by gergoerdi