clash-lang/clash-compiler #1182: NamedTypes annotationon vector fields

gergoerdi, 2020-03-09:
With Clash 1.2, given the following input:

```
import Clash.Prelude
import Clash.Annotations.TH

data SevenSegment dom (n :: Nat) = SevenSegment
    { anodes :: "AN" ::: Signal dom (Vec n Bool) }

topEntity
    :: "CLK" ::: Clock System
    -> "SS" ::: SevenSegment System 4
topEntity clk = withClockResetEnable clk resetGen enableGen board
  where
    board = SevenSegment{ anodes = pure $ repeat False }
makeTopEntity 'topEntity
```

this gives the following warning:

```
src/Calculator.hs:13:1: warning:
    Make sure HDL port names are correct:
Backtracked when constructing Clash.Sized.Vector.Vec
(Type appears recursive)
   |
13 | makeTopEntity 'topEntity
   | ^^^^^^^^^^^^^^^^^^^^^^^^
```

The generated Verilog is wrong, unbundling the `Vec 4 Bool` into four `wire`s:

```
/* AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
** GENERATED BY CLASH 1.2.0. DO NOT MODIFY.
*/
`timescale 100fs/100fs
module topEntity
    ( // Inputs
      input  CLK // clock

      // Outputs
    , output wire  SS_AN
    , output wire  SS_1
    , output wire  SS_2
    , output wire  SS_3
    );
  wire [3:0] SS;

  assign SS = {4 {1'b0}};

  assign SS_AN = SS[3:3];

  assign SS_1 = SS[2:2];

  assign SS_2 = SS[1:1];

  assign SS_3 = SS[0:0];

endmodule
```

martijnbastiaan, 2020-03-09:
Hmm, although it's not really "inside recursive data types" I guess it still triggers a case that's unlikely to work (according to the docs: "Unlikely to work: annotations inside recursive data types"). @blaxill, could you comment? :)

gergoerdi, 2020-03-09:
But what is recursive here?

If I change the type of the field to e.g. `Signal dom Bool` then it works.
Also, https://github.com/gergoerdi/retroclash-lib/blob/master/src/RetroClash/VGA.hs#L30 works, and that includes another nested datatype in `VGASync`.

martijnbastiaan, 2020-03-09:
`Vec` is defined recursively, I think that's what it's complaining about.

```haskell
data Vec :: Nat -> Type -> Type where
  Nil  :: Vec 0 a
  Cons :: a -> Vec n a -> Vec (n + 1) a
```

gergoerdi, 2020-03-09:
But surely `Vec` should be special-cased for this, like it is for many other purposes in Clash already.

martijnbastiaan, 2020-03-09:
I agree

blaxill, 2020-03-09:
A workaround could be to remove the `"AN" :::` annotation, in general this will cause the ports to be more granular, but since `SevenSegment` has one member I think this will be what you want?

1. It's been a while since I worked on this but I think my intention was for this case to be handled since it appears you aren't looking for annotations _inside_ the vec.  

2. It would also be nice to have `Vec` working as a special case, but since this runs on template haskell its more complex than it seems, particularly when the Nat is not a constant applied directly.

I'll have a look later and see if I can at least do (1) this but I'm afraid I have limited time available for Clash at the moment. 

Relevant code is [here](https://github.com/clash-lang/clash-compiler/blob/4ff664975167b8632e85cc13410f28e3456530cd/clash-prelude/src/Clash/Annotations/TH.hs#L275) if someone else wants to have a go.

leonschoorl, 2020-03-09:
With https://github.com/clash-lang/clash-compiler/pull/1149 this changed:
It now error with:
```
<no location info>: error:
    Clash error call:
    Clash.Netlist.Util(643): Ports were annotated as product, but type wasn't one: 
    
       Filtered was: FilteredHWType (Vector 4 Bool) [[(False,FilteredHWType (Vector 4 Bool) [[(False,FilteredHWType Bool []),(False,FilteredHWType Bool []),(False,FilteredHWType Bool []),(False,FilteredHWType Bool [])]])]]
    
       Ports was: PortProduct "SS" [PortName "AN"]
```

gergoerdi, 2020-03-09:
> A workaround could be to remove the `"AN" :::` annotation, in general this will cause the ports to be more granular, but since `SevenSegment` has one member I think this will be what you want?

That is only because I always try to cut down the code in my bug reports. In my real code, I have three fields i `SevenSegment`:

```
data SevenSegment dom n = SevenSegment
    { anodes :: "AN" ::: Signal dom (Vec n Bool)
    , segments :: "SEG" ::: Signal dom (Vec 7 Bool)
    , dp :: "DP" ::: Signal dom Bool
    }
```

blaxill, 2020-03-09:
> That is only because I always try to cut down the code in my bug reports. In my real code, I have three fields i `SevenSegment`:

Fair enough, another workaround would be to use BitVector. 

Looking at the test case and leonschoorl's notes, this is actually happening because Clash is squashing hierarchy and `makeTopEntity` isn't. `makeTopEntity` should still be working even with this test case and the warning should be ignorable. I'm surprised this is getting messed by the [tests](https://github.com/clash-lang/clash-compiler/blob/4ff664975167b8632e85cc13410f28e3456530cd/clash-prelude/tests/Clash/Tests/TopEntityGeneration.hs). I'll look into it deeper soon.

blaxill, 2020-03-09:
@gergoerdi  #1182 produces the following verilog, let me know if they aren't what you expect:

~~~haskell
data SevenSegment dom (n :: Nat) = SevenSegment
    { anodes :: "AN" ::: Signal dom (Vec n Bool) }

topEntity
    :: "CLK" ::: Clock System
    -> "SS" ::: SevenSegment System 4
topEntity clk = withClockResetEnable clk resetGen enableGen $
    SevenSegment{ anodes = pure $ repeat False }
makeTopEntity 'topEntity
~~~

~~~verilog
/* AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
** GENERATED BY CLASH 1.3.0. DO NOT MODIFY.
*/
`timescale 100fs/100fs
module topEntity
    ( // Inputs
      input  CLK // clock

      // Outputs
    , output wire [3:0] SS_AN
    );

  assign SS_AN = {4 {1'b0}};

endmodule
~~~

And the larger example mentioned:

~~~haskell
data SevenSegment dom (n :: Nat) = SevenSegment
    { anodes :: "AN" ::: Signal dom (Vec n Bool)
    , segments :: "SEG" ::: Signal dom (Vec 7 Bool)
    , dp :: "DP" ::: Signal dom Bool
    }

topEntity
    :: "CLK" ::: Clock System
    -> "SS" ::: SevenSegment System 8
topEntity clk = withClockResetEnable clk resetGen enableGen $
    SevenSegment{ anodes = pure $ repeat False, segments = pure $ repeat False, dp = pure False }
makeTopEntity 'topEntity
~~~

~~~
/* AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
** GENERATED BY CLASH 1.3.0. DO NOT MODIFY.
*/
`timescale 100fs/100fs
module topEntity
    ( // Inputs
      input  CLK // clock

      // Outputs
    , output wire [7:0] SS_AN
    , output wire [6:0] SS_SEG
    , output wire  SS_DP
    );
  wire [15:0] SS;

  assign SS = {{8 {1'b0}},{7 {1'b0}},1'b0};

  assign SS_AN = SS[15:8];

  assign SS_SEG = SS[7:1];

  assign SS_DP = SS[0:0];

endmodule
~~~

gergoerdi, 2020-03-10:
Yes that looks correct.

martijnbastiaan, 2020-03-10:
Fixed in https://github.com/clash-lang/clash-compiler/pull/1186. Should be backported soon :).

closed 2020-03-10 by martijnbastiaan

gergoerdi, 2020-03-11:
Even with 1c5c7c1c3d3c9fc833b1e2f641d0b97f3c67d783, my original example still produces this warning:

```
src/Calculator.hs:12:1: warning:
    Make sure HDL port names are correct:
Backtracked when constructing Clash.Sized.Vector.Vec
(Type appears recursive)
   |
12 | makeTopEntity 'topEntity
   | ^^^^^^^^^^^^^^^^^^^^^^^^
```

Albeit, the generated Verilog seems correct:

```
/* AUTOMATICALLY GENERATED VERILOG-2001 SOURCE CODE.
** GENERATED BY CLASH 1.3.0. DO NOT MODIFY.
*/
`timescale 100fs/100fs
module topEntity
    ( // Inputs
      input  CLK // clock

      // Outputs
    , output wire [3:0] SS_AN
    );

  assign SS_AN = {4 {1'b0}};

endmodule

```

blaxill, 2020-03-11:
The warning is intended with my reasoning being that there might be annotations deeper in the type tree that are being ignored but we have to give up at some point as the traversal is somewhat naive; type applications are applied where possible to resolve directly, but in general this is complex to do in template haskell. There has to be some termination criteria which may result in deeper annotations will be ignored.

We could silence the warning for Vec because the programmer likely doesn't have or at least doesn't intend for deeper annotations to be applied, but its a somewhat fuzzy situation so I believe having a warning to remind the user to check the generated HDL is warranted.

Ideally I'd like to only produce a warning if the programmer has _definitely_ added annotations deeper in the type hierarchy that will be ignored, but a more complete type checker in template Haskell would be non-trivial.

blaxill, 2020-03-11:
Perhaps the warning could be more descriptive- "Make sure the generated HDL port names are correct as annotations deeper than <TYPE> will not be applied"

alex-mckenna, 2020-03-11:
Even better if we can pick out the names of the ports where children might not be correct, e.g.

```
Warning: Annotations deeper than

  <NAME> ::: <TYPE>

are not being applied because <REASON>.

Ensure the port names in the generated HDL are correct.
```

Although granted, I haven't looked at this - so I don't know how feasible / useful it would be