clash-lang/clash-compiler #1183: `maxBound :: Signed 0` misbehaves

martijnbastiaan, 2020-03-09:
```haskell
Clash.Prelude> [0..] :: [Signed 0]
*** Exception: Negative exponent
Clash.Prelude> maxBound :: Signed 0
*** Exception: Negative exponent
Clash.Prelude> minBound :: Signed 0
*** Exception: Negative exponent
```

martijnbastiaan, 2020-03-09:
Possible solution: treat `Signed 0`, `Unsigned 0`, `Index 0`, and `BitVector 0` as void types. I.e., having no inhabitants. Implementation: https://github.com/clash-lang/clash-compiler/commit/1f5dbeab50ae13702c68329300f66cf5fa4a44b2. This turned out to be a controversial issue. I'll create a PR once it's settled.

rowanG077, 2020-03-09:
Don't forget `Index 1`.

rowanG077, 2020-03-09:
Shouldn't `Index 1` be included here?

alex-mckenna, 2020-03-09:
`Index 1` has a single inhabitant, `0`

martijnbastiaan, 2020-03-09:
I didn't include `Index 1` because it has an implementation that's conceptually sound, meets its own spec, and doesn't crash when you use `minBound` or `maxBound`. IMO, `Index` should be its own thing unrelated to `Unsigned`.

> For example the following will be invalid:
> 
>     x = (0 :: Unsigned 0)

IMO, yes, but it would result in an `XException` not a compile error, similar to `0 :: Index 0`. But like I said, it's a contentious issue. The current definitions for ranges say:

* `Signed n`: `[-(2^(n-1)) .. 2^(n-1)-1]` => `[-(2^(-1)) .. 2^(-1)-1]`
* `Unsigned n`: `[0 .. 2^n-1]` => `[0 .. 0]`

following this, we'd need to implement `Signed` as void and `Unsigned` as unit. I.e.:

```
>>> minBound :: Signed 0
XException: #boos 
>>> minBound :: Unsigned 0
0
>>> minBound :: Index 1
0
>>> minBound :: Index 0
XException: #boos 
```

Arguments in favor of making the 4 (`BV`, `Signed`, `Unsigned`, `Index`) with n=0 _all_ `XException`:

* "Here's how it works" is very easy to explain.
* It keeps the door open for generalizations (as opposed to enforcing it on the type level)
* It seems to me that both `Unsigned` and `Signed` are directly defined in terms of the number of bits they're defined with. IMO therefore there isn't a "good" answer to the question: how many numbers does a zero-width construct carry? And to invoke the, pardon my French, Python Zen: "In the face of ambiguity, refuse the temptation to guess."

Arguments in favor of implementing specwise:

 * It follows the spec (obviously) 
 * Implementation naturally follows from range definitions

Arguments in favor of making the 4 with n=0 _all_ `0`:

 * It's the easiest to implement from the current state of affairs. 
 * It keeps the ability to do `case (x :: Signed n) of {0 -> ..., 1 -> ...}`, instantiating it with `n=0` and expect it to simulate _and_ synthesize. (The latter even having a special case in `caseCon`.)

Hope I didn't miss anything. Please edit this post (and add your username :)) if you want to add extra points.

christiaanb, 2020-03-09:
BlueSpec treats 0-Bit Signed (`Int(n)` in their lingo) as unit: https://github.com/B-Lang-org/bsc/blob/1bcd48b48305f1ee33dd17f1d09fe3da2f072b36/src/Libraries/Base1/Prelude.bs#L1481

martijnbastiaan, 2020-03-10:
`Signed 0` as unit: https://github.com/clash-lang/clash-compiler/commit/99ff6684a57409b46c7411fa192a7dfa18a65101. I'll make a PR tomorrow.

leonschoorl, 2020-03-10:
Just discovery some more surprising stuff related to 0-bit numbers:
In ghci:
```
Prelude> 2^0
1
```

In clashi
```
Clash.Prelude> (8::Unsigned 8) ^ d0
0
Clash.Prelude> :t (8::Unsigned 8) ^ d0
(8::Unsigned 8) ^ d0 :: Unsigned 0
```
Because the result is 0-bits wide it can't return 1.

martijnbastiaan, 2020-03-11:
@leonschoorl Hmm :(. I guess we should change

```haskell
type ExpResult (Index m) n = Index (m ^ n)
```

to

```haskell
type ExpResult (Index m) n = Min 1 (Index (m ^ n))
```

but I fear type inference issues.

alex-mckenna, 2020-03-11:
> I guess we should change
> ```haskell
>type ExpResult (Index m) n = Index (m ^ n)
> ```
>to
> ```haskell
>type ExpResult (Index m) n = Max 1 (Index (m ^ n))
> ```
>but I fear type inference/checking issues.

Does this even make sense? Isn't

```haskell
1 :: Nat
Index (m ^ n) :: Type
```

alex-mckenna, 2020-03-11:
Oh wait, did you mean

```haskell
Index (Max 1 (m ^ n))
```

martijnbastiaan, 2020-03-11:
Yeah sorry I ninja-edited it, but I guess the e-mail got sent out immediately :(

commit aaad3f0f (2020-03-11): Treat `Signed 0`, `Unsigned 0`, `Index 1`, `BitVector 0` as unit. Fix #1183

See discussion at https://github.com/clash-lang/clash-compiler/issues/1183
commit 5b103ea2 (2020-03-11): Treat `Signed 0`, `Unsigned 0`, `Index 1`, `BitVector 0` as unit. Fix #1183

See discussion at https://github.com/clash-lang/clash-compiler/issues/1183
commit 47f0bf45 (2020-03-11): Treat `Signed 0`, `Unsigned 0`, `Index 1`, `BitVector 0` as unit. Fixes #1183

See discussion at https://github.com/clash-lang/clash-compiler/issues/1183
commit e4006b01 (2020-03-11): Treat `Signed 0`, `Unsigned 0`, `Index 1`, `BitVector 0` as unit. Fixes #1183

See discussion at https://github.com/clash-lang/clash-compiler/issues/1183
closed 2020-03-11 by martijnbastiaan