clash-lang/clash-compiler #1187: "Forced to evaluate unexpected function argument"

gergoerdi, 2020-03-10:
The following Clash code compiles fine with Clash 1.0, but crashes Clash 1.2:

```
import Clash.Prelude
import RetroClash.Keypad (inputKeypad)
import RetroClash.Clock
import RetroClash.SerialRx
import RetroClash.SerialTx
import Control.Monad
import Data.Maybe
import Data.Char
import Control.Monad.State

type Digit = Index 10

{-# ANN topEntity
  (Synthesize
    { t_name   = "Calculator"
    , t_inputs =
          [ PortName "CLK"
          , PortName "RX"
          , PortName "ROWS"
          ]
    , t_output = PortProduct ""
          [ PortName "TX"
          , PortName "COLS"
          ]
    }) #-}
topEntity
    :: Clock System
    -> Signal System Bit
    -> Signal System (Vec 4 Bool)
    -> ( Signal System Bit
      , Signal System (Vec 4 Bool)
      )
topEntity clk = withClockResetEnable clk resetGen enableGen board
  where
    board rx rows =
        ( tx
        , reverse <$> cols
        )
      where
        digits = logic @4 cmd

        (tx, ack) = serialTx @8 (SNat @9600) (fmap bitCoerce <$> serialDisplay ack digits)
        cmd = (const Nothing =<<) <$> (serialRx @8 (SNat @9600) rx)

        (cols, _) = inputKeypad (repeat $ repeat 0)  rows

serialDisplay
    :: forall n dom. (KnownNat n, HiddenClockResetEnable dom)
    => Signal dom Bool
    -> Signal dom (Vec n (Maybe Digit))
    -> Signal dom (Maybe (Unsigned 8))
serialDisplay ack digits = pure Nothing

logic
    :: forall n dom. (KnownNat n, HiddenClockResetEnable dom)
    => Signal dom (Maybe ())
    -> Signal dom (Vec n (Maybe Digit))
logic = const $ pure $ repeat Nothing
```

Compilation command:

```
clash -fclash-inline-limit=30 src/Calculator.hs  -isrc --verilog
```

Error message from Clash 1.2:

```

<no location info>: error:
    Clash error call:
    Clash.Netlist.BlackBox(243): Forced to evaluate unexpected function argument: (letrec
      x[1032705] :: GHC.Types.Bool[3674937295934324744]
      = Clash.Sized.Internal.Index.eq#
          @(GHC.TypeNats.Div[3674937295934325082]
              5000000000
              (Clash.Signal.Internal.DomainConfigurationPeriod[8214565720323794442]
                 (Clash.Signal.Internal.KnownConf[8214565720323789608]
                    "System")))
          result_11[262276][LocalId]
          (Clash.Sized.Internal.Index.fromInteger#
             @(GHC.TypeNats.Div[3674937295934325082]
                 5000000000
                 (Clash.Signal.Internal.DomainConfigurationPeriod[8214565720323794442]
                    (Clash.Signal.Internal.KnownConf[8214565720323789608]
                       "System")))
             500000
             499999)
    in <prefixName>"x_"
    Clash.Sized.Internal.Index.eq#
      @(GHC.TypeNats.Div[3674937295934325082]
          5000000000
          (Clash.Signal.Internal.DomainConfigurationPeriod[8214565720323794442]
             (Clash.Signal.Internal.KnownConf[8214565720323789608]
                "System")))
      result_11[262276][LocalId]
      (Clash.Sized.Internal.Index.fromInteger#
         @(GHC.TypeNats.Div[3674937295934325082]
             5000000000
             (Clash.Signal.Internal.DomainConfigurationPeriod[8214565720323794442]
                (Clash.Signal.Internal.KnownConf[8214565720323789608]
                   "System")))
         500000
         499999) :: GHC.Types.Bool[3674937295934324744])
    CallStack (from HasCallStack):
      error, called at src/Clash/Netlist/BlackBox.hs:243:32 in clash-lib-1.2.0-ClaqIO5T7YOLlk84wqbN3Y:Clash.Netlist.BlackBox
```

gergoerdi, 2020-03-10:
Unfortunately I wasn't able to cut down the repro code any further, since very small code removals are enough to suppress the issue. I've uploaded the full code without any external dependencies to https://github.com/gergoerdi/clash-issue-1187

martijnbastiaan, 2020-03-10:
Hmm, Clash is doing something seriously wrong here. It ends up with

```
(letrec
      x[1036161] :: GHC.Types.Bool[3674937295934324744]
      = Clash.Sized.Internal.Index.eq#
          @(GHC.TypeNats.Div[3674937295934325082]
              5000000000
              (Clash.Signal.Internal.DomainConfigurationPeriod[8214565720323794333]
                 (Clash.Signal.Internal.KnownConf[8214565720323789501]
                    "System")))
          result_10[265348][LocalId]
          (Clash.Sized.Internal.Index.fromInteger#
             @(GHC.TypeNats.Div[3674937295934325082]
                 5000000000
                 (Clash.Signal.Internal.DomainConfigurationPeriod[8214565720323794333]
                    (Clash.Signal.Internal.KnownConf[8214565720323789501]
                       "System")))
             500000
             499999)
    in <prefixName>"x_"
    Clash.Sized.Internal.Index.eq#
      @(GHC.TypeNats.Div[3674937295934325082]
          5000000000
          (Clash.Signal.Internal.DomainConfigurationPeriod[8214565720323794333]
             (Clash.Signal.Internal.KnownConf[8214565720323789501]
                "System")))
      result_10[265348][LocalId]
      (Clash.Sized.Internal.Index.fromInteger#
         @(GHC.TypeNats.Div[3674937295934325082]
             5000000000
             (Clash.Signal.Internal.DomainConfigurationPeriod[8214565720323794333]
                (Clash.Signal.Internal.KnownConf[8214565720323789501]
                   "System")))
         500000
         499999) :: GHC.Types.Bool[3674937295934324744])
```

as an enable line. I don't see _any_ enable-line manipulation in your program though, so it should simply see `True`. 

Thanks for the bug report!

martijnbastiaan, 2020-03-10:
Woops, spoken too soon, I completely missed the `regEn`s. I'll keep digging!

martijnbastiaan, 2020-03-10:
This bug uncovered two issues:

 * We use `reduceConstant` in Clash's cleanup/flattening stage and we didn't account for the fact that it might introduce let-bindings. `mkArgument` doesn't expect let-bindings, so it trips. Running deadcode elimination right after `reduceConstant` will help in this particular case, as `reduceConstant` made a binding that ended up being unused. Your immediate issue has therefore been fixed in https://github.com/clash-lang/clash-compiler/pull/1189.

* `mkArgument` should simply handle let-bindings. I'll submit a PR soon.

commit d4ca5635 (2020-03-11): Run 'deadCode' after successful 'reduceConstant'. Fix #1187

'reduceConstant' uses the evaluator ('whnfRW') to do its job. 'whnfRW'
uses 'bindPureHeap' to bind variables, even if these variables end up
unused in the resulting expression. This in turn trips 'mkArgument' in
the netlist stage, as it doesn't expect let-bindings as blackbox
arguments. Running 'deadCode' removes these "useless" bindings, hence
preventing an error in 'mkArgument'.
commit 3ba338bd (2020-03-11): Run 'deadCode' after successful 'reduceConstant'. Fix #1187

'reduceConstant' uses the evaluator ('whnfRW') to do its job. 'whnfRW'
uses 'bindPureHeap' to bind variables, even if these variables end up
unused in the resulting expression. This in turn trips 'mkArgument' in
the netlist stage, as it doesn't expect let-bindings as blackbox
arguments. Running 'deadCode' removes these "useless" bindings, hence
preventing an error in 'mkArgument'.

(cherry picked from commit d4ca56353bd23b25054f932759bd84980082ec65)
closed 2020-03-11 by martijnbastiaan