clash-lang/clash-compiler #1191: Symlinks in subdirectories break Stack

gergoerdi, 2020-03-10:
OK OK OK I know this is a Stack issue. I hate myself for even bringing this up. But.

Stack chokes on installing Clash from Git due to a very stupid, very superficial problem: symlinks in subdirectories. Here's a Stack ticket, there are others on this issue as well: https://github.com/commercialhaskell/stack/issues/5004 Here's a comment from the maintainer: https://github.com/commercialhaskell/stack/issues/4913#issuecomment-522432180 The summary is that things are not looking too good for this.

But what *are* these problematic symlinks in the Clash source tree? I am glad you asked, because the list is not long:

* `/docs/general/LICENSE -> ../../LICENSE`

Yyyyep. Just getting rid of that single multi-level symlink would solve this issue (which I have verified in https://github.com/gergoerdi/clash-compiler/tree/stack-issue-5004-workaround). And I don't imagine its content changes so much that maintaining a separate copy in this directory would be that much of a problem...

martijnbastiaan, 2020-03-10:
Yeah, I'm fine with copying it. If you open a PR I'll merge it :).

martijnbastiaan, 2020-03-10:
Although I'm curious why it doesn't fail for `clash-prelude/CHANGELOG.md`?

gergoerdi, 2020-03-10:
I think it needs to be a symlink that points out by at least two levels... :man_shrugging:

closed 2020-03-10 by martijnbastiaan