clash-lang/clash-compiler #1198: Using `Clash.Signal.Trace.traceSignal` gives an exception when generating HDL

JvWesterveld, 2020-03-10:
For example,
```
topEntity :: Signal System Bit
topEntity = traceSignal1 "top" (pure 0)
```
yields
```
*** Exception: Clash.Netlist.BlackBox(264): No blackbox found for: GHC.Prim.atomicModifyMutVar#. Did you forget to include directories containing primitives? You can use '-i/my/prim/dir' to achieve this.

CallStack (from HasCallStack):
  extractPrimWarnOrFail, called at src/Clash/Netlist/Util.hs:749:36 in clash-lib-1.2.0-inplace:Clash.Netlist.Util
```

martijnbastiaan, 2020-03-10:
None of the `traceSignal` functions appear to have blackbox definitions :(.

JvWesterveld, 2020-03-10:
Right, so adding `clash-lib/prims/common/Clash_Signal_Trace.json` with something like this for `traceSignal1`, `traceVecSignal1`, `traceSignal` and `traceVecSignal` should be the fix, right?
```
[ { "BlackBox" :
    { "name"      : "Clash.Signal.Trace.traceSignal1"
    , "workInfo"  : "Never"
    , "kind"      : "Expression"
    , "type"      :
"traceSignal1
  :: ( BitPack a      -- ARG[0]
     , NFDataX a      -- ARG[1]
     , Typeable a )   -- ARG[2]
  => String           -- ARG[3]
  -> Signal dom a     -- ARG[4]
  -> Signal dom a"
    , "template" : "~ARG[4]"
    }
  }
]
```

martijnbastiaan, 2020-03-10:
Yes! Could you make a PR? :) I'm not 100% sure whether it should be `WorkNever` or `WorkVariable` (the latter being the default) even after reading the documentation of `WorkInfo`. @christiaanb ?

JvWesterveld, 2020-03-10:
Sure, I'll wait for Christiaan to provide some insight into what `workInfo` ought to be :)

leonschoorl, 2020-03-12:
It should be `Never`, because the blackbox never add work: `~ARG[4]` might do work, but this blackbox doesn't add any additional work.

`Variable` is used for example for `+`, because `x + y` does have to do work (the addition).
And it is called "variable" because is if both arguments were constant (`1 + 2`) then it doesn't add work to the resulting hardware.
(This assumes that the synthesis tool would optimize this to `3`)

martijnbastiaan, 2020-03-12:
@leonschoorl Ah right, that explanation helps!

commit 2034872d (2020-03-12): Add blackboxes for Clash.Signal.Trace traceSignal functions (fixes #1198)
commit 1170b0b5 (2020-03-13): Add blackboxes for Clash.Signal.Trace traceSignal functions (fixes #1198)

(cherry picked from commit 2034872dc69227e65aee5c01b731362b05d61bea)
closed 2020-03-12 by martijnbastiaan