clash-lang/clash-compiler #1200: `WithSingleDomain` &c. not exported by `Clash.Prelude`

gergoerdi, 2020-03-11:
I'm not sure if this is intentional or not.

I've tried porting some code to Clash HEAD, and I've run into this new single-domain check. It meant that I needed to add some `TryDomain` type instances to some of my types, and also add a `WithSingleDomain` constraint to `withResetEnableGen`, which I defined as

```
withResetEnableGen
    :: (KnownDomain dom, WithSingleDomain dom r)
    => (HiddenClockResetEnable dom => r)
    -> Clock dom -> r
withResetEnableGen board clk = withClockResetEnable clk resetGen enableGen board
```

While the error message I got without these new constraints / type instances is top-notch, there's a bit of extra friction in needing to import `Clash.Class.HasDomain`, since it is not re-exported by `Clash.Prelude`.

martijnbastiaan, 2020-03-11:
I think we should export them.

Having said that: multiple hidden constructs are currently an experimental feature. It's currently disabled on release branches (`-f-multiple-hidden` @ clash-prelude), but enabled on master. This is due to a number of known issues (poorer type inference, increased compiled times, API not as nice as it could be). They're _probably_ not going to be enabled on releases any time soon. So depending on your reasons for moving to HEAD (keeping your book up-to-date?) you might want to consider disabling it :).

gergoerdi, 2020-03-11:
Oh, I didn't realize there's a way to opt out of this feature. I will make sure to do that, since everything in my book will very explicitly be single-domain:

> # Multiple clocks
> 
> The problems associated with interfacing components with different clocks is beyond the scope of this book. We will design circuits where everything runs at the same clock, using counter-based clock rate division where needed. For circuits that generate video output, the
> base clock rate will be chosen to match the pixel clock rate. 
> 
> In the last chapters, we will implement computers based on pre-existing designs from the late 1970's. The original machines we will replicate had a multi-clock design by neccesity: the video
> subsystem was required to be simple enough so that it can run at the speed needed to produce a valid video signal, but the CPU was too complex to run at that rate. On a modern FPGA, these simple CPUs can be easily run at the speed prescribed by the video system. By using a
> single clock domain for the whole computer, we will avoid all the headache associated with domain crossing; but this comes at the price of our computer not behaving exactly the same as the original ones: we will discuss the user-visible implications of this separately for each machine.
>