clash-lang/clash-compiler #1208: Infer port names from `:::`

martijnbastiaan, 2020-03-11:
As mentioned in [this comment](https://github.com/clash-lang/clash-compiler/pull/1207#issuecomment-597595320), we'd probably want to fold `makeTopEntity` functionality into the compiler itself to work around template haskell stage restrictions.

martijnbastiaan, 2023-10-31:
Before we start implementing this, we should have a good view of what it is we actually want. We need to account for:

 * Annotating void types
 * `Vec`/`RTree` (not arbitrary recursive types)
 * Ability to "stop" looking for names recursively
 * Naming (sum-of-)product types
 * What happens if we do `"a" ::: "b" ::: Bool`?

prabhupraveen, 2025-02-18:
This will be a very useful enhancement to Clash, especially for large projects.

Instead defining explicit port and module names, Is it possible to use the function names and variable names themselves and suffix it with random string/counter to make it unique?

DigitalBrains1, 2025-02-18:
Also we should consider including the functionality `ANN Synthesize` has for splitting `Maybe` types. Perhaps we need a specific constructor for that. Such that the following code

```haskell
foo :: Maybe A -> Maybe B
{-# OPAQUE foo #-}
{-# ANN
  foo
  ( Synthesize
      { t_name = "foo"
      , t_inputs =
          [ PortProduct
              "inp"
              [ PortName "valid"
              , PortName "data"
              ]
          ]
      , t_output =
            PortProduct
              "out"
              [ PortName "valid"
              , PortName "data"
              ]
      }
  )
  #-}
```

could be achieved by something like:
```haskell
foo ::
  MaybePort "inp_valid" "inp_data" ::: Maybe A ->
  MaybePort "out_valid" "out_data" ::: Maybe B
```

DigitalBrains1, 2025-02-18:
> to work around template haskell stage restrictions.

I don't know if all issues are caused by stage restrictions, but there's a small list of issues with `makeTopEntity`. It is great when it works, but it has too many issues to recommend it in general.