clash-lang/clash-compiler #1215: Include clock origin and phase in domain.

rowanG077, 2020-03-12:
In some cases a design requires clocks to be related somehow. For instance a low overhead clock crossing can be implemented if two clocks have the same origin, phase and the period of one clock is an integer multiple of the other. Clash currently doesn't provide a way to encode this in a domain meaning there is no way to specify two clocks are related in this way.

I would like to have two extra parameters in `DomainConfiguration`:

- `vOrigin :: String`: What the origin of the clock is. If not specified this would default to a unique string.
- `vPhase :: Natural`: Where Nat is between 0 and 359. If not specified defaults to 0.

Would you guys be oke with this? Am I missing some other parameter that could be useful?

leonschoorl, 2020-03-16:
Interesting idea
Do you have an idea of what this would look like in use?
For example what would be the type of this low overhead synchronizer?

rowanG077, 2020-03-18:
I haven't tested it in Hardware since I don't have an FPGA around at home. However I believe something like this should work. It was inspired by a [comment](https://groups.google.com/d/msg/clash-language/aZcM-45Li3o/R46kj1kYAAAJ) @christiaanb made in the google group.

```
{-# LANGUAGE UndecidableInstances #-}

module SimpleSynchronizer where

import Clash.Explicit.Prelude

import GHC.TypeLits          ( TypeError
                             , ErrorMessage(..)
                             )

import Data.Kind             ( Type )

type GearRatio (fastDom :: Domain) (slowDom :: Domain) = GearRatioCalc
                                                           (Div (DomainPeriod slowDom) (DomainPeriod fastDom))
                                                           ((DomainPeriod fastDom) <=? (DomainPeriod slowDom))
                                                           ((Mod (DomainPeriod slowDom) (DomainPeriod fastDom)) <=? 0)

type family GearRatioCalc (ratio :: Nat) (fastDomIsFaster :: Bool) (isMultiple :: Bool) :: Nat where
  GearRatioCalc r 'True 'True  = r
  GearRatioCalc _ 'False _ = TypeError ('Text "Invalid gearing:" ':$$:
                                        'Text "\tThe fast Clock domain does not have a " ':<>:
                                        'Text "smaller period then the slow Clock Domain.")
  GearRatioCalc _ 'True 'False = TypeError ('Text "Invalid gearing:" ':$$:
                                            'Text "\tThe fast Clock domain does not cleanly " ':<>:
                                            'Text "divide the slow Clock domain.")

fastToSlowGear
  :: forall (a             :: Type)
            (fastDom       :: Domain)
            (slowDom       :: Domain)
            (gearRatio     :: Nat)
   . ( NFDataX a
     , Default a
     , KnownNat gearRatio
     , KnownDomain fastDom
     , KnownDomain slowDom
     , (GearRatio fastDom slowDom) ~ gearRatio
     , 1 <= gearRatio
     )
   => Clock fastDom
   -> Clock slowDom
   -> Reset slowDom
   -> Signal fastDom a
   -> Signal slowDom (Vec gearRatio a)
fastToSlowGear fClk sClk sRst fInData = sOutData
  where
    fRst = convertReset sClk fClk sRst
    fOutData = moore fClk fRst enableGen writer snd def fInData
    sOutData = unsafeSynchronizer fClk sClk fOutData
    writer :: (Index gearRatio, Vec gearRatio a) -> a -> (Index gearRatio, Vec gearRatio a)
    writer (cnt, accu) x = (satSucc SatWrap cnt, replace cnt x accu)
```

A similar implementation can be made for a `slowToFastGear`. This won't work in hardware if the there is any phase difference between the fast clock and slow clock (Either introduced explicitly by configuring a PLL or because the clocks have different sources). Clash doesn't allow me to specify the clocks needs to have these properties. This is just one simple use case though. A SDRAM controller needs out of phase clocks and in clash you could model this at the type level with this inclusion.

Basically what I would to to be able to write is this:

```
{-# LANGUAGE UndecidableInstances #-}

module SimpleSynchronizer where

import Clash.Explicit.Prelude

import GHC.TypeLits          ( TypeError
                             , ErrorMessage(..)
                             )
[... 69 lines cut ...]
  -- ^ Whether the initial (or "power up") value of memory elements is
  -- unknown/undefined, or configurable to a specific value
  , _resetPolarity :: ResetPolarity
  -- ^ Whether resets are active high or active low
  , _Phase :: Nat
  -- ^ The phase of the clock domain from 0 to 359
  , _Origin :: Symbol
  -- ^ What the origin of the clock domain is
  }
  deriving (Typeable)

type family DomainConfigurationPhase (config :: DomainConfiguration) :: Nat where
  DomainConfigurationPhase ('DomainConfiguration name period edge reset init polarity phase origin) = phase

type family DomainConfigurationOrigin (config :: DomainConfiguration) :: Symbol where
  DomainConfigurationOrigin ('DomainConfiguration name period edge reset init polarity phase origin) = origin

type DomainPhase (dom :: Domain) =
  DomainConfigurationPhase (KnownConf dom)

type DomainEqOrigin (domA :: Domain) (domB :: Domain) =
  (DomainConfigurationOrigin (KnownConf domA)) == (DomainConfigurationOrigin (KnownConf domB))
```

Of course the singleton and record version would have these changes as well. I'm not sure whether to directly expose the origin or just expose `DomainEqOrigin`.

rowanG077, 2020-03-18:
Oh and btw shouldn't `convertReset` use a synchronizer in the case when going from a synchronous reset to a synchronous reset? I say this because the clocks could have any arbitrary skew between them which means a synchronous reset in domain A is not necessarily a synchronous reset in domain B.

https://github.com/clash-lang/clash-compiler/blob/aaf7af3074b10198013e9eab72f5628c62816b20/clash-prelude/src/Clash/Explicit/Signal.hs#L895-L915

christiaanb, 2020-03-19:
You're right... that's a bug. Given they have a `knownDomain` constraint, we can check at "run-time" whether to insert a dual-ff synchronizer.

rowanG077, 2020-06-26:
With the work being done on [https://github.com/clash-lang/clash-compiler/issues/1387](generating an SDC file) I was reminded of this. It would be cool if we can include this additional information in the SDC file as well. That is if you guys are willing to have this additional information in a Clock domain.

christiaanb, 2020-06-28:
So all the information currently in the domain affect the simulation behavior, where the `vPeriod` is used in `unsafeSynchronizer`:
https://github.com/clash-lang/clash-compiler/blob/dbcc07851e5a99e8feb7485fdb15f9a76dcd3500/clash-prelude/src/Clash/Explicit/Signal.hs#L501-L554
And all the other information is used by `register#` (and `delay#`): https://github.com/clash-lang/clash-compiler/blob/dbcc07851e5a99e8feb7485fdb15f9a76dcd3500/clash-prelude/src/Clash/Signal/Internal.hs#L1063-L1115
Additionally, the `vPeriod` is used in things like the `clockGen` so that Clash and HDL simulation behave the same: https://github.com/clash-lang/clash-compiler/blob/dbcc07851e5a99e8feb7485fdb15f9a76dcd3500/clash-lib/prims/vhdl/Clash_Signal_Internal.json#L106-L123
I'd be willing to include the phase as well, as long as we can see its effects in `unsafeSynchronizer`, matching the simulation behavior of the generated HDL.

Finally, before adding something like the `fastToSlowGear`, and thus something like `vOrigin` to `clash-prelude`, we first need something in Clash that emit min/max-delay constraints between the registers in the two domains; otherwise the `unsafeSynchronizer` is not a safe CDC even if the registers were to be clocked by clocks that have the same origin and are in-phase.

I guess what I'm trying to say is that I'd first like to see this work in some external package (which is definitely possible, since Clash has extensible primops), before merging it in `clash-prelude`.

martijnbastiaan, 2022-08-10:
Somewhat related to this PR (but not completely): https://github.com/clash-lang/clash-compiler/pull/2295. For starter I think adding `vPhase` would be a good addition, which should cause similar changes to the PR I just linked.