clash-lang/clash-compiler #1240: `iterate` cannot be used to describe a reset value

christiaanb, 2020-03-20:
The following fails to compile:

```haskell
module Test where

import Clash.Prelude

topEntity = register @System (iterate d1 not True)
```

giving the following error:

```
Test.hs:5:1: error:

    Clash.Netlist.BlackBox(188): Couldn't instantiate blackbox for Clash.Signal.Internal.register#. Verification procedure reported:

    Argument 5 should be literal, as blackbox used ~CONST[5], but was:

    Identifier "xs" Nothing

    The source location of the error is not exact, only indicative, as it is acquired
    after optimizations. The actual location of the error can be in a function that is
    inlined. To prevent inlining of those functions, annotate them with a NOINLINE pragma.
  |
5 | topEntity = register @System (iterate d1 not True)
  | ^^^^^^^^^
```

looking at the debug output we see:

```
  xs[8286623314361788996] :: Clash.Sized.Vector.Vec[8214565720323789569]
                               1
                               GHC.Types.Bool[3674937295934324744]
  = Clash.Sized.Vector.init @1
      @GHC.Types.Bool[3674937295934324744]
      (Clash.Sized.Vector.Cons[8214565720323789572]
         @(GHC.TypeNats.+[3674937295934325068] 1 1)
         @GHC.Types.Bool[3674937295934324744]
         @1
         (_CO_
            @(GHC.Prim.~#[3674937295934324842]
                GHC.Types.Nat[3674937295934325064]
                GHC.Types.Nat[3674937295934325064]
                (GHC.TypeNats.+[3674937295934325068] 1 1)
                (GHC.TypeNats.+[3674937295934325068] 1 1)))
         GHC.Types.True[3891110078048108589]
         (Clash.Sized.Vector.Cons[8214565720323789572] @1
            @GHC.Types.Bool[3674937295934324744]
            @0
            _CO_
            (GHC.Classes.not
               (Clash.Sized.Vector.head @0
                  @GHC.Types.Bool[3674937295934324744]
                  xs[8286623314361788996][LocalId]))
            (Clash.Sized.Vector.Nil[8214565720323789573] @0
               @GHC.Types.Bool[3674937295934324744]
               _CO_)))
```

i.e.

```haskell
let xs = init (Cons True (Cons (not (head xs)) Nil))
```

which is semantically a constant, because the second element of the list is never used, but not syntactically; and so Clash doesn't want to inline it.

christiaanb, 2020-03-20:
A work-around is to use TemplateHaskell:

```haskell
module Test where

import Clash.Prelude

topEntity = register @System $(lift (iterate d1 not True))
```

martijnbastiaan, 2020-08-24:
[PR #1240 is reverted in 1.2](https://github.com/clash-lang/clash-compiler/pull/1455), but should also be reverted in master, reopening this issue.

commit c08bb41c (2020-05-28): Add test for #1240

PR #1354 fixes #1240
commit 2b28a493 (2020-05-28): Add test for #1240

PR #1354 fixes #1240
commit cd7fca4b (2020-05-28): Add test for #1240

PR #1354 fixes #1240
commit 04b01d5e (2020-05-28): Add test for #1240

PR #1354 fixes #1240