clash-lang/clash-compiler #1245: Support GHC 8.10

alex-mckenna, 2020-03-25:
As GHC 8.10 is now released, we should add support to Clash. This should just be a straightforward case of adding a new `UI` Module, although the changes to module hierarchy in the 8.10 API may make this more challenging.

TODO

  - [ ] Support compiling clash with GHC 8.10
  - [ ] Add GHC 8.10 to the CI image and use for testing
  - [ ] Backport to 1.2 branch (?)

martijnbastiaan, 2020-03-25:
> although the changes to module hierarchy in the 8.10 API may make this more challenging.

We should use [ghc-api-compat](https://hackage.haskell.org/package/ghc-api-compat) until the hierarchy is stable and we've dropped all GHCs without the stable hierarchy IMO.

Should we drop GHC 8.4 after doing this?

alex-mckenna, 2020-03-25:
Will dropping support for 8.4 affect users? To be safe, I think it might be better to announce end of life for GHC 8.4 support (and then drop it in say, Clash 1.4.)

martijnbastiaan, 2020-03-25:
Oh yeah, we definitely wouldn't drop it for 1.2. Just master / Clash 1.4.

alex-mckenna, 2020-04-07:
Assigning @leonschoorl, as he's been working on this issue the past few days.

closed 2020-04-24 by leonschoorl