clash-lang/clash-compiler #1252: Compiler Pipeline Testing

alex-mckenna, 2020-04-02:
## Motivation

Currently, `clash-testsuite` allows tests to be defined which run clash from source to either core (recently added in #1251), netlist or HDL. However, there are some shortcomings to the current way the testsuite is used:

  * using clash to generate HDL in the testsuite requires spawning a new process
  * using clash to generate core / netlist requires using an ad-hoc and fragile API

It would be desirable to do away with both of these issues and provide an interface which allows clash to be run up to/from an arbitrary stage for the purpose of testing. This has the potential to

  * **better isolate the definition of tests**
    If it is possible to define tests which only use a certain section of the compiler, there is less risk of unrelated issues in the project to cause tests to fail. For example, tests could be defined on identifier generation for a given backend from a netlist definition.

  * **allow more meaningful assertions to be made**
    If it is possible to obtain any intermediate representation the compiler may use, then tests can be defined which perform assertions on the most relevant IR. For example, tests on a normalization pass can be defined with assertions on the core. This is more precise than trying to identify how the desired behaviour will propagate through later stages of the compiler.

  * **increase testsuite performance**
    By allowing tests to run using only a slice of the compiler pipeline, work which is not necessary to the test can be avoided. Instead of having to start from a known source file, tests can start from known core, or known netlist.

  * **simplify testsuite**
    Having the ability to run clash in this way would mean that the `clash-testsuite` library would no longer be needed (as it is only needed to provide `runToNetlistStage` and `runToCoreStage` for netlist and core tests). A more friendly API would also mean that running clash in the testsuite could be performed without needing to spawn a new process.

## Problems / Questions (Update on discussion)

  * Starting / stopping the compiler at different phases requires either a way to serialize the different intermediate representations used in clash, or a way to effectively generate elements of a given IR. Neither of these are currently done in the compiler.

  * This would likely require a large amount of engineering effort. If a reasonable solution is found it should be possible to break it down into a collection of smaller changes in order to be reviewable.

christiaanb, 2020-04-02:
We have serialization working: we needed it for profiling and benchmarking.

WRT `clash-testsuite`, we can’t throw it away completely as we still need it to call the HDL simulators with the correct arguments.

alex-mckenna, 2020-04-02:
Ah yes, I was working on the assumption that `Test.Tasty.Clash` and `Test.Tasty.Clash.Program` would go back to being `other-modules` of the executable, like they were before netlist tests were first being added (see 1718946). Moving them back to `other-modules` would also remove the need for some things that were only added when netlist tests were first introduced, like having `tests: True` in the `cabal.project.local`

alex-mckenna, 2020-04-02:
> We have serialization working: we needed it for profiling and benchmarking.

Is this for all IR in the compiler?

christiaanb, 2020-04-02:
> 
> 
> > We have serialization working: we needed it for profiling and benchmarking.
> 
> Is this for all IR in the compiler?

No, but the IR used as source for normalisaton and netlist generation (which are the same IR) can be seriaized. The only thing we cannot serialize right now is the Netlist format used as the source of the backends.

alex-mckenna, 2020-04-03:
Ah, I wasn't aware we had a `Binary` instance for `Term`. I'll update the issue