clash-lang/clash-compiler #1256: Space leak with memory simulators

gergoerdi, 2020-04-11:
I've asked around on the mailing list a couple weeks ago about tracking down a space leak in simulation, and I think now I have managed to show that there is some inherent problem with the simulation of memory primitives.

To set the scene, let's suppose I have the following Clash circuit:

```
module Circuit where

import Clash.Prelude
import Data.Word
import Control.Monad.State

type PC = Unsigned 12

logicBoard
    :: (HiddenClockResetEnable dom)
    => Signal dom ()
    -> Signal dom ()
logicBoard x = dummy <$> cpuOut
  where
    cpuOut = cpu cpuIn
    romRead = withStart 0 $ blockRam1 NoClearOnReset (SNat @10000) 0 (progAddr <$> cpuOut) (pure Nothing)
    cpuIn = CPUIn <$> romRead

withStart x = mux (register True $ pure False) (pure x)

data CPUIn = CPUIn
    { instr :: Word8
    }
    deriving (Generic, NFDataX)

data CPUOut = CPUOut
    { progAddr :: PC
    , dummy :: ()
    }
    deriving (Generic, NFDataX)

defaultOutput :: PC -> CPUOut
defaultOutput pc = CPUOut
    { progAddr = pc
    , dummy = ()
    }

cpu :: (HiddenClockResetEnable dom) => Signal dom CPUIn -> Signal dom CPUOut
cpu = mealy step 0
  where
    step s CPUIn{} = let (y, s') = (defaultOutput s, s + 1) in (s', y)
```

I then use the following function to simulate interactively:

```
module Sim where

import Clash.Prelude
import Control.Concurrent

driveIO_ :: ([i] -> [o]) -> i -> IO ((o -> IO i) -> IO ())
driveIO_ circuit input0 = do
    inChan <- newChan
    writeChan inChan input0

    ins <- getChanContents inChan
    outs <- newMVar $ circuit ins

    return $ \world -> do
        out <- modifyMVar outs $ \(out:outs) -> return (outs, out)
        input <- world out
        writeChan inChan input
```

Now, I will simulate my circuit at two levels: at `cpu` and at `logicBoard`. Note that in `cpu`, we are lazy in the fields of `CPUOut`, and `cpu` is lazy in the fields of `CPUIn`, so there is no "hidden" forcing of anything anywhere. 

```
module Main where

import Clash.Prelude
import Circuit
import Sim
import Control.Monad

main :: IO ()
main = do
    sim <- driveIO_ (simulate @System cpu) (CPUIn 0)
    replicateM_ 100000 $ sim $ \CPUOut{} -> return (CPUIn 0)
```

```
module Main where

import Clash.Prelude
import Circuit
import Sim
import Control.Monad

main :: IO ()
main = do
    sim <- driveIO_ (simulate @System logicBoard) ()
    replicateM_ 100000 $ sim $ \() -> return ()
```

I think that depending on how `simulate` is implemented, either both of these functions should leak, or neither should. However, testing with `+RTS -M2M`, i.e. with a heap limit of 2 Mb, I find that 

* `sim-cpu` works as expected, without leaking
* `sim-logicboard` quickly runs out of heap space
* `sim-logicboard`, with the type of the `instr` field of `CPUIn` changed to be strict, doesn't leak.

Note that if `CPUIn` has multiple fields, some connected to RAM or ROM, others connected to the outside world, then only making the RAM/ROM-connected fields strict is enough to fix the space leak.

All this is suggesting to me that the problem is with the implementation of the memory primitives.

gergoerdi, 2020-04-11:
I've added full code to https://github.com/gergoerdi/clash-issue-1256

christiaanb, 2020-04-11:
I think your cpu is leaking space, it never consumes its input to produce an output. i.e. I don't know how to change the implementation of `blockRam#` to deal with a circuit that is not forcing the result produced by `blockRam#`. What does the `-hy` heap profile say? are we leaking `Word8` or `Unsinged 12`?

christiaanb, 2020-04-11:
That is, looking at the implementation of blockRam#:

```haskell
  go !ram o ret@(~(re :- res)) rt@(~(r :- rs)) et@(~(e :- en)) wt@(~(w :- wr)) dt@(~(d :- din)) =
    let ram' = d `defaultSeqX` upd ram e (fromEnum w) d
        o'   = if re then ram `Seq.index` r else o
    in  o `seqX` o :- (ret `seq` rt `seq` et `seq` wt `seq` dt `seq` go ram' o' res rs en wr din)
```

I don't know how to make it any stricter:
* To get the tail of the stream, both the `ram` and the latched output `o` are forced to WHNF. 
* Forcing `ram` to WHNF forces data-in `d` and write-enable `e` from "the previous cycle" to WHNF, and if the write-enable was asserted, it would also force the write-address from "the previous cycle" to WHNF.
* Forcing `o` to WHNF forces the read-enable `re` to WHNF, and the read address `r` to WHNF whenever read-enable is asserted.
* Forcing the tail of the tail of the stream then also forces the stream constructor from "the previous cycle".

gergoerdi, 2020-04-11:
I can't really comment on this issue because I am not familiar with the kind of compatibility guarantees you hope to give.

If it were up to me, I'd just change the type of `simulate*` in 1.4, and tell to the users "we have a type system so you will know exactly how to change your code"...

gergoerdi, 2020-04-11:
If I let it run to 1G heap with `-M1G`, `-hy` shows me a mysterious `*` as the type. `-hr` gives me `register#.goAsync` (and `blockRam#.go` at a much lower level).

christiaanb, 2020-04-11:
When I change `logicBoard` to:

```
logicBoard x = seq <$> x <*> (dummy <$> cpuOut)
```

and `step` from `CPU` to:

```
step s (CPUIn x) = x `seq` let (y, s') = (defaultOutput s, s + 1) in (s', y)
```

the heap profile is flat. i.e. when there's demand for the input.

christiaanb, 2020-04-12:
even putting the demand deeper into the `step` function still results in a flat heap profile:
```
step s (CPUIn x) = let (y, s') = (defaultOutput s, x `seq` s + 1) in (s', y)
```

christiaanb, 2020-04-12:
So in a way, `mux`, and ultimately `fmap` for `Signal` is "too lazy"; i.e. they leak space when there's no demand. That is, `mux` is defined as:

```haskell
mux bs ts fs = ite `fmap` bs <*> ts <*> fs
  where  ite b t f = if b then t else f
```

when we change that to:

```haskell
import Clash.Signal.Internal

fmapS :: (a -> b) -> Signal dom a -> Signal dom b
fmapS f = go
  where go (a :- as) = f a :- (a `seqX` go as)

mux bs ts fs = ite `fmapS` bs <*> ts <*> fs
  where  ite b t f = if b then t else f
```

you get the flat heap profile.

Now, you can change the behavior of `fmap` to be that of `fmapS` by enabling the `strict-mapSignal` cabal flag for `clash-prelude`.

Finally... I'm not a big fan of the `strict-mapSignal` flag, it can give a 20% or higher performance penalty in case when there's sufficient demand. I have seen this leaky behavior of `mux` before though. So I will introduce the `fmapS` primitive, and use it inside `mux`.

christiaanb, 2020-04-12:
@gergoerdi Would the above change to `mux` be sufficient for your use case? is that something you could check in your larger code base?

gergoerdi, 2020-04-12:
I will try the change on the full codebase and get back to you.

gergoerdi, 2020-04-13:
Do I get this right? I should replace the definition of `mux` in `clash-prelude` with [this one](https://github.com/clash-lang/clash-compiler/issues/1256#issuecomment-612619353), recompile with that `clash-prelude` but without any bang-annotations on `CPUIn`, and I should see no more space leaks? Because I just tried that, and nope, still leaks.

christiaanb, 2020-04-13:
Was there any demand on the input of `logicBoard`? This is the Circuit.hs that is giving me the flat heap profile:

```haskell
module Circuit where

import Clash.Prelude hiding (mux)
import Data.Word
import Control.Monad.State
import Clash.Signal.Internal hiding (mux)

fmapS :: (a -> b) -> Signal dom a -> Signal dom b
fmapS f = go
  where go (a :- as) = f a :- (a `seqX` go as)

mux bs ts fs = ite `fmapS` bs <*> ts <*> fs
  where
    ite b t f = if b then t else f

type PC = Unsigned 12

logicBoard
    :: (HiddenClockResetEnable dom)
    => Signal dom ()
    -> Signal dom ()
logicBoard x = seqX <$> x <*> (dummy <$> cpuOut)
  where
    cpuOut = cpu cpuIn
    romRead = withStart 0 $ blockRam1 NoClearOnReset (SNat @10000) 0 (progAddr <$> cpuOut) (pure Nothing)
    cpuIn = CPUIn <$> romRead

withStart x = mux (register True $ pure False) (pure x)

data CPUIn = CPUIn
    { instr :: Word8
    }
    deriving (Generic, NFDataX)

data CPUOut = CPUOut
    { progAddr :: PC
    , dummy :: ()
    }
    deriving (Generic, NFDataX)

defaultOutput :: PC -> CPUOut
defaultOutput pc = CPUOut
    { progAddr = pc
    , dummy = ()
    }

cpu :: (HiddenClockResetEnable dom) => Signal dom CPUIn -> Signal dom CPUOut
cpu = mealy step 0
  where
    step s CPUIn{} = let (y, s') = (defaultOutput s, s + 1) in (s', y)
```

notice the:

```haskell
logicBoard x = seqX <$> x <*> (dummy <$> cpuOut)
```

gergoerdi, 2020-04-13:
Here's my full code: https://github.com/gergoerdi/clash-brainfuck/tree/780740c1680721e0d4230f9c6b055c63c77df4a9

gergoerdi, 2020-04-13:
The following behaviour remains: if only `instr` and `memRead` (the two fields fed from memory elements) are strict in `CPUIn`, then there is no space leak in `bf-sim-logicboard`. The other fields that are connected to `logicBoard`'s inputs, don't have to be made strict for the space leak to go away.

christiaanb, 2020-04-13:
I can confirm your code really does need the `strict-mapSignal` cabal flag to be turned on to not have the space-leak, and that the stricter `mux` isn't sufficient. It will take some time to figure out why.

christiaanb, 2020-04-13:
Alright, so it seems we need the strict mux, but also this slightly change definition of `romFile#`:

```haskell
romFile# clk en sz file rd =
  -- delay clk en (deepErrorX "First value of romFile is undefined") ((content !) <$> rd)
  go (deepErrorX "First value of romFile is undefined") (fromEnable en) rd
 where
  go o (e :- es) ~(r :- rs) =
    let o' = if e then content ! r else o
    in  o `seqX` o :- (go o' es rs)
```

where I've basically inlined the definition of `delay`. I'm not quite sure what's going on...

christiaanb, 2020-04-13:
Or alternatively, when I implement `romFile#` as:

```haskell
romFile# clk en sz file rd =
  asyncRomFile# sz file `fmapS`
  delay clk en (deepErrorX "First value of romFile is undefined") rd
```

christiaanb, 2020-04-13:
I'll go with the above change to `romFile#` and the change to `mux` to close this issue (for now...)

christiaanb, 2020-04-13:
@gergoerdi the issues are solved by this proper fix: https://github.com/clash-lang/clash-compiler/pull/1258

Turns out that we were missing fixity declarations, which made our space-leak prevention in `register#` and `delay#` not work as intended. Thanks for pushing us on fixing this!! It was really helpful.

closed 2020-04-14 by martijnbastiaan

gergoerdi, 2020-04-14:
> @gergoerdi the issues are solved by this proper fix: #1258
> 
> Turns out that we were missing fixity declarations, which made our space-leak prevention in `register#` and `delay#` not work as intended. Thanks for pushing us on fixing this!! It was really helpful.

Thanks, I was starting to feel like an asshole for pushing on "my code is perfect! it is *obviously* the compiler's fault" :)

gergoerdi, 2020-04-15:
On a whim, I have tested this with `sim-hs` in https://github.com/gergoerdi/clash-bounce-bench and it seems there is a ~5% performance improvement from 702fea65584f57e0ff6cd19f3e80af92a27ed6e7 to 55757a8158d53ddb17ab8a81d37f22e584b3ea00.

PR #1258 (2020-04-14): Infix use of `defaultSeqX` and `deepSeqX` righ-assoc
In the register and delay primitive we were writing:

```
q `defaultSeqX` q :- rest
```

which was parsed as:

```
(q `defaultSeqX` q) :- rest
```

because of the missing fixity declaration, our intention was the following however: !!!

```
q `defaultSeqX` (q :- rest)
```

so that the currunt value in the register was forced to WHNF (or NF) whenever the tail of the Signal was requested in order to prevent space-leaks. Without the fixity declaration on `defaultSeqX`, this space-leak prevention was of course not happening! And we were leaking  [... cut]
commit 55757a81 (2020-04-14): Infix use of `defaultSeqX` and `deepSeqX` righ-assoc

In the register and delay primitive we were writing:

```
q `defaultSeqX` q :- rest
```

which was parsed as:

```
(q `defaultSeqX` q) :- rest
```

because of the missing fixity declaration, our intention
was the following however: !!!

```
q `defaultSeqX` (q :- rest)
```

so that the currunt value in the register was forced to
WHNF (or NF) whenever the tail of the Signal was requested
in order to prevent space-leaks. Without the fixity declaration
on `defaultSeqX`, this space-leak prevention was of course not
happening! And we were leaking [... cut]
commit 0677a8df (2020-04-14): Infix use of `defaultSeqX` and `deepSeqX` righ-assoc

In the register and delay primitive we were writing:

```
q `defaultSeqX` q :- rest
```

which was parsed as:

```
(q `defaultSeqX` q) :- rest
```

because of the missing fixity declaration, our intention
was the following however: !!!

```
q `defaultSeqX` (q :- rest)
```

so that the currunt value in the register was forced to
WHNF (or NF) whenever the tail of the Signal was requested
in order to prevent space-leaks. Without the fixity declaration
on `defaultSeqX`, this space-leak prevention was of course not
happening! And we were leaking [... cut]