clash-lang/clash-compiler #1291: How to use and install external packages

ra1u, 2020-04-26:
What would be workflow for creating and installing external haskell packages? I am running snap from release 1.2. I used to work  with cabal in past, but I am out of loop on current packet management in new environment. 

Any pointers would be much appreciated.

martijnbastiaan, 2020-04-28:
Hi @ra1u! We've been (internally) focused on Nix, so we haven't got a good flow for "snapclash" yet. having said that, I've looked into it and it seems straightforward with a few changes to the snap (see: https://github.com/clash-lang/clash-compiler/pull/1293). I plan on adding an example project soon.

I'll ping you when it's done.

ra1u, 2020-04-28:
Let me take this opportunity for asking about nix setup. How should I include cabal in default nix install to work with clash? I am unfortunately not that familiar with nix. 

Kind regards, Luka

martijnbastiaan, 2020-05-01:
@ra1u I've been working on an example project. Check it out over here: https://github.com/clash-lang/clash-compiler/tree/8f49b0e1f4ac824de96fa4587d9ee647297eeac9/clash-starters/clash-example-project#table-of-contents. The good news is that the approach mentioned in the tutorial works. The bad news is that it only works due to a number of features we introduced after releasing 1.2. So for now, it only works when you install Clash using `snap install clash --edge` (or when you use one of the other methods described there). 

The nix integration has mostly done by other people, so I'll query them and do a write up once I get the time.

martijnbastiaan, 2020-12-05:
@ra1u I've (finally!) published a nix starter project over here:

https://github.com/clash-lang/clash-starters/tree/main/simple-nix

Is that enough for you to get started with nix?

ra1u, 2020-12-05:
Thank you for your effort.

To be honest it has been quiet a while since poking with clash.
For sure I will take a look into this to put dust off some of  projects i have in my backlog.

closed 2020-12-05 by ra1u

martijnbastiaan, 2020-12-05:
Great! Let me know if there's anything you're missing.