clash-lang/clash-compiler #1308: Allow clockWizard to generate multiple output domains and consume multiple input domains

rickwebiii, 2020-05-04:
`clockWizard` only allows a single output domain, but the clock wizard in Vivado can generate multiple output clocks.

Creating multiple IP blocks and calling createWizard multiple times results in a design that doesn't generate a bitstream. The synthesis logs give me this critical warning:
```
Failed to create I/OLOGIC Route Through shape for instance clockwizard_0.clockwizard_inst_0/inst/clkin1_ibufg. Found overlapping instances within the shape: clockwizard.clockwizard_inst/inst/clkin1_ibufg and clockwizard_0.clockwizard_inst_0/inst/clkin1_ibufg
```
which I think is telling me the input buffer on the PLL is a dedicated resource that can't be assigned to 2 different IP components. (That would make sense to me anyways)

The IP generator also supports multiple input clocks, which isn't supported by `clockWizard`.

Ideally, I can call a new function clockWizardMN and pass Vecs for input and output domain symbols and get returned a Vec of tuples. This would allow the existing function to continue to exist with the same interface and not break existing users.

martijnbastiaan, 2020-05-04:
Hi again @rickwebiii :)

This exactly what we did for Intel's alteraPll:

https://github.com/clash-lang/clash-compiler/blob/5517ebbbf4ac1e31486cdfa1003e9b623174702f/clash-prelude/src/Clash/Intel/ClockGen.hs#L59-L98

The corresponding HDL is generated by Haskell blackboxes:

https://github.com/clash-lang/clash-compiler/blob/5517ebbbf4ac1e31486cdfa1003e9b623174702f/clash-lib/prims/common/Clash_Intel_ClockGen.json#L15-L29

and

https://github.com/clash-lang/clash-compiler/blob/5517ebbbf4ac1e31486cdfa1003e9b623174702f/clash-lib/src/Clash/Primitives/Intel/ClockGen.hs

My guess is that the Xilinx ones would look very similar!

christiaanb, 2020-05-04:
Yes, except that instead of generating `.qsys` files, we would have to generate `.tcl` files that invoke the `create_ip` function to instantiate the MCMM or ClockWizard with the correct parameters.

rickwebiii, 2020-05-04:
I'm using Clash 1.2.0, in case something has changed since then.

Today, the createWizard function today doesn't call invoke the clock wizard as far as I could tell. I believe it just assumes a black box with the shape of something that takes 4 signals with the SSymbol name you give it. I think this behavior is fine (I'd almost prefer it) as it separates the how you get a clocks of the speeds you want (go run clock wizard in vivado and copy sources or make a tcl script) from how you consume it in Clash.

Generating tcl files that I have to go run after calling clash --vhdl isn't a deal breaker, but it does tie Clash more closely to Xilinx's toolchain rather than just using a generic black box. I'd eventually like to experiment with Symbiflow for making my projects and I don't know if they use Xilinx IP in exactly the same way or not.

I guess another option is for users to just make a topentitywrapper that injects multiple clock domains into clash's top entity.

christiaanb, 2020-05-05:
Ah, I see, I hadn't considered using something else than the vendor's toolchain for synthesis yet. Generating the `.tcl` is definitely not mandatory, it's mostly an ease of use thing; i.e. we can follow @martijnbastiaan suggestion of reusing most of the machinery we made for Intel's `alteraPll` but skip the `"includes"` section of the primitive/blackbox definition.