clash-lang/clash-compiler #1314: Output directory structure doesn't contain full module name

gergoerdi, 2020-05-07:
I have a file `MyCircuit/Foo/Top.hs` with the following contents:

```
module MyCircuit.Foo.Top where

import Clash.Prelude
import Clash.Annotations.TH

topEntity
    :: "CLK" ::: Clock System
    -> "INP" ::: Signal System Bit
    -> "OUT" ::: Signal System Bit
topEntity clk = withClock clk $ fmap complement 

makeTopEntity 'topEntity
```

If I synthesize this with

```
clash -outputdir /tmp/foo --verilog MyCircuit.Foo.Top
```

then the resulting files are put in `verilog/MyCircuit/topEntity`, instead of `verilog.MyCircuit.Foo.Top.topEntity`:

```
$ tree /tmp/foo/verilog/
/tmp/foo/verilog/
└── MyCircuit
    └── topEntity
        ├── topEntity.manifest
        └── topEntity.v

2 directories, 2 files
```

This is surprising, and makes code that needs to programmatically access the `.manifest` file, and the generated code, needlessly ad-hoc.

christiaanb, 2020-05-17:
I guess it is somewhat arbitrary, and can lead to issues when you have a `topEntity` without a Synthesize annotation in both `Foo.Bar` and `Foo.Bax`; as two separate runs, one for `Foo.Bar` and one for `Foo.Bax`, will output files too `/outputdir/hdl/Foo`

gergoerdi, 2021-02-26:
Can this be fixed for 1.4 so that it can be included with other incompatible changes?

martijnbastiaan, 2021-02-26:
Wouldn't it make more sense to put files in `$outputdir/$qualifiedname`? E.g., `$outputdir/MyCircuit.Foo.Top.topEntity`.

martijnbastiaan, 2021-02-26:
@gergoerdi https://github.com/clash-lang/clash-compiler/blob/a397d9bb4a484bdc8822c2d4b797195c9ecd53e2/clash-lib/src/Clash/Driver.hs#L343-L345 Would you be willing to make a PR?

martijnbastiaan, 2021-02-26:
@gergoerdi Christiaan just pinged me saying the patch might not be as simple as it seems, as it affects our testsuite. If you're up for it anyway, let me know. Otherwise, I'll pick it up after my lunch break (in ~3 hours).

martijnbastiaan, 2021-02-26:
So there's been some ~~bikeshedding~~ discussion on our Slack and we've settled on:

* By default: `$hdl/$qualifiedname`
* If `hdldir` flag is set or if GHC output dir flags are set (this is a hidden feature currently in there! :-)): `$hdldir/$qualifiedname/`

There was some discussion on what to do when we add a multi-hdl-output feature to Clash (e.g., generate VHDL + Verilog from one normalization result) too. If it gets that far, we'd introduce the use of variables in `-fclash-hdldir`. So users would be able to specify things like: `-fclash-hdldir={target_hdl}/{name_slug}`.

gergoerdi, 2021-02-26:
> If you're up for it anyway, let me know. Otherwise, I'll pick it up after my lunch break (in ~3 hours).

Bad timing, I'm in a seminar...

martijnbastiaan, 2021-02-26:
Np! I'll pick it up!

closed 2021-03-09 by martijnbastiaan

martijnbastiaan, 2021-03-09:
@gergoerdi So this turned out to be quite annoying to fix as our test suite quite heavily relied on a lot of implicit assumptions. They've now been generalized and Clash generates a separate folder for every top entity. See the example in https://github.com/clash-lang/clash-compiler/pull/1687.