clash-lang/clash-compiler #1337: `inlineCleanup` introduces free variables

martijnbastiaan, 2020-05-18:
Compiling the following:

```haskell
{-# LANGUAGE GADTs               #-}
{-# LANGUAGE NoImplicitPrelude   #-}
{-# LANGUAGE RecordWildCards     #-}

module Test where

import Clash.Prelude

type Lanes = 2

data SigFn a b = SigFn { runSigFn :: SigType a -> SigType b }

class Sig a where
  type SigType a

instance Sig Bit where
  type SigType Bit = Signal System Bit

instance Sig a => Sig (Vec n a) where
  type SigType (Vec n a) = Vec n (SigType a)

instance (Sig a, Sig b) => Sig (a, b) where
  type SigType (a, b) = (SigType a, SigType b)

singletonSigFn :: SigFn a (Vec 1 a)
singletonSigFn = SigFn singleton

headSigFn :: SigFn (Vec 1 a) a
headSigFn = SigFn head

defSigFn :: Default (SigType t) => SigFn s t
defSigFn = SigFn (const def)

(>>>) :: SigFn a b -> SigFn b c -> SigFn a c
SigFn f >>> SigFn g = SigFn (g . f)

(<=>) :: SigFn a b -> SigFn c d -> SigFn (a, c) (b, d)
SigFn f <=> SigFn g = SigFn (\(x, y) -> (f x, g y))

splitSigFn :: KnownNat n => SigFn (Vec (n + m) a) (Vec n a, Vec m a)
splitSigFn = SigFn splitAtI

catSigFn :: SigFn (Vec n a, Vec m a) (Vec (n + m) a)
catSigFn = SigFn (\(x, y) -> x ++ y)

data BitLayout t where
  BitLayout :: forall bitCount t
            .  ( KnownNat bitCount )
            => { bitCount :: SNat bitCount
               , inFn :: SigFn t (Vec bitCount Bit)
               , outFn :: SigFn (Vec bitCount Bit) t
               }
            -> BitLayout t

unitLayout :: BitLayout Bit
unitLayout = BitLayout SNat singletonSigFn headSigFn

voidLayout :: forall t . Default (SigType t) => BitLayout t
voidLayout = BitLayout d0 defSigFn defSigFn

catLayout
  :: forall n m s
  .  ( KnownNat n
     , KnownNat m
     , 1 <= n )
  => BitLayout (Vec n s)
  -> BitLayout (Vec m s)
[... 1257 lines cut ...]
                    (rest[11] :: Clash.Sized.Vector.Vec[8214565720323788085]
                                   1
                                   (Clash.Sized.Internal.Index.Index[8214565720323788078]
                                      (GHC.TypeNats.+[3674937295934325068] 2 1))) ->
                  el[12][LocalId])
             (Clash.Sized.Vector.Nil[8214565720323788089] @0
                @(Clash.Sized.Internal.Index.Index[8214565720323788078]
                    (GHC.TypeNats.+[3674937295934325068] 2 1))
                _CO_))
    in result[31996][LocalId]
    It introduces free variables.
    Before: c$arg[17344] :: Test.SigType[8214565720323817080]
                      (Clash.Sized.Vector.Vec[8214565720323788085]
                         2
                         Clash.Sized.Internal.BitVector.Bit[8214565720323788076])
    After: elH0[279] :: Clash.Sized.Internal.Index.Index[8214565720323788078]
                   (GHC.TypeNats.+[3674937295934325068] 2 1)
    c$arg[17344] :: Test.SigType[8214565720323817080]
                      (Clash.Sized.Vector.Vec[8214565720323788085]
                         2
                         Clash.Sized.Internal.BitVector.Bit[8214565720323788076])
    CallStack (from HasCallStack):
      error, called at src/Clash/Rewrite/Util.hs:232:13 in clash-lib-1.3.0-inplace:Clash.Rewrite.Util

```

martijnbastiaan, 2020-05-18:
Ah, this only triggers after this patch:

```diff
diff --git a/clash-lib/src/Clash/Normalize/Transformations.hs b/clash-lib/src/Clash/Normalize/Transformations.hs
index 8706b3cc..bb2dde34 100644
--- a/clash-lib/src/Clash/Normalize/Transformations.hs
+++ b/clash-lib/src/Clash/Normalize/Transformations.hs
@@ -2082,8 +2082,8 @@ reduceNonRepPrim c@(TransformContext is0 ctx) e@(App _ _) | (Prim p, args, ticks
         case runExcept (tyNatSize tcm nTy) of
           Right n -> do
             shouldReduce1 <- List.orM [ pure (ultra || n < 2)
-                                 , shouldReduce ctx
-                                 , List.anyM isUntranslatableType_not_poly
+                                      , shouldReduce ctx
+                                      , List.anyM isUntranslatableType_not_poly
                                         [lhsElTy,rhsElty,resElTy] ]
             if shouldReduce1
                then let [fun,lhsArg,rhsArg] = Either.lefts args
@@ -2096,8 +2096,8 @@ reduceNonRepPrim c@(TransformContext is0 ctx) e@(App _ _) | (Prim p, args, ticks
         case runExcept (tyNatSize tcm nTy) of
           Right n -> do
             shouldReduce1 <- List.orM [ pure (ultra || n < 2 )
-                                 , shouldReduce ctx
-                                 , List.anyM isUntranslatableType_not_poly
+                                      , shouldReduce ctx
+                                      , List.anyM isUntranslatableType_not_poly
                                         [argElTy,resElTy] ]
             if shouldReduce1
                then let [fun,arg] = Either.lefts args
@@ -2116,8 +2116,8 @@ reduceNonRepPrim c@(TransformContext is0 ctx) e@(App _ _) | (Prim p, args, ticks
         case runExcept (tyNatSize tcm nTy) of
           Right n -> do
             shouldReduce1 <- List.orM [ pure (ultra || n == 0)
-                                 , shouldReduce ctx
-                                 , isUntranslatableType_not_poly aTy ]
+                                      , shouldReduce ctx
+                                      , isUntranslatableType_not_poly aTy ]
             if shouldReduce1 then
               let [fun,arg] = Either.lefts args
               in  (`mkTicks` ticks) <$> reduceFold c (n + 1) aTy fun arg
@@ -2128,8 +2128,8 @@ reduceNonRepPrim c@(TransformContext is0 ctx) e@(App _ _) | (Prim p, args, ticks
         in  case runExcept (tyNatSize tcm nTy) of
           Right n -> do
             shouldReduce1 <- List.orM [ pure ultra
-                                 , shouldReduce ctx
-                                 , List.anyM isUntranslatableType_not_poly [aTy,bTy] ]
+                                      , shouldReduce ctx
+                                      , List.anyM isUntranslatableType_not_poly [aTy,bTy] ]
             if shouldReduce1
               then let [fun,start,arg] = Either.lefts args
                    in  (`mkTicks` ticks) <$> reduceFoldr c n aTy fun start arg
@@ -2148,8 +2148,9 @@ reduceNonRepPrim c@(TransformContext is0 ctx) e@(App _ _) | (Prim p, args, ticks
                 | n == 0 -> changed rArg
                 | m == 0 -> changed lArg
                 | otherwise -> do
-                    shouldReduce1 <- List.orM [ shouldReduce ctx
-                                         , isUntranslatableType_not_poly aTy ]
+                    shouldReduce1 <- List.orM [ pure ultra
+                                              , shouldReduce ctx
+                                              , isUntranslatableType_not_poly aTy ]
                     if shouldReduce1
                        then (`mkTicks` ticks) <$> reduceAppend is0 n m aTy lArg rArg
                        else return e
@@ -2159,8 +2160,9 @@ reduceNonRepPrim c@(TransformContext is0 ctx) e@(App _ _) | (Prim p, args, ticks
             [vArg]    = Either.lefts args
         case runExcept (tyNatSize tcm nTy) of
           Right n -> do
-            shouldReduce1 <- List.orM [ shouldReduce ctx
-                                 , isUntranslatableType_not_poly aTy ]
+            shouldReduce1 <- List.orM [ pure ultra
[... 82 lines cut ...]
         case runExcept (tyNatSize tcm nTy) of
           Right n -> do
             shouldReduce1 <- List.orM [ pure (ultra || n < 2)
-                                 , shouldReduce ctx
-                                 , List.anyM isUntranslatableType_not_poly [argElTy,resElTy] ]
+                                      , shouldReduce ctx
+                                      , List.anyM isUntranslatableType_not_poly [argElTy,resElTy] ]
             if shouldReduce1
                then let [_,fun,arg] = Either.lefts args
                     in  (`mkTicks` ticks) <$> reduceImap c n argElTy resElTy fun arg
@@ -2279,8 +2285,9 @@ reduceNonRepPrim c@(TransformContext is0 ctx) e@(App _ _) | (Prim p, args, ticks
         let ([_sArg,vArg],[nTy,aTy]) = Either.partitionEithers args
         case runExcept (tyNatSize tcm nTy) of
           Right n -> do
-            shouldReduce1 <- List.orM [ shouldReduce ctx
-                                 , isUntranslatableType False aTy ]
+            shouldReduce1 <- List.orM [ pure ultra
+                                      , shouldReduce ctx
+                                      , isUntranslatableType False aTy ]
             if shouldReduce1
                then (`mkTicks` ticks) <$> reduceTReplicate n aTy eTy vArg
                else return e
```

so it might just be `reduceNonRepPrim` violating some assumption.

martijnbastiaan, 2020-05-18:
Actually, I just need to make `init`, `head`, or both sensitive to `-fclash-compile-ultra` and this happens.

leonschoorl, 2020-05-18:
A lot of that patch just seems to be whitespace changes

commit d07f08e3 (2020-05-20): Carry substitution all the way through `inlineBndrsCleanup` loop. Fix #1337
commit 44ab8700 (2020-05-20): Carry substitution all the way through `inlineBndrsCleanup` loop. Fix #1337
commit 2746782d (2020-05-20): Carry substitution all the way through `inlineBndrsCleanup` loop. Fix #1337
commit 30709445 (2020-05-20): Inline left-over binders in recursive binders in `inlineBndrsCleanup`.

Fixes #1337
closed 2020-05-25 by christiaanb