clash-lang/clash-compiler #1345: Seemingly small change drastically impairs circuit complexity

gergoerdi, 2020-05-25:
I have a small circuit that generates VGA signal from a frame buffer. A tiny refactoring that really just moves code around turns the generated Verilog from 1052 lines to 2948 lines (so ~3 times as much Verilog), and the Verilator performance also goes down from 

> 60 frames at 3938 ms, 15.24 fps

to 

> 60 frames at 7155 ms, 8.39 fps

The commit in question is [this one](https://github.com/gergoerdi/clash-draw-toy/commit/58bd6df890b41f713a827c6c07afd25d2b02c220); as you can see, all it does is it moves the VGA sync signal generation from the `topEntity` to another function.

christiaanb, 2020-05-25:
Just to humor me, could you add `-XNoMonomorphismRestriction` to the list of language pragma's at the top of the file and check whether it makes a difference.

gergoerdi, 2020-05-25:
Indeed, with the monomorphism restriction turned on, I am back in the "good" state.

christiaanb, 2020-05-25:
Right... we should really go through with https://github.com/clash-lang/clash-compiler/issues/1270 then.

gergoerdi, 2020-05-25:
So what is being duplicated here? The type of `vgaX` and `vgaY` is monomorphic (because `vga640x480at60` specifies it fully), and I'd think this is enough to force everything else to also be...?

christiaanb, 2020-05-25:
You would have to look at the outputs of `-ddump-simpl` for with and without the monomorphism restriction. You might see certain let-bindings lifted out because they take a dictionary as argument when the monomorphism restriction is disabled, which leads to loss of sharing.

christiaanb, 2020-05-25:
Another option is to compile with `-XMonomorphismRestriction -Wmonomorphism-restriction` to see where the monomorhism restriction kicked in.

christiaanb, 2020-05-26:
Doing the latter I see:

```
/home/christiaan/clash-draw-toy/src/DrawToy.hs:81:5: warning: [-Wmonomorphism-restriction]
    • The Monomorphism Restriction applies to the bindings
      for ‘vgaSync’, ‘vgaX’, ‘vgaY’
        Consider giving them a type signature
    • In an equation for ‘drawToy’:
          drawToy input
            = (frameEnd, delayVGA vgaSync rgb)
            where
                VGADriver {..} = vgaDriver vga640x480at60
                frameEnd = isFalling False (isJust <$> vgaY)
                x = center @128 . scale (SNat @5) $ vgaX
                y = center @64 . scale (SNat @5) $ vgaY
                ....
   |
81 |     VGADriver{..} = vgaDriver vga640x480at60
   |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
```

gergoerdi, 2020-05-26:
Is it because the fields of `VGADriver` don't, individually, determine the full type of `VGADriver` itself?

```
data VGADriver dom w h = VGADriver
    { vgaSync :: VGASync dom
    , vgaX :: Signal dom (Maybe (Index w))
    , vgaY :: Signal dom (Maybe (Index h))
    }
```

So if you have `vgaX :: Signal dom0 (Maybe (Index w0))`, that means the full `VGADriver` is still polymorphic in `h`.

What I find strange is that in this particular case, the `VGADriver` in question is defined as

```
vgaDriver vga640x480at60
```

Which should be monomorphic with `w ~ 640` and `h ~ 480`, since the type of `vgaDriver` is:

```
vgaDriver
    :: (HiddenClockResetEnable dom, KnownNat w, KnownNat h)
    => (DomainPeriod dom ~ ps)
    => VGATimings ps w h
    -> VGADriver dom w h

vga640x480at60 :: VGATimings (HzToPeriod 25_175_000) 640 480
```

So all in all, I am still a bit confused.

christiaanb, 2020-05-26:
Perhaps GHC distinguishes between: `x :: (Num p, p ~ Int) => p)`  and `x :: Int` where the former is considered polymorphic (though constrained)

commit 7134eff0 (2020-05-26): Do not disable the monomorphism restriction

Warn when it is turned off when converting Hs to HDL. Also
enable -Wmonomorphism-restriction by default so you know when
you might want to add a type signature to prevent unexpected
monomorphism.

This commit also demonstrates the fallout of leaving the
MonomorphismRestriction enabled by default, and can be used
as a guide for fixing Clash code suffering from the same
issues once this commit lands in a released version of the
compiler.

Fixes #1270
Fixes #1345
commit 6187b964 (2020-05-26): Do not disable the monomorphism restriction

Warn when it is turned off when converting Hs to HDL. Also
enable -Wmonomorphism-restriction by default so you know when
you might want to add a type signature to prevent unexpected
monomorphism.

This commit also demonstrates the fallout of leaving the
MonomorphismRestriction enabled by default, and can be used
as a guide for fixing Clash code suffering from the same
issues once this commit lands in a released version of the
compiler.

Fixes #1270
Fixes #1345
commit 46c5468c (2020-05-28): Do not disable the monomorphism restriction

Warn when it is turned off when converting Hs to HDL. Also
enable -Wmonomorphism-restriction by default so you know when
you might want to add a type signature to prevent unexpected
monomorphism.

This commit also demonstrates the fallout of leaving the
MonomorphismRestriction enabled by default, and can be used
as a guide for fixing Clash code suffering from the same
issues once this commit lands in a released version of the
compiler.

Fixes #1270
Fixes #1345
PR #1349 (2020-06-02): Do not disable the monomorphism restriction
Warn when it is turned off when converting Hs to HDL. Also enable -Wmonomorphism-restriction by default so you know when you might want to add a type signature to prevent unexpected monomorphism.

This commit also demonstrates the fallout of leaving the `MonomorphismRestriction` enabled by default, and can be used as a guide for fixing Clash code suffering from the same
issues once this commit lands in a released version of the compiler.

Fixes #1270
Fixes #1345
closed 2020-06-02 by christiaanb