clash-lang/clash-compiler #1359: Updated tutorial on CLASH with an FPGA

apache8080, 2020-06-01:
I was trying to follow this [tutorial ](http://christiaanb.github.io/posts/clash-fpga-starter/) with my Terasuc DE0-Nano board but it is out of date for the new CLASH version. Is there any updated tutorial out there on how to use clash to program an Altera FPGA?

rowanG077, 2020-06-01:
The updated code used in that tutorial is located in this repo as well.

https://github.com/clash-lang/clash-compiler/blob/master/examples/Blinker.hs

It contains a lot more comments to explain what's going on.

apache8080, 2020-06-01:
I tried using that code and following the rest of the tutorial and it was able to synthesize but when I upload it to the FPGA it does nothing. Does something extra need to be done with the .qsys file that is generated?

rowanG077, 2020-06-01:
What step created a qsys file? The tutorial doesn't use the system designer and no qsys file should be generated.

apache8080, 2020-06-01:
If your run `stack run -- clash --vhdl examples/Blinker.hs` then it generates the vhdl files but it also generates the .qsys file for the altpll50

rowanG077, 2020-06-01:
Can you attach your Quartus project so I can take a look? If Clash has generated a valid qsys file then it should be as simple as adding that to the project as well and running quartus build pipeline. You can ignore the entire MegaWizard pll stuff from the tutorial you linked.

apache8080, 2020-06-02:
This is what CLASH generates:
```
altpllEE5D850CE4873837.qsys  blinker.manifest  blinker.vhdl  blinker_types.vhdl
```
When I added the vhdl files and the qsys files quartus asked me to upgrade the IP automatically. I did that and also ignored the MegaWizard step but after programming the fpga nothing happens.

I can't attach the QPF file to this github issue but here are the following steps I did:
- Used the Terasic DE0-Nano System Builder to generate the Quartus project file
- Modified the .qsf file to use CLOCK_50[0], KEY0[0], KEY1[0]
- Modified the .sdc file to say `derive_pll_clocks`
- Then added the `altpllEE5D850CE4873837.qsys  blinker.manifest  blinker.vhdl  blinker_types.vhdl` files to the project
- compiled the code
- used the programmer to flash the FPGA

rowanG077, 2020-06-02:
That looks good to me. Could you try the following version which removes the PLL entirely:

```Haskell
module Blinker where

import Clash.Prelude

data LedMode
  = Rotate
  -- ^ After some period, rotate active led to the left
  | Complement
  -- ^ After some period, turn on all disable LEDs, and vice versa
  deriving (Generic, NFDataX)

-- Define a synthesis domain with a clock with a period of 20000 /ps/.
createDomain vSystem{vName="Input", vPeriod=20000}

{-# ANN topEntity
  (Synthesize
    { t_name   = "blinker"
    , t_inputs = [ PortName "CLOCK_50"
                 , PortName "KEY0"
                 , PortName "KEY1"
                 ]
    , t_output = PortName "LED"
    }) #-}
topEntity
  :: Clock Input
  -- ^ Incoming clock
  -> Signal Input Bool
  -- ^ Reset signal, straight from KEY0
  -> Signal Input Bit
  -- ^ Mode choice, straight from KEY1. See 'LedMode'.
  -> Signal Input (BitVector 8)
  -- ^ Output containing 8 bits, corresponding to 8 LEDs
topEntity clk20 rstBtn modeBtn =
  exposeClockResetEnable
    (mealy blinkerT initialStateBlinkerT . isRising 1)
    clk20
    rst
    en
    modeBtn
 where
  -- | Enable line for subcomponents: we'll keep it always running
  en = enableGen

  -- Start with the first LED turned on, in rotate mode, with the counter on zero
  initialStateBlinkerT = (1, Rotate, 0)

  -- Signal coming from the reset button is low when pressed, and high when
  -- not pressed. We convert this signal to the polarity of our domain with
  -- 'unsafeFromActiveLow'.
  rst = unsafeFromLowPolarity rstBtn

flipMode :: LedMode -> LedMode
flipMode Rotate = Complement
flipMode Complement = Rotate

blinkerT
  :: (BitVector 8, LedMode, Index 16650001)
  -> Bool
  -> ((BitVector 8, LedMode, Index 16650001), BitVector 8)
blinkerT (leds, mode, cntr) key1R = ((leds', mode', cntr'), leds)
  where
    -- clock frequency = 50e6  (50 MHz)
    -- led update rate = 333e-3 (every 333ms)
    cnt_max = 16650000 :: Index 16650001 -- 50e6 * 333e-3

    cntr' | cntr == cnt_max = 0
          | otherwise       = cntr + 1

    mode' | key1R     = flipMode mode
          | otherwise = mode

    leds' | cntr == 0 =
              case mode of
                Rotate -> rotateL leds 1
                Complement -> complement leds
          | otherwise = leds
```

If that still doesn't work try an even simpler circuit without inputs(no resets even) that only counts up and outputs it to the LEDs.

christiaanb, 2020-07-12:
I've updated the tutorial to the latest version of Clash over at: https://qbaylogic.com/blog/2020/07/09/new-clash-fpga-starter.html

closed 2020-07-12 by christiaanb