clash-lang/clash-compiler #1360: Problem synthesising toSLV for sum types in VHDL with Vivado with -fclash-hdlsyn Vivado

xavieran, 2020-06-02:
Please refer to the following gist for code:
https://gist.github.com/xavieran/f7aa0409bbb42ed4550b553b276e41ac

When compiling this clash 
```
data StatusA = InactiveA | ActiveA

topEntity :: HiddenClockResetEnable System =>
  Signal System (Vec 2 StatusA) -> Signal System (Index 2) -> Signal System (Vec 2 StatusA)
topEntity input idx = out
  where
    out = replace <$> idx <*> pure InactiveA <*> input
```
with -fclash-hdlsyn Vivado, the "replace" function sets the value of the Vec to 
`ivec(vec_index) := test_types.toSLV("0");`
instead of 
`ivec(vec_index) := "0";`
without -fclash-hdlsyn Vivado

This causes an error during synthesis with Vivado (2019.2)
```
ERROR: [Synth 8-2380] near toslv ; 3 visible identifiers match here [~/vhdl/Test/topentity.vhdl:53]
INFO: [Synth 8-1047] first match for 'toslv' found here [~/vhdl/Test/test_types.vhdl:22]
INFO: [Synth 8-1047] another match for 'toslv' found here [~/vhdl/Test/test_types.vhdl:26]
INFO: [Synth 8-1047] another match for 'toslv' found here [~/vhdl/Test/test_types.vhdl:30]
INFO: [Synth 8-2810] unit structural ignored due to previous errors [~/vhdl/Test/topentity.vhdl:22]
```

It seems that toSLV is ambiguous when applied to the "0".

christiaanb, 2020-06-02:
Right, it should type-qualify the argument so it knows which `toSLV` to use.

closed 2020-06-11 by martijnbastiaan

xavieran, 2020-06-12:
Thanks for the fix!

commit 004f7244 (2020-06-09): VHDL: automatically generate a type qualification inside ~TOBV

This makes sure VHDL always knows which toSLV to call.
Fixes #1360

This also leave out to- and fromSLV calls when type is already std_logic_vector.
And fixes the layout of the entity declaration end keyword.
commit 0bd71766 (2020-06-11): VHDL: automatically generate a type qualification inside ~TOBV

This makes sure VHDL always knows which toSLV to call.
Fixes #1360

This also leave out to- and fromSLV calls when type is already std_logic_vector.
And fixes the layout of the entity declaration end keyword.
PR #1365 (2020-06-11): VHDL: automatically generate a type qualification inside ~TOBV
This makes sure VHDL always knows which `toSLV `to call.
Fixes #1360

This also leave out `to`- and `fromSLV` calls when the type is already `std_logic_vector`.
And fixes the layout of the entity declaration `end` keyword.
commit ff4f2e84 (2020-06-11): VHDL: automatically generate a type qualification inside ~TOBV

This makes sure VHDL always knows which toSLV to call.
Fixes #1360

This also leave out to- and fromSLV calls when type is already std_logic_vector.
And fixes the layout of the entity declaration end keyword.