clash-lang/clash-compiler #1363: `delayI` vs `delayedI`

gergoerdi, 2020-06-07:
In `Clash.Signal.Delayed` we have both `delayI` and `delayedI` with the same type and same meaning; it is not clear which one I should use in my book. I assume one of them should be deprecated?

Looking at other definitions in that same module, we have functionality that exists with the name `delayed`, and functionality which exists with the name `delayN`, so it is not clear if it should be `delay` / `delayI` / `delayN` or `delayed` / `delayedI` / `delayedN`...

christiaanb, 2020-06-07:
Right... so `delayI` and `delayedI` used to be different. `delayI` used to initialize the registers with `deepErrorX "initial value"`, while `delayedI` would initialize the registers with `Default.def`. While these day they both take an initialization argument. The difference that still remains is that `delayed / delayedI` have a reset, while `delay / delayI / delayN` do not have a reset.

martijnbastiaan, 2020-06-07:
We could do with a makeover IMO. Ideally, we should copy all register-functions from the explicit prelude and make them into a delayed variant, exposing them under the same name. We could keep the old API exposed as `Clash.Deprecated_12.Explicit.Signal.Delayed` or something similar.

christiaanb, 2020-06-08:
I wonder if we should introduce a `clash-prelude-compat` package to deal with using newer prelude's in older code bases; instead of polluting `clash-prelude`.

martijnbastiaan, 2020-06-08:
Excellent idea :)