clash-lang/clash-compiler #1364: `satSub` has its arguments flipped compared to normal `subtract`

gergoerdi, 2020-06-07:
It seems that the meaning of `satSub _ x y` is `x - y` (modulo whatever saturating behaviour you specify), which can be very surprising if you are used to the Haskell prelude function `subtract`, which is the other way round.

In other words, the partially applied `subtract x` is the function that subtracts `x` from whatever it is applied on. I think it would make more sense if `satSub _ x` was similarly the function that subtracts `x`.

christiaanb, 2020-06-07:
Mmm... I get the confusion w.r.t. the Haskell Prelude `subtract`. But wouldn't it be confusing to have:
```
satAdd _ x y  ~ x + y
satMul _ x y ~ x * y
satSub _ x y ~ y - x
```

gergoerdi, 2020-06-08:
Sorry, I gave myself a full day, but I just can't not take the bait. It's too good a trolling opportunity.

How about

```
satAdd _ x y ~ y + x
satMul _ x y ~ y * x
satSub _ x y ~ y - x
```
Here's a patch that I think changes `satMul` and `satAdd` in all existing `ExtendingNum` instances to this new semantics:

```
```

christiaanb, 2020-06-08:
I think I'd rather introduce a new `satSubtract` then, as existing code would break quite unexpectedly.

christiaanb, 2020-06-10:
I guess if we make a `clash-prelude-compat` package as suggested in https://github.com/clash-lang/clash-compiler/issues/1363 we can flip the arguments.

dramforever, 2020-07-03:
Maybe we could give them operator aliases? I'm thinking something suggesting the 'bounded' nature of the operations, so maybe [`>+<` `>-<` `>*<`] or [`|+|` `|-|` `|*|`].