clash-lang/clash-compiler #1386: Use Edalize to interface EDA tools

olofk, 2020-06-16:
[Edalize](https://github.com/olofk/edalize) is an abstraction library for interfacing EDA tools Edalize uses language-agnostic yaml for configuration and support 20+ EDA tools.

To make use of FuseSoC, Clash would only need to generate an [EDAM](https://edalize.readthedocs.io/en/latest/edam/api.html) file to go with the generated HDL (+constraints and what else uou need) and pass it to Edalize

christiaanb, 2020-06-16:
Thanks for the suggestion. We’ll definitely take a look, and ask questions when we need some guidance on what exact info we need to put in the EDAM files.

christiaanb, 2020-06-17:
@olofk so for https://github.com/clash-lang/clash-compiler/blob/master/examples/Blinker.hs Clash generates the following vhdl files:
* `blinker.vhdl`
* `blinker_types.vhdl`
* `altpllEE5D850CE4873837.qsys`

Where the top-level entity of the design is called `blinker`.

Is it then sufficient for Clash to generate the following `blinker.edam` file:
```json
{ "files" : [ {"name" : "blinker.vhdl", "file_type" : "vhdlSource" }
            , {"name" : "blinker_types.vhdl", "file_type" : "vhdlSource" }
            , {"name" : "altpllEE5D850CE4873837.qsys", "file_type" : "QSYS" }
            ] 
, "name" : "blinker"
, "toplevel" : "blinker"
}                 ,
```
or does it need to do more?

And also, given the above EDAM file, how would one invoke FuseSoC (the executable, not the Edalize library) to build a bitstream using Quartus?

olofk, 2020-06-17:
The EDAM looks correct to me from a first look but if you want to run this through quartus you would also need to specify some info about the part you're targeting (device, family etc) and constraints

There are two ways to go about this. Either use FuseSoC or use Edalize directly. My first thought was that using Edalize would be the preferred way for your workflow but I'm not sure so let me just summarize the options. 

If you're using FuseSoC, you would create a [core description file](https://fusesoc.readthedocs.io/en/master/ref/capi2.html) instead. This would be the preferred way if you want to combine clash sources with existing verilog/vhdl cores since you can pull in the latter as dependencies. You could even specify to build the verilog/vhdl from clash on-the-fly using a [generator](https://fusesoc.readthedocs.io/en/master/user/generators.html). Converting HDLs not directly supported by EDA toools (e.g. clash, migen, chisel, spinalhdl...) into verilog/vhdl is one of the main intended uses of generators.

A core file might look like this

```yaml
CAPI=2:

name : ::blinker:0

filesets:
  rtl:
    files:
      - blinker.vhdl
      - blinker_types.vhdl
      - altpllEE5D850CE4873837.qsys : {file_type : QSYS}
    file_type : vhdlSource

targets:
  default:
    filesets : [rtl]
    toplevel : blinker
```

You would then run with `fusesoc run blinker`. Note however that this wouldn't work right away since we haven't told FuseSoC/Edalize which FPGA device we're targeting and there are no constraints specified and so on.
(An example using a generator to automatically convert the clash source would look a bit different)

Using Edalize directly might be preferred if you want to bring it into an existing build system just to talk to EDA tools. I've been meaning to add an executable wrapper around the library, but forgot that I haven't done that yet so I can see why you got confused how to call edalize. :)

I feel that the above was not a super pedagogical explanation but I'm happy to work with you to see how we can make the best use of this.

christiaanb, 2020-06-17:
Thanks for the details. That clears up the misunderstanding. I was just confused by your initial suggestion that:
> To make use of FuseSoC, Clash would only need to generate an EDAM file to go with the generated HDL

And I couldn't figure out from the documentation how FuseSoC could be called with anything else than a `.core` file, so I assumed there was some undocumented feature that allows you to use EDAM files directly.

You're detailed explanation above is sufficient to move forward at this point. We'll come back with any questions once we've explored some more.

PR #1410 (2020-07-01): Generate EDAM file in Clash
Fixes #1386 

This PR allows clash to optionally generate metadata for use with Edalize, by using the `-fclash-edalize` flag.
The generated file includes dummy configuration for tools which are favoured by Clash (i.e. for testing purposes)
with fields that need to be specified before Edalize can be used marked as `TODO`.
closed 2020-07-01 by alex-mckenna