clash-lang/clash-compiler #1387: Generate SDC file with clock specifications

christiaanb, 2020-06-17:
Clash should generate an SDC file with `create_clock` commands for all `Clock` arguments of functions that are a synthesis boundary (`topEntity` or functions with a `Synthesize` annotation). For example, for https://github.com/clash-lang/clash-compiler/blob/master/examples/Blinker.hs  Clash should generate a file called `blinker.sdc` with the following content:
```tcl
create_clock -name {CLOCK_50} -period 20.000 -waveform { 0.000 10.000 } [get_ports {CLOCK_50}]
```

alex-mckenna, 2020-06-18:
Should this change based on `hdlSyn`? Vivado uses XDC which is (from the looks of it) SDC with some Xilinx-specific extensions. A cursory search also seems to suggest SDC is largely associated with Quartus, although that may just be because Quartus is so widespread

christiaanb, 2020-06-18:
Synopsys Design Constraint, i.e. SDC, is the industry-standard name; but I guess we could change the extension to `.xdc` if the `hdlSyn` says we're building for Vivado.

alex-mckenna, 2020-06-18:
If it's industry standard, presumably Vivado will accept it anyway (at least, you would hope it does). I was more curious about whether this was aimed at a specific EDA tool or not

PR #1400 (2020-06-26): Generate SDC files for top entity
Fixes #1387

This PR uses createHDL to generate SDC files for each component
with a clock as an input. If a component has multiple clocks as an
input, each clock will appear in the SDC file.
closed 2020-06-26 by alex-mckenna