clash-lang/clash-compiler #1388: Clash generates illegal VHDL (according to Modelsim)

xavieran, 2020-06-19:
clash-compiler commit: e230a8713f92b356debe1e9e775c6346a1bc7f39
Vivado 2019.2
Modelsim 2020.1

Please refer to the following gist (test.hs)
https://gist.github.com/xavieran/b5b73a7330892cb8f7a9f8d41cdee765

The function convertTwoTypeAs generates an instantiation of bytesToWords that contains a "fromSLV" expression in the port map, which modelsim's VHDL compiler considers illegal.

**Clash**
```
 {-# NOINLINE bytesToWords #-}                                                  
 bytesToWords :: Bytes 4 -> TypeB                                               
 bytesToWords = TypeB . fmap (\(Byte a) -> Word $ ((unpack . resize .  pack) a))

 data TypeAs = TypeAS TypeA | Nop

 {-# NOINLINE convertTwoTypeAs #-}            
 convertTwoTypeAs :: TypeAs -> Maybe TypeB    
 convertTwoTypeAs op = case op of             
   TypeAS (TypeA a) -> Just $ (bytesToWords a)
   Nop        -> Nothing                      
```
**VHDL**
```
 bytestowords_capp_arg : entity bytestowords                                                     
   port map                                                                                      
     ( result  => \c$app_arg\                                                                    
     , \c$arg\ => test_types.array_of_std_logic_vector_8'(test_types.fromSLV(op(31 downto 0))) );
```

Modelsim Error:
```
** Error: ../../../../../repos/haskell/vhdl/Test/converttwotypeas.vhdl(22): (vcom-1436) Actual expression (function call "fromSLV") of formal "\c$arg\" is not globally static.
** Note: ../../../../../repos/haskell/vhdl/Test/converttwotypeas.vhdl(32): VHDL Compiler exiting
```

If I remove "Nop" from the type "TypeAs", clash compiles without "fromSLV" in the port map.

martijnbastiaan, 2020-06-19:
Looks like the function generating the port map should use the equivalent of `~VAR`. Alternatively, the hdl backends should make sure the RHS of a port map is "simple enough".

christiaanb, 2020-06-19:
https://github.com/clash-lang/clash-compiler/blob/460f0a172aeb9c032db63977d500111bd68be54e/clash-lib/src/Clash/Netlist.hs#L693 should check whether the `Modifier` is "simple" enough. It seems that most projections out of an SOP type are not simple.

xavieran, 2020-06-19:
Quick update on my end, as a work around it looks like I can write this as
```
{-# NOINLINE convertTwoTypeAs #-}            
convertTwoTypeAs :: TypeAs -> Maybe TypeB    
convertTwoTypeAs op = case op of             
 TypeAS a -> Just $ (bytesToWords (getTypeA a))
 Nop        -> Nothing
  where
    getTypeA (TypeA a) = a
```

and that correctly places the fromSLV outside of the portMap.

christiaanb, 2020-06-19:
Fixed by #1389

closed 2020-06-19 by christiaanb