clash-lang/clash-compiler #1392: Monomorphism restriction vs. `HiddenClockResetEnable`

gergoerdi, 2020-06-20:
Starting at f8f1d2b6162ae27d0191c7fdbc096e88e6f2662c, the following code doesn't work anymore:

```
import Clash.Prelude

topEntity
    :: "CLK_25MHZ" ::: Clock System
    -> "RESET"     ::: Reset System
    -> "FOO"       ::: Signal System Bool
topEntity clk rst = withClockResetEnable clk rst enableGen vga
  where
    vga = video $ pure False

video :: (HiddenClockResetEnable System) => Signal System Bool -> Signal System Bool
video = id
```

It fails with the following error message at f8f1d2b6162ae27d0191c7fdbc096e88e6f2662c (and works before that):

```
/tmp/Test2.hs:9:11: error:
    • Unbound implicit parameters
        (GHC.Classes.IP
           (Clash.Signal.HiddenClockName System) (Clock System),
         GHC.Classes.IP
           (Clash.Signal.HiddenEnableName System) (Enable System),
         GHC.Classes.IP
           (Clash.Signal.HiddenResetName System) (Reset System))
        arising from a use of ‘video’
    • In the expression: video $ pure False
      In an equation for ‘vga’: vga = video $ pure False
      In an equation for ‘topEntity’:
          topEntity clk rst
            = withClockResetEnable clk rst enableGen vga
            where
                vga = video $ pure False
  |
9 |     vga = video $ pure False
  |           ^^^^^^^^^^^^^^^^^^
```

This is very annoying, because it breaks the following pattern, where circuitry is described as a set of mutually recursive `Signal` definitions, with the input signals in scope for them:

```
topEntity
    :: "CLK_25MHZ" ::: Clock Dom25
    -> "RESET" ::: Reset Dom25
    -> "INPUTS ::: Signal Dom25 (Vec 4 (Active Low))
    -> "OUTPUTS" ::: Signal Dom25 (Vec 4 (Active Low))
topEntity = withEnableGen board
  where
    board inps = outs
      where
        -- Mutually recursive definitions of signals using `inps`
```

christiaanb, 2020-06-20:
Yes, there is no win, sadly. You could use the following instead:
```haskell
topEntity
    :: "CLK_25MHZ" ::: Clock System
    -> "RESET"     ::: Reset System
    -> "FOO"       ::: Signal System Bool
topEntity clk rst = withClockResetEnable clk rst enableGen $
  let
    vga = video $ pure False
  in
    vga
```

Or, as a variation on the second example you gave:
```haskell
topEntity
    :: "CLK_25MHZ" ::: Clock System
    -> "RESET"     ::: Reset System
    -> "FOO"       ::: Signal System Bool
topEntity clk rst = withClockResetEnable clk rst enableGen $
  let
    vga inps = videoOut
      where
        videoOut = video inps
  in
    vga (pure False)
```

gergoerdi, 2020-06-20:
The horrible-ness goes further with the warnings too. The following code triggers a monomorphism restriction warning:

```
import Clash.Prelude

topEntity
    :: "CLK_25MHZ" ::: Clock System
    -> "RESET"     ::: Reset System
    -> "FOO"       ::: Signal System Bool
topEntity clk rst = withClockResetEnable clk rst enableGen $
    let bar = riseEvery (SNat @1000)
        foo = regEn True bar $ complement <$> foo
    in foo
```

To silence the warning, I am required to write this as

```
bar = riseEvery @System (SNat @1000)
```

I can't even imagine the full extent of the pain this is going to cause me...

gergoerdi, 2020-06-20:
Here's a really tiny circuit that I migrated to Clash with the monomorphism restriction... Look at all these changes, annotating all overloaded literals even though the signals they are used in are all used monomorphically... 

https://github.com/gergoerdi/clash-framebuffer/commit/a1e0d277e058970e13ece3c4822e340fc0d72e47

christiaanb, 2020-06-20:
We could disable the warning.

christiaanb, 2020-06-20:
Unless you’re saying you got type errors instead of warnings

gergoerdi, 2020-06-21:
> Unless you’re saying you got type errors instead of warnings

No, these are just warnings. Except, of course, for the original one in this ticket that breaks the `where` pattern.

christiaanb, 2020-06-22:
@gergoerdi HEAD of `master` now has the warning disabled

closed 2020-06-22 by christiaanb