clash-lang/clash-compiler #1393: Clarify License for example project

mb64, 2020-06-20:
At the moment, the example project has an [example LICENSE file](https://github.com/clash-lang/clash-compiler/blob/master/clash-starters/clash-example-project/LICENSE) which is likely not intended to apply to the example project itself.  It would be good to specify how the example project is licensed.

However, I don't know if the BSD-style license that applies to the Clash compiler is the right license for the sample project, either.  Personally, I would like to copy the list of extensions and plugins from the example project's `.cabal` file into my own `.cabal` file.  Do you want to legally require me to include your LICENSE file and copyright statement in my project, and to state that parts of my project are derived from code the Clash compiler?

If so, the same BSD-style license as the compiler would be fine.  If not, I recommend CC0 or some other public domain license.  I believe that making the example project CC0 would provide a significantly lower barrier of entry to trying out Clash.  (Either way, I expect you'll get enough attribution – personally, I know I'll link to Clash either way.)

martijnbastiaan, 2020-06-21:
You're right. That was not my intention when I included that LICENSE file. CC0 seems like the proper license for the starter project. We should add a warning that makes people change the licence for their own projects though, as it's unlikely they'd want their project to be CC0 too.

I'm taking some time off of work, so it'd be nice if someone else could pick this up. @christiaanb Could you replace the license with CC0 and replace the current `starter_project.zip` for Clash 1.2.2?

closed 2020-06-21 by christiaanb

christiaanb, 2020-06-21:
@mb64 The `starter_project.zip` bundled with the 1.2.2 release now has the version dedicated to the public domain.

PR #1394 (2020-06-21): Dedicate the "Clash Example Project" to the public domain
Fixes #1393
commit 30af2f65 (2020-06-21): Dedicate the "Clash Example Project" to the public domain

Fixes #1393