clash-lang/clash-compiler #1401: Prelude.^ fails on Bit, BitVector

leonschoorl, 2020-06-23:
```
Clash.Prelude> import qualified Prelude as P

Clash.Prelude P> 42  P.^  (0 :: Int)
1
Clash.Prelude P> 42  P.^  (1 :: Int)
42

Clash.Prelude P> 42  P.^  (0 :: Unsigned 1)
1
Clash.Prelude P> 42  P.^  (1 :: Unsigned 1)
*** Exception: divide by zero

Clash.Prelude P> 42  P.^  (0 :: BitVector 1)
1
Clash.Prelude P> 42  P.^  (1 :: BitVector 1)
*** Exception: divide by zero

Clash.Prelude P> 42  P.^  (0 :: Bit)
1
Clash.Prelude P> 42  P.^  (1 :: Bit)
^CInterrupted.
```

leonschoorl, 2020-06-23:
The `divide by zero` is because [implementation of Prelude.^] uses `Prelude.even` and that doesn't work for types that can't represent the value 2, see https://github.com/clash-lang/clash-compiler/issues/959

This hang with Bit is  because Bit's Integral instance is a hardcoded and doesn't throw `divide by zero` exceptions like the other instances do: https://github.com/clash-lang/clash-compiler/blob/d547dee8a33e2a3cddd7772bbf0df8e587f45d27/clash-prelude/src/Clash/Sized/Internal/BitVector.hs#L297-L303

[implementation of Prelude.^]: https://gitlab.haskell.org/ghc/ghc/-/blob/d4a0be758003f32b9d9d89cfd14b9839ac002f4d/libraries/base/GHC/Real.hs#L625

leonschoorl, 2020-06-25:
Ideally for us we would put the `| y == 1 = x` first as the first case in [f in base's implementation of `(^)`](https://gitlab.haskell.org/ghc/ghc/-/blob/a1f34d37b47826e86343e368a5c00f1a4b1f2bce/libraries/base/GHC/Real.hs#L630)

We should try [these benchmarks](https://gitlab.haskell.org/ghc/ghc/-/issues/1687) to see if the order of the cases of this `f` are just an implementation detail or important for performance.

dramforever, 2020-07-10:
Just to throw in an additional note, `negate` for `Bit` looks suspicious: It's `complement` rather than `id`

https://github.com/clash-lang/clash-compiler/blob/d547dee8a33e2a3cddd7772bbf0df8e587f45d27/clash-prelude/src/Clash/Sized/Internal/BitVector.hs#L281-L288

Is this intended? It isn't negation modulo 2 (which is `id`, which in turn justifies `(+) = (-)`), so it doesn't even match with `BitVector 1`. And also now `0 - x` isn't `- x`.

```
ghci> negate 1 :: BitVector 1
1
ghci> negate 0 :: BitVector 1
0
ghci> negate 1 :: Bit
0
ghci> negate 0 :: Bit
1
``

Maybe whoever wrote that thought `(- x)` might be a handy shortcut, and to be honest, negating a (non-0) unsigned value would cause an overflow in the first place, so it probably wouldn't have worked in generic code anyway. But I thought I should bring it up.