clash-lang/clash-compiler #1402: Clash 1.2.2 Produces Invalid VHDL Output

paddytheplaster, 2020-06-24:
Hello,

I'm getting a bug for a rather large project.

The following is the entire code for the top entity:

    {-# ANN topEntity
        (Synthesize
        { t_name    = "paddy"
        , t_inputs  = [ PortName "CLOCK_50"
                      , PortName "KEY0"
                      , PortName "RX"
                      ]
        , t_output  = PortProduct "" [ PortName "TX"
                                     , PortName "LED"
                                     ]
        }
      ) #-}
    topEntity
      :: Clock  Dom50                    -- clock
      -> Signal Dom50 Bool               -- reset
      -> Signal Dom50 Bit                -- raw RX input
      -> Signal Dom50 (Bit,BitVector 8)
    topEntity clock50 reset_button rx_raw
      = exposeClockResetEnable (top rx_raw) clock50 rst en
      where en = enableGen
            rst = unsafeFromLowPolarity reset_button

    top
      :: HiddenClockResetEnable dom
      => Signal dom Bit
      -> Signal dom (Bit,BitVector 8)
    top rx_raw
      = (,) <$> tx_raw <*> leds
      where
        ss = mealy (\s (mb_bit,acc) -> (paddy_transition s mb_bit acc,s)) initial_paddy ps
        uart_out = uart data_bits ticks_per_bit rx_raw tx_tent
        ps = (,) <$> rx_out <*> tx_acc
        leds = register (0 :: BitVector 8) $ fmap test_output ss
        tx_tent = fmap tx_out ss
        rx_out = fmap fst uart_out
        tx_raw = fmap (fst.snd) uart_out
        tx_acc = fmap (snd.snd) uart_out

The generated/relevant vhdl is the following. The 'mb_bit' in the previous function is the only line of code in my entire project which uses the name 'mb_bit'.

    with (mb_bit(1 downto 1)) select
      \c$tupIn\ <= ( tup4_1_sel0_boolean => false
                   , tup4_1_sel1_std_logic => '0'
                   , tup4_1_sel2_unsigned_0 => signed'(0 to 63 => '-')
                   , tup4_1_sel3_unsigned_1 => resize(unsigned(stuff1),64) ) when "0",
                   ( tup4_1_sel0_boolean => true
                   , tup4_1_sel1_std_logic => x_1
                   , tup4_1_sel2_unsigned_0 => signed'(0 to 63 => '-')
                   , tup4_1_sel3_unsigned_1 => resize(unsigned(stuff1),64) ) when others;

The following is the vhdl compiler error. Line 4023 refers to the 4th line of the previous snippet.

Error (10511): VHDL Qualified Expression error at paddy.vhdl(4023): SIGNED type specified in Qualified Expression must match UNSIGNED type that is implied for expression by context File: /home/paddy/Altera/Clash/vhdl/TopEntity/paddy/paddy.vhdl Line: 4023
Error (10511): VHDL Qualified Expression error at paddy.vhdl(4027): SIGNED type specified in Qualified Expression must match UNSIGNED type that is implied for expression by context File: /home/paddy/Altera/Clash/vhdl/TopEntity/paddy/paddy.vhdl Line: 4027
Error: Quartus Prime Analysis & Synthesis was unsuccessful. 2 errors, 0 warnings
    Error: Peak virtual memory: 966 megabytes
    Error: Processing ended: Wed Jun 24 06:21:43 2020
    Error: Elapsed time: 00:00:09
    Error: Total CPU time (on all processors): 00:00:24

I'll try to narrow this down a bit further but this may take a while.

In the mean time I'm happy to accept any suggestions about how to avoid the error.

Regards,

Paddy

christiaanb, 2020-06-24:
It's probably somewhere inside `paddy_transition`, if you're at liberty to share it, we might be able to infer from that what's going on without having to create a minimal test case.

paddytheplaster, 2020-06-24:
A minor update.

The signal of 'Maybe' values is a signal of 'Maybe Bit' values, which is changed to a '(Bool,Bit)' pair as follows:

    (req,b) = maybe (False,low) (\b -> (True,b)) mb_bit

If I change this to the following, the error disappears.

    (req,b) = (False,low)

Regards,

Paddy

paddytheplaster, 2020-06-24:
Apologies for accidentally closing the issue....

christiaanb, 2020-06-24:
Right, I would then check which code branches on the value of `req`.

paddytheplaster, 2020-06-24:
Hi again,

Doing just that (checking the code which branches on the value of 'req').

FYI At some stage, clash synthesis to VHDL went fine but I got the following:

    Clash.Backend.VHDL(2195): WARNING: rendering bitvector mask as dontcare
    Clash.Backend.VHDL(2195): WARNING: rendering bitvector mask as dontcare
    Clash.Backend.VHDL(2188): WARNING: result smaller than argument
    Clash.Backend.VHDL(2188): WARNING: result smaller than argument

FWIW The first one is interesting because the signal we're investigating is used to control the change of a 'BitVector'. The second may be worrying?

I am making so many changes to simplify things that may not be able to reproduce this.

Regards,

Paddy

christiaanb, 2020-06-24:
Are you by any change using `fromIntegral` anywhere? Because those warnings indicate one of two things:
1. GHCs Worker/Wrapper transformation kicked in for BitVector, which we don't see happening often
2. You are using functions like `Clash.Sized.BitVector.Internal.unsafeToNatural`

christiaanb, 2020-06-24:
Thanks for those hints wrt the WARNINGs @paddytheplaster, I think I found the bug.

christiaanb, 2020-06-24:
So I'm able to get something that triggers the `result smaller than argument` WARNING by:
```haskell
{-# OPTIONS_GHC -O #-}
module Test where

import Clash.Prelude

topEntity :: BitVector 128 -> BitVector 128
topEntity = f

f :: BitVector 128 -> BitVector 128
f x = case x of
  90 -> x
  _ -> 8
```
the warning about the "rendering bitvector mask as dontcare" is a bit more difficult so far.

christiaanb, 2020-06-24:
Alright, I have a reproducer:
```haskell
{-# OPTIONS_GHC -O #-}
module Test where

import Clash.Prelude

topEntity :: Maybe (BitVector 128) -> (BitVector 128,Bool)
topEntity q = case q of
  Just b -> f True b
  _      -> f False 3

f :: Bool -> BitVector 128 ->  (BitVector 128, Bool)
f t x = case x of
  90 -> (x,t)
  _ -> (8,t)
```

christiaanb, 2020-06-24:
@paddytheplaster Are you running Clash with the `-O` or `-O2` flag by any chance?

paddytheplaster, 2020-06-24:
Hi again,

@christiaanb Yes, there are some 'fromIntegral' calls. I'm not sure if they're relevant. I need more time to anwer this.

@christiaanb Yes, I'm using optimisation. The following are (some of) the flags: '-O3 -fclash-no-cache '. The error doesn't go away if I remove 'O3'.

@christiaanb I noticed your minimal example which reproduces a similar error.

FYI In my case the "culprit signal" just transforms the bitvector for the next state of the bitvector by shifting in a bit. (This is done using 'split#' and '++#' because there's no real support for things like this.) The next state of the bitvector isn't used until the next clock cycle. The signal isn't used anywhere, except for changing the state. I can't simplify much more???

Regards,

Paddy

christiaanb, 2020-06-24:
@paddytheplaster Can you add this flag: `-fno-worker-wrapper` and check whether the error goes away?

paddytheplaster, 2020-06-24:
Hi again,

@chritiaanb Error doesn't go away when I use '-fno-worker-wrapper'.

Regards,

Paddy

christiaanb, 2020-06-24:
Have you packaged up any of your Clash code into another cabal package that you are now importing for this project?

paddytheplaster, 2020-06-24:
No, I'm using clash in a nixshell environment.

Regards,

Paddy

paddytheplaster, 2020-06-24:
@christiaanb I don't understand your comment.

christiaanb, 2020-06-24:
Well, the error refers to the lines with this vhdl code:
```vhdl
, tup4_1_sel2_unsigned_0 => signed'(0 to 63 => '-')
```
With the patch, that will now be rendered as:
```vhdl
, tup4_1_sel2_unsigned_0 => unsigned'(0 to 63 => '-')
```

paddytheplaster, 2020-06-24:
@christiaanb  I didn't realise your comment included a hyperlink. Will follow link and will revert.

paddytheplaster, 2020-06-25:
@christiaanb As already mentioned in the Clash user group, the patch fixes the bug.

About the warning messages, these aren't always easy to notice. It may be an idea to highlight them (this could be flag-controlled: on or off).

Regards,

Paddy

paddytheplaster, 2020-06-25:
Thanks.

P.

PR #1403 (2020-06-25): Use `unsigned` typemark when rendering the mask of a BitVector
Fixes #1402
commit e6b8b478 (2020-06-25): Use unsigned typemark when rendering BV mask

It was being rendered with a signed typemark, a leftover from
when the underlying representation of BV were two Integers
instead of two Naturals.

Fixes #1402
commit 79d87057 (2020-06-25): Use unsigned typemark when rendering BV mask

It was being rendered with a signed typemark, a leftover from
when the underlying representation of BV were two Integers
instead of two Naturals.

Fixes #1402

(cherry picked from commit e6b8b4784ebf1c224f8ba5e6e9c386e786849e2f)
closed 2020-06-25 by christiaanb