clash-lang/clash-compiler #1436: Multiple PrimitiveGuard annotations makes Clash yield no output HDL

JvWesterveld, 2020-07-09:
It seems that compilation of any Clash function with (accidentally) multiple PrimitiveGuard annotations does not yield any HDL as output. This may, for example, easily happen if one does not know that warnAlways implies hasBlackBox:
```
import Clash.Prelude
import Clash.Annotations.Primitive
import Data.String.Interpolate (i)
import Data.String.Interpolate.Util (unindent)

test :: Bool
test = True
{-# NOINLINE test #-}
{-# ANN test hasBlackBox #-}
{-# ANN test (warnAlways "You should know that ...") #-}
{-# ANN test (InlinePrimitive [VHDL] $ unindent [i|
  [ { "BlackBox" :
      { "name" : "VoidTest.test"
      , "kind" : "Expression"
      , "template" : "true"
      }
    }
  ]
 |]) #-}

topEntity :: Bool
topEntity = test
```

This is a user error, but I think the compiler could warn better about this. Currently it only prints this message, not making clear the consequences:
```The following value has multiple 'PrimitiveGuard' annotations: test```

Tested on Clash 1.2.2 using GHC 8.6.5.

JvWesterveld, 2020-08-24:
Somehow the `Panic.pgmError` calls in `LoadModules.hs`, such as [the one relevant here](https://github.com/clash-lang/clash-compiler/blob/ccebebb4fb2ccf70098b6b75a717baf1775c3a3e/clash-ghc/src-ghc/Clash/GHC/LoadModules.hs#L593-L595), don't have "`*** Exception: `" prepended when encountered during Clash compilation.

leonschoorl, 2020-08-24:
`*** Exception: ` comes from the default exception handler inside of [clashi](https://github.com/clash-lang/clash-compiler/blob/ccebebb4fb2ccf70098b6b75a717baf1775c3a3e/clash-ghc/src-bin-8.10/Clash/GHCi/UI.hs#L4207)/ghci
It's not part of the actual error.
And if you run `clash --SOME-HDL ..`, you'll get an slightly different error.

commit bcbad59f (2023-08-16): Allow values to have multiple PrimitiveGuard annotations

Fixes #1436
commit 43e0e49f (2023-08-16): Allow values to have multiple PrimitiveGuard annotations

Fixes #1436
commit a713dfba (2023-08-17): Allow values to have multiple PrimitiveGuard annotations

Fixes #1436
commit 6ceb6dd5 (2023-08-17): Allow binders to have multiple PrimitiveGuard annotations

Fixes #1436
closed 2023-08-18 by leonschoorl