clash-lang/clash-compiler #1439: More Than One Use Clause Imports

paddytheplaster, 2020-07-10:
Hi again,

When I compile the following with Clash 1.3.0 the generated VHDL causes a VHDL compilation error.

    rotate_right
        :: forall n . (KnownNat n, 1 <= n)
        => Bool
        -- ^ Client request
        -> BitVector n
        -- ^ Object of shift operation
        -> BitVector n
        -- ^ Result.
    rotate_right bv
        = leToPlus @1 @n (rotate_right' bv)
    {-# NOINLINE rotate_right #-}

    rotate_right'
        :: forall n . (KnownNat n)
        => Bool
        -- ^ Client request
        -> BitVector (n+1)
        -- ^ Object of shift in operation
        -> BitVector (n+1)
        -- ^ Result.
    rotate_right' bool bv
        | bool      = pack b ++# (fst $ split# bv)
        | otherwise = bv
        where b = lsb bv
    {-# NOINLINE rotate_right' #-}

The compilation error is as follows:

    Error (10621): VHDL Use Clause error at rotate_right_0.vhdl(20): more than one Use Clause imports a declaration of simple name "rotate_right" -- none of the declarations are directly visible File: /home/paddy/Altera/Clash/vhdl/TopEntity/circuit/rotate_right_0.vhdl Line: 20
    Error (10784): HDL error at numeric_std_vhdl1993.vhd(584): see declaration for object "ROTATE_RIGHT" File: /opt/altera/quartus/libraries/vhdl/ieee/1993/numeric_std_vhdl1993.vhd Line: 584
    Error (10609): VHDL error at numeric_std_vhdl1993.vhd(595): another match for "rotate_right" File: /opt/altera/quartus/libraries/vhdl/ieee/1993/numeric_std_vhdl1993.vhd Line: 595
    Error (10609): VHDL error at rotate_right.vhdl(10): another match for "rotate_right" File: /home/paddy/Altera/Clash/vhdl/TopEntity/circuit/rotate_right.vhdl Line: 10
    Error (10476): VHDL error at rotate_right_0.vhdl(20): type of identifier "ROTATE_RIGHT" does not agree with its usage as "entity" type File: /home/paddy/Altera/Clash/vhdl/TopEntity/circuit/rotate_right_0.vhdl Line: 20
    Error (10467): VHDL Component Configuration or Component Instantiation Statement error at rotate_right_0.vhdl(20): component "ROTATE_RIGHT" is used but not declared File: /home/paddy/Altera/Clash/vhdl/TopEntity/circuit/rotate_right_0.vhdl Line: 20
    Error: Quartus Prime Analysis & Synthesis was unsuccessful. 6 errors, 0 warnings
        Error: Peak virtual memory: 955 megabytes
        Error: Processing ended: Fri Jul 10 10:18:03 2020
        Error: Elapsed time: 00:00:11
        Error: Total CPU time (on all processors): 00:00:23

The following is the relevant generated VHDL. The second line of the third block is Line 20.

    entity rotate_right_0 is
      port(eta2    : in boolean;
           \c$arg\ : in std_logic_vector(31 downto 0);
           result  : out std_logic_vector(31 downto 0));
    end;

    architecture structural of rotate_right_0 is

    begin
      rotate_right_result : entity rotate_right
        port map
          ( result => result
          , bool   => eta2
          , bv     => \c$arg\ );
    end;

Regards,

Paddy

paddytheplaster, 2020-07-10:
I just noticed the Haskell should have been rejected because it doesn't take the definition of `rotate_right` should have been as follows:

    rotate_right bool bv  = leToPlus @1 @n (rotate_right' bool bv)

Regards,

Paddy

paddytheplaster, 2020-07-10:
When I turn the global definition of `shift_right` into a local definition in a `where` clause, the error disappears.

Regards,

Paddy

alex-mckenna, 2020-07-10:
VHDL defines `rotate_right` in `numeric_std`, I suspect that may be part of the problem:

```vhdl
-- Id: S.6
function ROTATE_RIGHT (ARG: UNSIGNED; COUNT: NATURAL) return UNSIGNED;
-- Result subtype: UNSIGNED(ARG'LENGTH-1 downto 0)
-- Result: Performs a rotate-right of an UNSIGNED vector COUNT times.

-- Id: S.8
function ROTATE_RIGHT (ARG: SIGNED; COUNT: NATURAL) return SIGNED;
-- Result subtype: SIGNED(ARG'LENGTH-1 downto 0)
-- Result: Performs a logical rotate-right of a SIGNED
--         vector COUNT times.
```

paddytheplaster, 2020-07-10:
@alex-mckenna Thanks. More than likely the cause.
Regards,

Paddy

alex-mckenna, 2020-07-10:
Just checked on my machine, changing `rotate_right` to `rot_r` in the example you gave fixed it for me.
Maybe it would be worthwhile to have a clash warning if a component name clashes with something from a standard module, i.e.

```
IEEE.STD_LOGIC_1164.ALL;
IEEE.NUMERIC_STD.ALL;
IEEE.MATH_REAL.ALL;
std.textio.all;
```

for VHDL. What do you think @christiaanb / @martijnbastiaan / @leonschoorl ?

christiaanb, 2020-07-10:
Clash should not generate those names, it should be added to the reserved words list: https://github.com/clash-lang/clash-compiler/blob/d009ce0b48af9af08d314a291636bc0559514e3f/clash-lib/src/Clash/Backend/VHDL.hs#L234-L250

alex-mckenna, 2020-07-10:
Right, I'll put together a patch

alex-mckenna, 2020-07-10:
Hi @paddytheplaster, this should be fixed on `master` and the `1.2` branch now. Can you check and let us know?

paddytheplaster, 2020-07-11:
@alex-mckenna Thanks Alex. I can confirm it works now.

Regards,

Paddy

closed 2020-07-11 by alex-mckenna