clash-lang/clash-compiler #1469: `normalizeType` doesn't always normalize type

alex-mckenna, 2020-08-07:
Working on the partial evaluator, I found the following case where `normalizeType` doesn't seem to work as advertised.
Attempting to normalize
```haskell
BitSize (Maybe (Vec 2 (Maybe Bool)))
```
gives
```haskell
CLog 2 (GConstructorCount (Rep (Maybe (Vec 2 (Maybe Bool))))) + GFieldSize (Rep (Maybe (Vec 2 (Maybe Bool))))
```
but using `normalizeType tcm (normalizeType tcm ty)` gives
```haskell
LitTy (NumTy 5)
```
which is what was initially expected.

martijnbastiaan, 2020-08-07:
An (naive?) implementation could change

https://github.com/clash-lang/clash-compiler/blob/5477a2fe8b2944f31c8645211a712bf89a2d957f/clash-lib/src/Clash/Core/Type.hs#L648-L650

to

```haskell
             in case reduceTypeFamily tcMap ty' of
               Just ty'' -> go ty''
               Nothing  -> ty'
```

right?

alex-mckenna, 2020-08-07:
This appears to solve the issue, and with what must be one of the shortest diffs known to this project.
I give it :star: :star: :star: :star: :star:

martijnbastiaan, 2020-08-07:
This still needs a proper fix described in the TODO.

PR #1470 (2020-08-07): Fully normalize type families in `normalizeType`.
Normalize type did not recurse on the result of reducing a type
family, meaning if a type family reduced to another type family it
would not be normalized.

Closes #1469