clash-lang/clash-compiler #1480: Add more constraints on `RankNTypes` function args

DigitalBrains1, 2020-08-13:
We don't always specify all constraints that could be specified on `Nat` in our functions. For instance, `riseEvery` requires an `SNat` larger than zero, but it also accepts zero, which causes a runtime error. Usually it's not a problem because we can pass a more-constrained argument to a less-constrained function.

But when we're talking about the argument of a higher-order function (using higher-rank type), the reverse is true. We have
```
smap :: forall k a b. KnownNat k => (forall l. SNat l -> a -> b) -> Vec k a -> Vec k b
```
(same for `dfold`, `dtfold`, `vfold`). This precludes using functions that restrict `l` (ell) to be smaller than `k`, like `at`, `take`, `drop`, `select`, `splitAt`, etc. These all require `l` be smaller than `k`, and GHC/CλaSH can't prove that this is the case.

This lead to [this](https://groups.google.com/d/topic/clash-language/OSpK5WINytg/discussion) Google Group thread.

We should add constraints on these function arguments to allow using constrained functions.

leonschoorl, 2024-10-23:
Same idea as #1054

closed 2024-10-23 by leonschoorl

DigitalBrains1, 2024-10-24:
This issue mentions `smap`, `dfold`, `dtfold` and `vfold`. #1054 only mentions `dfold` and `dtfold`. I'm worried we might forget about the other ones if you close this one in favour of #1054.