clash-lang/clash-compiler #1486: history.dat keeps growing

leonschoorl, 2020-08-24:
When you enable the `history` flag on `clash-lib` clash will record the rewrite history in `history.dat`, for analysis with the `clash-term` utility.  
But clash will just append data to the end of that file, so the rewrites of the current clash run will just be appended to the end of the rewrites of the previous clash run.

This can greatly confuse and/or slowdown `clash-term`.

Workaround: delete `history.dat` between clash runs  
Fix would be just to put a `removeFile` at an appropriate point in the startup code.

martijnbastiaan, 2020-08-24:
A slightly more sophisticated way would be to save the histories in `histories/$(date --iso-8601=seconds).dat`. This way could compare various runs. Maybe even with a flag to influence naming?

leonschoorl, 2020-08-24:
Or maybe just `-fclash-history FILENAME`?
Then the user can use any shell magic to set that filename if they want.
Since the rewrite history can be fairly big, I would be against keeping more then 2 around by default.

commit e16e9012 (2021-02-23): Don't keep appending to history.dat

Start a new history.dat so it'll only contain the transformations of one run of clash.

Fixes #1486
closed 2021-02-23 by leonschoorl