clash-lang/clash-compiler #1493: Incompatibility of Lengths of Vecs on LHS and RHS of Defining Clauses

paddytheplaster, 2020-09-01:
Hi again,

[Clash, version 1.3.0 (using clash-lib, version: 1.3.0)]

Maybe I am overlooking something but the following is odd to say the least. The lengths of the 'Vec's which are pattern matched in the first definition can never be equal and neither can the lengths of the results in the second function.

     module Bug where

     import Clash.Prelude

     bug :: Vec n () -> Bool
     bug b = case b of
             (() `Cons` _) -> True
             Nil           -> True

     bug' :: KnownNat n => Vec n () -> Vec n Bool
     bug' (() `Cons` _) = True `Cons` repeat True
     bug' Nil           = Nil

Regards,

Paddy

rowanG077, 2020-09-01:
I'm not sure what you mean. Why is this a bug? The vector could be empty or it could contain something which is what the pattern match is checking.

paddytheplaster, 2020-09-02:
Hi Rowan,

I realise I mad a mistake and that I've seen (many) similar definitions before. If the system allows, I'll remove this thread tomorrow. That should give you enough time to read my reply.

Regards,

Paddy

martijnbastiaan, 2020-09-02:
I'll just close the issue, thanks for the concern @paddytheplaster !

closed 2020-09-02 by martijnbastiaan