clash-lang/clash-compiler #1494: Blackbox template function with BiSignals error

hcab14, 2020-09-02:
minimal example triggering the error
https://gist.github.com/hcab14/4b6d03a1e69a4950a850b40bdbdccd62

`TopWorking.hs` uses `Signal`s (no error and correct HDL code generated) while `TopNotWorking` is using `BiSignalIn`,
`BiSignalOut` and triggers the following error:
```
cabal new-run clash -- -i. --verilog TopNotWorking.hs
```
```
<no location info>: error:
    Clash error call:
    Clash.Backend.Verilog(1126): DataCon Bit (DC (Bit,0)) [Identifier "dIn" Nothing]
    CallStack (from HasCallStack):
      error, called at src/Clash/Backend/Verilog.hs:1125:13 in clash-lib-1.3.0-inplace:Clash.Backend.Verilog
```

I have put some `Debug.Trace` statements into Clash but could not figure out why this error is triggered.

https://github.com/clash-lang/clash-compiler/blob/7e45191ece8853154eb361898bc03e06c28c8ed0/clash-lib/src/Clash/Backend/Verilog.hs#L1125

christiaanb, 2020-09-02:
This is happening because `BiSignalOut` is filtered out, i.e. a tuple `(BiSignalOut, Bit)` becomes `Bit`.

I haven't tested it, but I think if you change https://gist.github.com/hcab14/4b6d03a1e69a4950a850b40bdbdccd62#file-primtest-hs-L57 to:

```haskell
, Assignment result (Identifier dIn Nothing)
```

It will work, and you get the Verilog you wanted.

hcab14, 2020-09-02:
Hi @christiaanb 

thanks a lot! It does work as you suggested.

Closing

closed 2020-09-02 by hcab14