clash-lang/clash-compiler #1500: Presence of unused variable definition affects compilation

JvWesterveld, 2020-09-03:
For example:
```
a
  :: HiddenClockResetEnable System
  => Signal System (Signed 3)
  -> Signal System Bool
a = register False . fmap (== 0)

system
  :: HiddenClockResetEnable System
  => Signal System Bool
system = a (pure 1)

topEntity
  :: Clock System
  -> Reset System
  -> Enable System
  -> Signal System Bool
topEntity = exposeClockResetEnable system

something = exposeClockResetEnable system clockGen resetGen enableGen
```
This compiles to VHDL as you would expect when leaving out the last line. With the last line included, the generated VHDL is significantly different and even includes `clockGen` and `resetGen`:
```
-- Automatically generated VHDL-93
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;
use IEEE.MATH_REAL.ALL;
use std.textio.all;
use work.all;
use work.test_types.all;

entity topentity is
  port(-- clock
       \c$arg\   : in test_types.clk_system;
       -- reset
       \c$arg_0\ : in test_types.rst_system;
       \c$arg_1\ : in boolean;
       result    : out boolean);
end;

architecture structural of topentity is
  signal result_0      : std_logic_vector(1 downto 0);
  signal g1            : test_types.index_2;
  signal a1            : test_types.maybe;
  signal \c$bindCsr\   : test_types.clk_system;
  signal \c$bindCsr_0\ : test_types.rst_system;
  signal result_1      : boolean;

begin
  result <= (result_1);

  with (a1(1 downto 1)) select
    result_0 <= std_logic_vector'("0-") when "0",
                std_logic_vector'(std_logic_vector'(std_logic_vector'("1")) & std_logic_vector'((std_logic_vector'((std_logic_vector(g1)))))) when others;

  g1 <= test_types.index_2'(test_types.fromSLV(a1(0 downto 0)));

  a1 <= std_logic_vector'("1" & (std_logic_vector(to_unsigned(0,1)))) when result_1 else
        std_logic_vector'("0" & "-");

  -- clockGen begin
  -- pragma translate_off
  clkgen : process is
    constant half_periodh : time := 10000000 fs / 2;
    constant half_periodl : time := 10000000 fs - half_periodh;
  begin
    \c$bindCsr\ <= '0';
    wait for 3000 ps;
    loop
      \c$bindCsr\ <= not \c$bindCsr\;
      wait for half_periodh;
      \c$bindCsr\ <= not \c$bindCsr\;
      wait for half_periodl;
    end loop;
    wait;
  end process;
  -- pragma translate_on
  -- clockGen end

  -- resetGen begin
  -- pragma translate_off
  \c$bindCsr_0\ <=  '1' ,
            '0'  after (3001 ps + integer'(1 - 1) * 10000 ps);
  -- pragma translate_on
  -- resetGen end

  somefunction_a_result_1 : entity a
    port map
      ( result        => result_1
      , \c$bindCsr\   => \c$bindCsr\
      , \c$bindCsr_0\ => \c$bindCsr_0\ );
end;
```
Using Clash 1.2.4 and current git master with GHC 8.6.5.

leonschoorl, 2020-09-03:
This looks similar to #399

leonschoorl, 2020-09-03:
Simplified test case:
```haskell
{-# LANGUAGE ImplicitParams #-}
{-# LANGUAGE ConstraintKinds,FlexibleContexts #-}

module Test where
import Prelude

type Hidden a =
  ( ?enable :: a
  , Eq a  -- removing this "fixes" the issue
  )

{-# NOINLINE a #-}
a
  :: Hidden Bool
  -- => Int -- OK
  => Integer -- problem
  -> Bool
a _ = ?enable

system
  :: Hidden Bool
  => Bool

system = a 0

topEntity
  :: Bool
  -> Bool
topEntity ena =
  let ?enable = ena
   in system

someFunction =
  let ?enable = True
   in system
```

Which when compiled with: `ghc-8.10.2  -ddump-simpl -O1 -fno-full-laziness Test.hs` 
it outputs:
```
...
topEntity :: Bool -> Bool
[GblId,
 Arity=1,
 Str=<L,A>,
 Unf=Unf{Src=InlineStable, TopLvl=True, Value=True, ConLike=True,
         WorkFree=True, Expandable=True,
         Guidance=ALWAYS_IF(arity=1,unsat_ok=True,boring_ok=True)
         Tmpl= \ _ [Occ=Dead] -> someFunction}]
topEntity = \ _ [Occ=Dead] -> someFunction
...
```
After simplification GHC considers the argument to topEntity unused.

But good news is this is fixed in the [current ghc head](https://gitlab.haskell.org/ghc/ghc/-/tree/85e621234916e9b5d40174831a3b422bd99f8f83)

christiaanb, 2020-09-03:
It's broken in `ghc-9.0.1-alpha1`

leonschoorl, 2020-09-04:
`git bisect` tells me it was fixed by [this commit](https://gitlab.haskell.org/ghc/ghc/-/merge_requests/3318/diffs?commit_id=c221207aa69092e9dff21310e797d0d35c47afde)

JvWesterveld, 2020-09-04:
Reported upstream: https://gitlab.haskell.org/ghc/ghc/-/issues/18649

JvWesterveld, 2020-12-18:
This should have been fixed upstream in GHC [master](https://gitlab.haskell.org/ghc/ghc/-/merge_requests/4007) and backported to [9.0](https://gitlab.haskell.org/ghc/ghc/-/merge_requests/4288).

martijnbastiaan, 2021-03-19:
Thanks @JvWesterveld! I guess that fixes this issue.

closed 2021-03-19 by martijnbastiaan