clash-lang/clash-compiler #1507: `autoReg @System @(Maybe Int)` can't be used with undefined reset values

martijnbastiaan, 2020-09-09:
Minimal example:

```haskell
import Clash.Prelude

topEntity :: 
  SystemClockResetEnable => 
  Signal System (Maybe Int) ->
  Signal System (Maybe Int)
topEntity = autoReg  (errorX "foo bar")       
```

Error:

```
Test.hs:7:1: error:
    
    Clash.Netlist.BlackBox(188): Couldn't instantiate blackbox for Clash.Signal.Internal.register#. Verification procedure reported:
    
    Argument 5 should be literal, as blackbox used ~CONST[5], but was:
    
    Identifier "initial_0" Nothing
    
    The source location of the error is not exact, only indicative, as it is acquired 
    after optimizations. The actual location of the error can be in a function that is 
    inlined. To prevent inlining of those functions, annotate them with a NOINLINE pragma.
  |
7 | topEntity = autoReg  (errorX "foo bar")
  | ^^^^^^^^^
```

When running with `-fclash-debug DebugSilent` I get:

```
Irreducible constant as case subject: Clash.XException.errorX
  @(GHC.Maybe.Maybe[3674937295934324792] GHC.Types.Int[3674937295934324766])
  (Clash.Transformations.removedArg @GHC.Stack.Types.CallStack[3674937295934325108])
  (Clash.Transformations.removedArg
     @(GHC.Types.[][3674937295934324788] GHC.Types.Char[3674937295934324752]))
Can be reduced to: Clash.XException.errorX
  @(GHC.Maybe.Maybe[3674937295934324792] GHC.Types.Int[3674937295934324766])
  (Clash.Transformations.removedArg @GHC.Stack.Types.CallStack[3674937295934325108])
  (Clash.Transformations.removedArg
     @(GHC.Types.[][3674937295934324788] GHC.Types.Char[3674937295934324752]))
Irreducible constant as case subject: Clash.XException.errorX
  @(GHC.Maybe.Maybe[3674937295934324792] GHC.Types.Int[3674937295934324766])
  (Clash.Transformations.removedArg @GHC.Stack.Types.CallStack[3674937295934325108])
  (Clash.Transformations.removedArg
     @(GHC.Types.[][3674937295934324788] GHC.Types.Char[3674937295934324752]))
Can be reduced to: Clash.XException.errorX
  @(GHC.Maybe.Maybe[3674937295934324792] GHC.Types.Int[3674937295934324766])
  (Clash.Transformations.removedArg @GHC.Stack.Types.CallStack[3674937295934325108])
  (Clash.Transformations.removedArg
     @(GHC.Types.[][3674937295934324788] GHC.Types.Char[3674937295934324752]))
Irreducible constant as case subject: Clash.XException.errorX
  @(GHC.Maybe.Maybe[3674937295934324792] GHC.Types.Int[3674937295934324766])
  (Clash.Transformations.removedArg @GHC.Stack.Types.CallStack[3674937295934325108])
  (Clash.Transformations.removedArg
     @(GHC.Types.[][3674937295934324788] GHC.Types.Char[3674937295934324752]))
Can be reduced to: Clash.XException.errorX
  @(GHC.Maybe.Maybe[3674937295934324792] GHC.Types.Int[3674937295934324766])
  (Clash.Transformations.removedArg @GHC.Stack.Types.CallStack[3674937295934325108])
  (Clash.Transformations.removedArg
     @(GHC.Types.[][3674937295934324788] GHC.Types.Char[3674937295934324752]))
Irreducible constant as case subject: Clash.XException.errorX
  @(GHC.Maybe.Maybe[3674937295934324792] GHC.Types.Int[3674937295934324766])
  (Clash.Transformations.removedArg @GHC.Stack.Types.CallStack[3674937295934325108])
  (Clash.Transformations.removedArg
     @(GHC.Types.[][3674937295934324788] GHC.Types.Char[3674937295934324752]))
Can be reduced to: Clash.XException.errorX
[... 160 lines cut ...]
  @(GHC.Maybe.Maybe[3674937295934324792] GHC.Types.Int[3674937295934324766])
  (Clash.Transformations.removedArg @GHC.Stack.Types.CallStack[3674937295934325108])
  (Clash.Transformations.removedArg
     @(GHC.Types.[][3674937295934324788] GHC.Types.Char[3674937295934324752]))
Irreducible constant as case subject: Clash.XException.errorX
  @(GHC.Maybe.Maybe[3674937295934324792] GHC.Types.Int[3674937295934324766])
  (Clash.Transformations.removedArg @GHC.Stack.Types.CallStack[3674937295934325108])
  (Clash.Transformations.removedArg
     @(GHC.Types.[][3674937295934324788] GHC.Types.Char[3674937295934324752]))
Can be reduced to: Clash.XException.errorX
  @(GHC.Maybe.Maybe[3674937295934324792] GHC.Types.Int[3674937295934324766])
  (Clash.Transformations.removedArg @GHC.Stack.Types.CallStack[3674937295934325108])
  (Clash.Transformations.removedArg
     @(GHC.Types.[][3674937295934324788] GHC.Types.Char[3674937295934324752]))
Irreducible constant as case subject: Clash.XException.errorX
  @(GHC.Maybe.Maybe[3674937295934324792] GHC.Types.Int[3674937295934324766])
  (Clash.Transformations.removedArg @GHC.Stack.Types.CallStack[3674937295934325108])
  (Clash.Transformations.removedArg
     @(GHC.Types.[][3674937295934324788] GHC.Types.Char[3674937295934324752]))
Can be reduced to: Clash.XException.errorX
  @(GHC.Maybe.Maybe[3674937295934324792] GHC.Types.Int[3674937295934324766])
  (Clash.Transformations.removedArg @GHC.Stack.Types.CallStack[3674937295934325108])
  (Clash.Transformations.removedArg
     @(GHC.Types.[][3674937295934324788] GHC.Types.Char[3674937295934324752]))
```

christiaanb, 2020-09-09:
It's because `errorX` is not in this list: https://github.com/clash-lang/clash-compiler/blob/7385be8170ac3c918cd28fd54f46f8ff9abaa7c7/clash-lib/src/Clash/Normalize/Transformations.hs#L585-L606

leonschoorl, 2020-09-09:
Because of that it fail to evaluate this:
https://github.com/clash-lang/clash-compiler/blob/7385be8170ac3c918cd28fd54f46f8ff9abaa7c7/clash-prelude/src/Clash/Class/AutoReg/Internal.hs#L149
And can't get the initial value of the tag register

commit 78216e1b (2020-09-10): `caseCon` and the evaluator recognise `errorX` as bottom

Fixes #1507
commit 30f73331 (2020-09-10): `caseCon` and the evaluator recognise `errorX` as bottom

Fixes #1507
PR #1512 (2020-09-10): `caseCon` and the evaluator recognise `errorX` as bottom
Fixes #1507
commit cedd0eea (2020-09-10): `caseCon` and the evaluator recognise `errorX` as bottom

Fixes #1507
closed 2020-09-10 by martijnbastiaan