clash-lang/clash-compiler #1524: Not Enough Functions Rendered

paddytheplaster, 2020-09-16:
Hello again,

I get the following when I try to synthesize my `topEntity`.

    Clash error call:
    renderElem: not enough functions rendered? Needed 2 got only 1
    CallStack (from HasCallStack):
      error, called at src/Clash/Util.hs:214:38 in clash-lib-1.3.0-JvRYOpSVzUa7pk3nQEdcto:Clash.Util
      indexNote, called at src/Clash/Util.hs:201:21 in clash-lib-1.3.0-JvRYOpSVzUa7pk3nQEdcto:Clash.Util
      indexNote', called at src/Clash/Netlist/BlackBox/Util.hs:333:15 in clash-lib-1.3.0-JvRYOpSVzUa7pk3nQEdcto:Clash.Netlist.BlackBox.Util
      renderElem, called at src/Clash/Primitives/Sized/Vector.hs:179:30 in clash-lib-1.3.0-JvRYOpSVzUa7pk3nQEdcto:Clash.Primitives.Sized.Vector
      foldTF', called at src/Clash/Primitives/Sized/Vector.hs:134:43 in clash-lib-1.3.0-JvRYOpSVzUa7pk3nQEdcto:Clash.Primitives.Sized.Vector

FWIW This may be related to the following: when I tried to synthesize my `topEntity` with a slightly smaller size.

/home/paddy/Altera/Clash/src/Forest.lhs:226:3: error:

        Hit specialisation limit 20 on function `c$Forest.next_forest_sel3[609793]'.
        [ many lines deleted ]
        Argument (in position: 1) that triggered termination:
        20
        Run with '-fclash-spec-limit=N' to increase the specialisation limit to N.
        [snip]

I then used `-fclash-spec-limit=100`, which got rid of the error. However when I synthesised for the slightly larger `topEntity`, I got the error I mentioned at the top.

Any suggestions?

Regards,

Paddy

martijnbastiaan, 2020-09-16:
Interesting. I guess you're using `fold`? Would it be easy to make a small example triggering the issue?

paddytheplaster, 2020-09-16:
Hi Martijn,

Yes, I'm using `fold`.

I estimate it will take a while to narrow this down but I'll see what I can do.

Regards,

Paddy

paddytheplaster, 2020-09-16:
> Interesting. I guess you're using `fold`? Would it be easy to make a small example triggering the issue?

I have a short-ish example. I can send off line.

Regards,

Paddy

martijnbastiaan, 2020-09-16:
Cool, go for it :)

paddytheplaster, 2020-09-16:
Where do I send it, please?

paddytheplaster, 2020-09-17:
> Interesting. I guess you're using `fold`? Would it be easy to make a small example triggering the issue?

Hello again,

The files `TopEntity.lhs`, `Blur.lhs`, and `Blah.lhs` demonstrate the problem.

Compile with different flags gives you different errors.

    o clash -O1 --vhdl TopEntity.lhs
    o clash -O0 --vhdl TopEntity.lhs

The problem is the combination of `fold` and `unconcat` in the last lines in Blah.lhs.

FWIW When I was simulating a the original application I had already noticed a huge space leak which showed that `split`, `map`, `concat`, and `zipWith` produced a lot of `Vec` space.

Regards,

Paddy
[Paddy.zip](https://github.com/clash-lang/clash-compiler/files/5236379/Paddy.zip)

paddytheplaster, 2020-09-17:
> > Interesting. I guess you're using `fold`? Would it be easy to make a small example triggering the issue?
> 
> Hello again,
> 
> The files `TopEntity.lhs`, `Blur.lhs`, and `Blah.lhs` demonstrate the problem.
> 
> Compiling with different flags gives you different errors.
> 
> ```
> o clash -O1 --vhdl TopEntity.lhs
> o clash -O0 --vhdl TopEntity.lhs
> ```
> 
> The problem is the combination of `fold` and `unconcat` in the last lines in Blah.lhs.
> 
> FWIW When I was simulating the original application I had already noticed a huge space leak which showed that `split`, `map`, `concat`, and `zipWith` produced a lot of `Vec` space.
> 
> Regards,
> 
> Paddy
> [Paddy.zip](https://github.com/clash-lang/clash-compiler/files/5236379/Paddy.zip)

Further to this I remember noticing that _sometimes_ changing expresions like `SNat @(n-k) :: SNat (n-k)` to `undefined :: SNat (n-k)` got rid of the error. (I know the explicit type specification for the first epression is redundant...)

Regards,

Paddy

paddytheplaster, 2020-09-22:
Any updates/suggestions for workarounds? (I still can't synthesize.)

TIA

Paddy

paddytheplaster, 2020-10-13:
> Cool, go for it :)

Hello again,

I spent about one month generalising my computations and the resulting code caused this bug. Is there any idea how long it will take (approximately) to resolve this bug? The last thing I want is having to reverse my changes because of lack of progress with this bug. However, in the end I can't wait forever and may have to revert if needed. Please provide some educated guess so I can make some decision on how to proceed. 

TIA

Paddy

alex-mckenna, 2020-10-14:
Hi Paddy,

Sorry for the delay looking at this issue / giving an update, I believe it should be one of the next issues to be tackled. I can't give you an educated guess on how long it will take to fix, but if it's straightforward hopefully it won't be too long. If it looks like we won't be able to fix it soon then someone will update the issue so you're not left in the dark

paddytheplaster, 2020-10-14:
Thanks Alex.

leonschoorl, 2020-10-15:
Simplified it to a fold inside of a map:

```haskell
topEntity = trans2 @2

trans2 :: Vec k (Vec (k + 1) Bool) -> Vec k Bool
trans2 input = map (fold (||)) input
```

I'll look further into it tomorrow

leonschoorl, 2020-10-16:
If you'd like to experiment Paddy, I've found a quick hack that seems to help:
https://github.com/leonschoorl/clash-compiler/commit/88ba403864324bf2c273c4d1298e026876d4f065

I'll talk with @martijnbastiaan on Monday about a proper fix

paddytheplaster, 2020-10-19:
Thanks. Will have a look later this week. Will get back to let you know.

Regards,

Paddy

leonschoorl, 2020-10-19:
That hack didn't work for (System)Verilog and only sort of accidentally worked for VHDL.
The proper fix is in #1553

commit e6d4fe05 (2020-10-19): Provide mkBlackBoxContext with the correct primName

Fixes #1524

    map (fold f) ...

Would fail to generate the netlist because:
 1. `mkBlackBoxContext` for `fold` would be called  with "__INTERNAL__" as name
 2. this  would cause  `funcPlurality` to default to 1
 3. and thus not enough copies of `f` would be pre-rendered
 4. causing the rendering of `fold` to fail: "not enough functions rendered"
commit 0e37e382 (2020-10-19): Provide mkBlackBoxContext with the correct primName

Fixes #1524

    map (fold f) ...

Would fail to generate the netlist because:
 1. `mkBlackBoxContext` for `fold` would be called  with "__INTERNAL__" as name
 2. this  would cause  `funcPlurality` to default to 1
 3. and thus not enough copies of `f` would be pre-rendered
 4. causing the rendering of `fold` to fail: "not enough functions rendered"
PR #1553 (2020-10-20): Provide mkBlackBoxContext with the correct primName
Fixes #1524

    map (fold f) ...

Would fail to generate the netlist because:
 1. `mkBlackBoxContext` for `fold` would be called  with `"__INTERNAL__"` as name
 2. this  would cause  `funcPlurality` to default to 1
 3. and thus not enough copies of `f` would be pre-rendered
 4. causing the rendering of `fold` to fail: "not enough functions rendered"
commit 4eda7203 (2020-10-20): Provide mkBlackBoxContext with the correct primName (#1553)

Fixes #1524

    map (fold f) ...

Would fail to generate the netlist because:
 1. `mkBlackBoxContext` for `fold` would be called  with "__INTERNAL__" as name
 2. this  would cause  `funcPlurality` to default to 1
 3. and thus not enough copies of `f` would be pre-rendered
 4. causing the rendering of `fold` to fail: "not enough functions rendered"
closed 2020-10-20 by martijnbastiaan