clash-lang/clash-compiler #1534: Cannot reduce to an integer: closed type family

martijnbastiaan, 2020-09-24:
```haskell
{-# LANGUAGE UndecidableInstances #-}

module Issue where

import Clash.Prelude

type family Width a where
  Width a = BitSize a

topEntity :: BitVector (Width Bool) -> ()
topEntity _ = ()
```

gives:

```
    Clash.Netlist.Util(910): Clash.Core.Util(322): Cannot reduce to an integer:
    Clash.Class.BitPack.BitSize[8214565720323786306]
      GHC.Types.Bool[3674937295934324744]
```

however:

```haskell
*Issue> :kind! (BitVector (Width Bool))
(BitVector (Width Bool)) :: Type
= BitVector 1
```

Apparently this violates one of Clash's invariants. When using:

```haskell
topEntity :: BitVector (Width Bool)
topEntity = 0
```

It gives me a proper (but still wrong?) error message:

```haskell
    Clash.Normalize(223): This bndr has a non-representable return type and can't be normalized:
    Issue.topEntity :: Clash.Sized.Internal.BitVector.BitVector
                         (Issue.Width GHC.Types.Bool)
```

christiaanb, 2020-09-24:
I can't reproduce on `master` using GHC 8.8.3

martijnbastiaan, 2020-09-24:
This seems to only happen when using the clash-starter-project. Definitions in the starter projected are loaded from interface files, so it seems natural for it to be some issue in `LoadModules`..

martijnbastiaan, 2020-09-24:
from chat:

> it only fails for the instances defined in Clash.Class.BitPack :
> ```
> BitVector n
> Bit
> Bool
> Int
> Word
> etc
> ```
> But works for any instance defined outside of it, e.g.:
>
>```
>Vec n a
>Signed n
>```

I'll use `-isrc` as a workaround for now and continue with my other tasks.

martijnbastiaan, 2020-09-24:
Another workaround; add:

```
Signal dom (BitVector (BitSize ()))
```

Now it does find `BitSize` and compiles