clash-lang/clash-compiler #1536: Internal error in 'reduceBindersCleanup'

gergoerdi, 2020-09-25:
> Internal error: `reduceBindersCleanup` encountered a variable reference that was 
neither in `doneInl`, `origInl`, or in the transformation's in scope set. Unique 
was: `3`.

This will be quite difficult to minimize... I have uploaded the whole shebang to https://github.com/gergoerdi/clash-spaceinvaders/tree/clash-bug-reduceBindersCleanup.

Naive bisection shows that the first change on my side that causes it is somewhere between 
https://github.com/gergoerdi/clash-intel8080/commits/66ca632 and https://github.com/gergoerdi/clash-intel8080/commits/ff9e07b.

gergoerdi, 2020-09-25:
I have uploaded a version that at least doesn't contain a full Space Invaders machine :)

https://github.com/gergoerdi/clash-intel8080/tree/clash-issue-1536

Build with `stack build && stack exec -- runhaskell build.hs`.

martijnbastiaan, 2020-09-25:
A few pointers for people looking to solve this bug:

1. It's probably useful to trace `origInl`. A thing with a unique of `3` heavily implies it was generated by Clash. It'd be interesting to see how it's used.
2. `reduceBindersCleanup` is tested in isolation here: https://github.com/clash-lang/clash-compiler/blob/b0a239e1a6b0e0d0f7cf63a0c85f2c1448a0a13e/clash-lib/tests/Clash/Tests/Normalize/Transformations.hs. If there's nothing obvious wrong in step (1), try to put the entirety of the graph in `origInl` in there (like `t1337a`) and start reducing it until it doesn't fail.

@gergoerdi This might just be a frivolous error message. If you're in a hurry you might want to try removing it entirely from `reduceBindersCleanup` and see if your design still works.

gergoerdi, 2020-09-27:
I tried removing the error message, but then I get this:

```
<no location info>: error:
    Clash error call:
    Internal error: unknown normalized binder:
    
      Hardware.Intel8080.Steps.stepsOf_go[1638913] :: GHC.Tuple.(,)[3746994889972252676]
                                                        (Hardware.Intel8080.Steps.IMaybe[8214565720324086854]
                                                           GHC.Types.False[3891110078048108556]
                                                           Hardware.Intel8080.Microcode.InAddr[8214565720324083765])
                                                        (Clash.Sized.Vector.Vec[8214565720323789687]
                                                           1
                                                           (GHC.Tuple.(,)[3746994889972252676]
                                                              Hardware.Intel8080.Microcode.MicroInstr[8214565720324083787]
                                                              (GHC.Maybe.Maybe[3674937295934324792]
                                                                 (Data.Either.Either[3674937295934324904]
                                                                    Hardware.Intel8080.Microcode.OutAddr[8214565720324083764]
                                                                    Hardware.Intel8080.Microcode.InAddr[8214565720324083765]))))
    CallStack (from HasCallStack):
      error, called at src/Clash/Netlist.hs:610:20 in clash-lib-1.3.0-7LqQya9E8am3HCoXGJ8tNP:Clash.Netlist
      mkFunApp, called at src/Clash/Netlist.hs:360:3 in clash-lib-1.3.0-7LqQya9E8am3HCoXGJ8tNP:Clash.Netlist
      mkDeclarations', called at src/Clash/Netlist.hs:345:18 in clash-lib-1.3.0-7LqQya9E8am3HCoXGJ8tNP:Clash.Netlist
      mkDeclarations, called at src/Clash/Netlist.hs:251:40 in clash-lib-1.3.0-7LqQya9E8am3HCoXGJ8tNP:Clash.Netlist
      genComponentT, called at src/Clash/Netlist.hs:211:41 in clash-lib-1.3.0-7LqQya9E8am3HCoXGJ8tNP:Clash.Netlist
      genComponent, called at src/Clash/Netlist.hs:123:14 in clash-lib-1.3.0-7LqQya9E8am3HCoXGJ8tNP:Clash.Netlist
Error when running Shake build system:
  at want, called at src/Development/Shake/Internal/Args.hs:83:67 in shake-0.19.1-9C801xFJmW0658JJbFxRUX:Development.Shake.Internal.Args
* Depends on: clash
* Depends on: _build/clash-syn/verilog/Main/topEntity/topEntity.manifest
* Raised the exception:
```

gergoerdi, 2020-09-27:
It seems I can work around this issue by [doing some Template Haskell](https://github.com/gergoerdi/clash-intel8080/commit/f24eb0cab90ac75f388d5ceb83f5e70f3fab96af) to force microcode to normal form before hitting Clash.

martijnbastiaan, 2020-09-30:
@gergoerdi When I try to run

```
stack build && stack runhaskell -- build.hs
```

I get: 

```
<no location info>: error: can't find file: build.hs
```

Did you forget to add `build.hs`? :)

gergoerdi, 2020-09-30:
Are you on the right branch? It should be `clash-issue-1536`. I can see the file [right there](https://github.com/gergoerdi/clash-intel8080/blob/clash-issue-1536/build.hs).

martijnbastiaan, 2020-09-30:
*facepalm*

I removed the repo because I forgot `--recursive` (and didn't want to bother looking up how to clone submodules after the fact), and _then_ forgot to checkout the branch again.

gergoerdi, 2020-12-06:
As of e6738d181690c22e931779f47ad058eb50c266f4 I managed to trigger this again:

```
<no location info>: error:
    Clash error call:
    Internal error: 'reduceBindersCleanup' encountered a variable reference that was 
    neither in 'doneInl', 'origInl', or in the transformation's in scope set. Unique 
    was: '6989586621679302210'.
    CallStack (from HasCallStack):
      error, called at src/Clash/Normalize/Transformations.hs:2664:11 in clash-lib-1.3.0-BtaSQT9Z6y828nb9DyhjHh:Clash.Normalize.Transformations
      reduceBindersCleanup, called at src/Clash/Normalize/Transformations.hs:2625:30 in clash-lib-1.3.0-BtaSQT9Z6y828nb9DyhjHh:Clash.Normalize.Transformations
      inlineBndrsCleanup, called at src/Clash/Normalize/Transformations.hs:2510:19 in clash-lib-1.3.0-BtaSQT9Z6y828nb9DyhjHh:Clash.Normalize.Transformations
      inlineCleanup, called at src/Clash/Normalize/Strategy.hs:57:53 in clash-lib-1.3.0-BtaSQT9Z6y828nb9DyhjHh:Clash.Normalize.Strategy
```

Full code is at https://github.com/gergoerdi/clash-issue-1536/tree/5a4f39db790eb1bac425e1d3883e27a353df6d42 just run `build.sh` after checkout

gergoerdi, 2020-12-06:
I will keep pushing to https://github.com/gergoerdi/clash-issue-1536/tree/clash-issue-1536 as I try to simplify my code.

gergoerdi, 2020-12-06:
Hrmpf, I managed to simplify my code to the point where I am getting "`Clash.Normalize`(192): Clash can only normalize monomorphic functions, but this is polymorphic" instead, so I guess I'll have to completely rethink my approach in https://github.com/gergoerdi/retroclash-lib/blob/05b6f8cdaebfb77edb4db4a422b0817feeaca770/src/RetroClash/Memory.hs anyway...

christiaanb, 2020-12-06:
Adding `-fclash-debug DebugSilent` shows that it goes wrong earlier: ANF introduces free variables. However, that again is a proxy for a failure upstream. That is, I'm seeing that the expression on which ANF fails is:
```
<no location info>: error:
    Clash error call:
    Clash.Rewrite.Util(249): Error when applying rewrite ANF to:
    λ(r1[8286623314361845567] :: Clash.Signal.Internal.Signal[8214565720323789676]
                                   "System"
                                   (GHC.Maybe.Maybe[3674937295934324792]
                                      (Clash.Sized.Internal.Unsigned.Unsigned[8214565720323789711]
                                         16))) ->
    case Data.Dependent.Map.unionWithKey
```
where I expect that that `unionWithKey` should have been compile-time evaluated. (The ANF transform happens after compile-time evaluation). The problem is that ANF only works correctly when certain invariants hold, but those invariants aren't checked (adding that check would make the common case expensive, so we don't do it), and those invariants only don't hold when something is bugged earlier in the compile process.

Anyhow... investigating further why the compile-time evaluation of that `unionWithKey` isn't being evaluated at compile-time.

gergoerdi, 2020-12-06:
Well ideally the whole of this `memoryMap` stuff should be evaluated at compile-time. In the meantime, in desperation I started rewriting the whole shebang using Template Haskell but it is looking way worse...

christiaanb, 2020-12-06:
Mmm... if that `Data.Dependent.Map.unionWithKey` is this one: https://hackage.haskell.org/package/dependent-map-0.4.0.0/docs/src/Data.Dependent.Map.html#unionWithKey

Than I think what is going on is that there is not unfolding / Core expression corresponding to that `unionWithKey`, as it is recursive, and so Clash cannot evaluate at compile time. And recursive bindings never get an unfolding, unless the library is build with `-fexpose-all-unfoldings`. @gergoerdi Are you building all your dependencies with `-fexpose-all-unfoldings`? Is that possbible with `stack`?

With `cabal` you could add the following to your `cabal.project` file:
```
package *
  ghc-options: -fexpose-all-unfoldings
```

gergoerdi, 2020-12-06:
Hmm that's a very good point -- I build all my own libraries with `-fexpose-all-unfoldings` but I don't know how to pass that to all eternal dependencies as well. Let me look around.

gergoerdi, 2020-12-06:
> @gergoerdi Are you building all your dependencies with -fexpose-all-unfoldings? Is that possbible with stack?

[There is an option](https://docs.haskellstack.org/en/stable/yaml_configuration/#ghc-options), but the details [aren't pretty](https://github.com/commercialhaskell/stack/issues/849#issuecomment-320892095). I am now trying to compile with the flag set on only the `dependent-map` package, and, well, all I can say so far is that it hasn't errored out yet... it's been running for 5+ minutes, so let's wait and see.

gergoerdi, 2020-12-06:
BTW, what exactly does the 'inlining limit' mean? I am getting error messages from Clash that an inlining limit of e.g. 100 isn't enough to deal with the `unionWithKey` stuff, but I am building tiny `DMap`s with 4 entry...

christiaanb, 2020-12-06:
It limits how often a (global) function `f` is inlined in the subject of the expression `case (f ...) of`. The inlining currently only happens when `f` returns a data type that cannot be represented by a bundle of wires. It is limited because Clash cannot determine beforehand whether inlining will ever terminate.

gergoerdi, 2020-12-06:
So does that mean I am somehow getting a 100-deep recursive chain of calls to `go` in `unionWithKey`? That would be very surprising for a 4-element `DMap`.

christiaanb, 2020-12-06:
Yes, that’s what it means. The eager parallel reduction is not necessarily efficient. Actually, Clash doesn’t have a well defined compile-time semantics at all. The underlying rewrite mechanism is only guaranteed to be able to translate a Haskell program to a circuit when all transformations are applied exhaustively; the implementation has picked a somewhat ad-hoc set of tree traversals to achieve this exhaustive rewrite.

What this means is that Clash is/can be very bad at compile-time evaluation, especially partial evaluation (where the expression under evaluation has variables). That is why @alex-mckenna is hard at work to create a proper partial evaluator. But the problem is very non-trivial to solve, so an industrial-strength implementation is still far away.

gergoerdi, 2020-12-06:
Soooo does this mean this is hopeless, and I should start thinking seriously about the TH based approach?

gergoerdi, 2020-12-06:
Is there some debug flag I could pass to `clash` to see what exactly (if anything) it is doing while it is spinning at 100% CPU with `-fexpose-all-unfoldings` on `dependent-map` and a large enough `-fclash-inline-limit`?

martijnbastiaan, 2020-12-06:
You could try `-fclash-debug DebugName` to see a list of (applied) transformations. [More verbose options exist](https://clash-lang.readthedocs.io/en/latest/developing-hardware/flags.html) but aren't guaranteed to producte readable output :^).

gergoerdi, 2020-12-07:
My hunch is that the problem is going to be [this recursive piece of code](https://github.com/gergoerdi/retroclash-lib/blob/a304e25567a581f57770d77541e62fa0f78ff952/src/RetroClash/Memory.hs#L72):

```
{-# INLINE memoryMap #-}
memoryMap
    :: Signal dom (Maybe addr)
    -> Signal dom (Maybe dat)
    -> (forall s. Addressing s dom dat addr a)
    -> (Signal dom (Maybe dat), a)
memoryMap addr wr body = (join <$> firstIn read, x)
  where
    (x, (read, conns)) = evalRWS (unAddressing body) (fanInMaybe addr, wr, conns) 0
```

Here, we tie the knot on `conns` which is an essential part of the code: we need to know the address line going into a memory element when we create that memory element, but we need memory element handles so we can connect address lines to them.

gergoerdi, 2020-12-07:
> You could try `-fclash-debug DebugName` to see a list of (applied) transformations. 

It ends up cycling on the following:

```
caseCase {205290}
caseCase {205291}
applicationPropagation {205292}
bindOrLiftNonRep {205293}
caseCon {205294}
applicationPropagation {205295}
bindOrLiftNonRep {205296}
bindOrLiftNonRep {205297}
caseCase {205298}
applicationPropagation {205299}
bindOrLiftNonRep {205300}
caseCon {205301}
caseCase {205302}
caseCase {205303}
applicationPropagation {205304}
caseCase {205305}
applicationPropagation {205306}
caseCase {205307}
applicationPropagation {205308}
bindOrLiftNonRep {205309}
```

christiaanb, 2020-12-07:
Does it terminate if you don't use a knot-tied defined `conns`? But something like:
```
{-# INLINE memoryMap #-}
memoryMap
    :: Signal dom (Maybe addr)
    -> Signal dom (Maybe dat)
    -> (forall s. Addressing s dom dat addr a)
    -> _  
    -> (Signal dom (Maybe dat), a)
memoryMap addr wr body bogusConns = (join <$> firstIn read, x)
  where
    (x, (read, _conns)) = evalRWS (unAddressing body) (fanInMaybe addr, wr, bogusConns) 0
```

gergoerdi, 2020-12-08:
The only `bogusConns` I can easily come up with is `mempty`, so I tried that, and that causes Clash to synthesize in 40 seconds. Of course, the resulting circuit isn't what I want :)

christiaanb, 2020-12-08:
There's not some `singleton singleBogusCon` implementation where you can float the `singleBogusCon` value all the way up? So that we can at least get some additional structure.

gergoerdi, 2020-12-08:
While trying to minimize https://github.com/clash-lang/clash-compiler/issues/1611, I managed to trigger this bug again, with a fairly small & self-contained way of reproducing it!

https://github.com/gergoerdi/clash-issue-1611/tree/clash-issue-1536

```
<no location info>: error:
    Clash error call:
    Internal error: 'reduceBindersCleanup' encountered a variable reference that was 
    neither in 'doneInl', 'origInl', or in the transformation's in scope set. Unique 
    was: '6989586621679258275'.
    CallStack (from HasCallStack):
      error, called at src/Clash/Normalize/Transformations.hs:2664:11 in clash-lib-1.3.0-BtaSQT9Z6y828nb9DyhjHh:Clash.Normalize.Transformations
      reduceBindersCleanup, called at src/Clash/Normalize/Transformations.hs:2625:30 in clash-lib-1.3.0-BtaSQT9Z6y828nb9DyhjHh:Clash.Normalize.Transformations
      inlineBndrsCleanup, called at src/Clash/Normalize/Transformations.hs:2510:19 in clash-lib-1.3.0-BtaSQT9Z6y828nb9DyhjHh:Clash.Normalize.Transformations
      inlineCleanup, called at src/Clash/Normalize/Strategy.hs:57:53 in clash-lib-1.3.0-BtaSQT9Z6y828nb9DyhjHh:Clash.Normalize.Strategy
```

gergoerdi, 2020-12-08:
> There's not some `singleton singleBogusCon` implementation where you can float the `singleBogusCon` value all the way up? So that we can at least get some additional structure.

OK I have now done that (see [this commit](https://github.com/gergoerdi/clash-issue-1536/tree/67cd952fc9d72bd661bb929a912f7fe7a231009e)), but I'm not sure what we're supposed to see here. Compiling with an inliner limit of 1000, it doesn't diverge (so that's an improvement compared to the knot-tying version), but it still complains about not enough inlinings:

```
Clash.Normalize.Transformations(432): InlineNonRep: c$Hardware.SpaceInvaders.mainBoard_go_go[660417] already inlined 1000 times in:c$Hardware.SpaceInvaders.mainBoard_go[358191]
Type of the subject is: Data.Dependent.Map.Internal.DMap[8214565720324068564]
  (GHC.Prim.TYPE[3674937295934324912]
     GHC.Types.LiftedRep[3891110078048108766])
  (RetroClash.Memory.Component[8214565720324068547]
     (GHC.Types.Any[3674937295934325098]
        (GHC.Prim.TYPE[3674937295934324912]
           GHC.Types.LiftedRep[3891110078048108766]))
     (GHC.Types.Any[3674937295934325098]
        (GHC.Types.Any[3674937295934325098]
           (GHC.Prim.TYPE[3674937295934324912]
              GHC.Types.LiftedRep[3891110078048108766]))))
  (RetroClash.Memory.FanIn[8214565720324068556] "Dom25")
Function c$Hardware.SpaceInvaders.mainBoard_go[358191] will not reach a normal form, and compilation might fail.
Run with '-fclash-inline-limit=N' to increase the inlining limit to N.
```

gergoerdi, 2020-12-08:
Would it help if I put in the effort to make a single-module version of this (similar to https://github.com/clash-lang/clash-compiler/issues/1611#issuecomment-740572475)? Or it's not going to change anything, because even if the non-knotted version could be synthesized, the current evaluator has no chance of normalizing the knotted one?

christiaanb, 2020-12-08:
Right now, I don't see any reason why the compiler should not be able to normalize the knotted one, aside from:
1. There being a bug in the compiler
2. Some function from a dependency not having a Core unfolding

gergoerdi, 2020-12-09:
Here's a managable (<130 lines!) standalone version of this code:

```
{-# LANGUAGE DerivingStrategies, GeneralizedNewtypeDeriving #-}
{-# LANGUAGE RankNTypes #-}
module Board where

import Clash.Prelude
import Data.Maybe
import Control.Monad
import Control.Monad.RWS

import Data.Kind
import Data.Dependent.Map as DMap
import Data.GADT.Compare
import Type.Reflection

data Component s (addr :: Type) = Component (TypeRep addr) Int

instance GEq (Component s) where
    geq (Component a _) (Component b _) = geq a b

instance GCompare (Component s) where
    gcompare (Component a _) (Component b _) = gcompare a b

newtype FanIn dom a = FanIn{ getFanIn :: Signal dom `Ap` First a }
    deriving newtype (Semigroup, Monoid)

newtype AddrMap s dom = AddrMap{ addrMap :: DMap (Component s) (FanIn dom) }
    deriving newtype (Monoid)

instance Semigroup (AddrMap s dom) where
    AddrMap map1 <> AddrMap map2 = AddrMap $ unionWithKey (const mappend) map1 map2

newtype Addressing s dom dat addr a = Addressing
    { unAddressing :: RWS
          (FanIn dom addr, Signal dom (Maybe dat), AddrMap s dom)
          (FanIn dom (Maybe dat), AddrMap s dom)
          Int
          a
    }
    deriving newtype (Functor, Applicative, Monad)

memoryMap
    :: Signal dom (Maybe addr)
    -> Signal dom (Maybe dat)
    -> (forall s. Addressing s dom dat addr a)
    -> (Signal dom (Maybe dat), a)
memoryMap addr wr body = (join <$> firstIn read, x)
  where
    (x, (read, conns)) = evalRWS (unAddressing body) (fanInMaybe addr, wr, conns) 0

readWrite_
    :: (HiddenClockResetEnable dom, Typeable addr')
    => (Signal dom (Maybe addr') -> Signal dom (Maybe dat) -> Signal dom (Maybe dat))
    -> Addressing s dom dat addr (Component s addr')
readWrite_ mkComponent = Addressing $ do
    component <- Component typeRep <$> get <* modify succ
    (_, wr, addrs) <- ask
    let addr = firstIn . fromMaybe mempty $ DMap.lookup component (addrMap addrs)
        selected = isJust <$> addr
    let read = mkComponent addr wr
    tell (gated (delay False selected) (fanIn read), mempty)
    return component

ram0
    :: (HiddenClockResetEnable dom, 1 <= n, NFDataX dat, Num dat)
    => SNat n
    -> Addressing s dom dat addr (Component s (Index n))
ram0 size@SNat = readWrite_ $ \addr wr ->
[... 641 lines cut ...]
                ds4[8214565720324004468][LocalId]
    in Board.$fApplicativeAddressing2[1194969][GlobalId]
         addr1[32274][LocalId]
         clk[6989586621679208910][LocalId]
         rst[6989586621679208911][LocalId]
         wr[32275][LocalId]
    Run with '-fclash-spec-limit=N' to increase the specialisation limit to N.
    
    The source location of the error is not exact, only indicative, as it is acquired 
    after optimizations. The actual location of the error can be in a function that is 
    inlined. To prevent inlining of those functions, annotate them with a NOINLINE pragma.
    |
123 | topEntity clk rst addr wr = withClockResetEnable clk rst enableGen $ memoryMap addr wr $ do
    | ^^^^^^^^^
```

However, similar to #1611, if I remove the part that actually uses `TypeRep`s, i.e. if I change `topEntity` to

```
topEntity clk rst addr wr = withClockResetEnable clk rst enableGen $ memoryMap addr wr $ do
    () <- return ()
    return ()
```

then it synthesizes without complaints. So now I don't know if this is the same problem as #1611 or not...

gergoerdi, 2020-12-11:
Using the Clash instrumentation from https://github.com/clash-lang/clash-compiler/issues/1611#issuecomment-743139000, I see the following set of unfoldable definitions:

```
--
Control.Monad.Trans.RWS.Lazy.$tcRWST1 
Data.Monoid.$fSemigroupAp_$cstimes 
Data.Monoid.$fSemigroupFirst1 
Data.Monoid.$tcAp1 
Data.OldList.intercalate_$spoly_go 
Data.OldList.prependToAll 
Data.Semigroup.Internal.$wstimesDefault 
Data.Sequence.Internal.$fFunctorSeq_$cfmap 
Data.Sequence.Internal.$fIsListSeq_mkTree 
Data.Sequence.Internal.$w$slookupTree 
Data.Sequence.Internal.$wupdate 
Data.Typeable.Internal.$tcAppOrCon1 
Data.Typeable.Internal.$wmkTrCon 
Data.Typeable.Internal.eqTypeRep 
Data.Typeable.Internal.fpTYPELiftedRep 
Data.Typeable.Internal.typeRepTyCon 
Foreign.Storable.$fStorableFingerprint_$s$wpeekW64 
Foreign.Storable.$fStorableFingerprint_$s$wpokeW64 
GHC.Base.++ 
GHC.Base.++_$s++ 
GHC.Base.map 
GHC.CString.unpackAppendCString# 
GHC.Fingerprint.fingerprintString_go 
GHC.Integer.Type.czeroBigNat 
GHC.Integer.Type.nullBigNat 
GHC.Integer.Type.oneBigNat 
GHC.Integer.Type.zeroBigNat 
GHC.List.$wlenAcc 
GHC.List.reverse1 
GHC.List.splitAt_$s$wsplitAt' 
GHC.Prim.addWordC# 
GHC.Prim.byteArrayContents# 
GHC.Prim.catch# 
GHC.Prim.getSizeofMutableByteArray# 
GHC.Prim.indexWordArray# 
GHC.Prim.newAlignedPinnedByteArray# 
GHC.Prim.newByteArray# 
GHC.Prim.newPinnedByteArray# 
GHC.Prim.plusAddr# 
GHC.Prim.proxy# 
GHC.Prim.raise# 
GHC.Prim.raiseIO# 
GHC.Prim.readWord8OffAddr# 
GHC.Prim.readWordArray# 
GHC.Prim.realWorld# 
GHC.Prim.reallyUnsafePtrEquality# 
GHC.Prim.resizeMutableByteArray# 
GHC.Prim.seq# 
GHC.Prim.setByteArray# 
GHC.Prim.shrinkMutableByteArray# 
GHC.Prim.sizeofByteArray# 
GHC.Prim.timesWord2# 
GHC.Prim.touch# 
GHC.Prim.unsafeFreezeByteArray# 
GHC.Prim.void# 
GHC.Prim.writeWord8OffAddr# 
GHC.Prim.writeWordArray# 
GHC.Show.$fShowInteger_jprintb 
GHC.Show.$witos' 
GHC.Show.$wjblock' 
GHC.Show.$wjsplitf 
GHC.Show.$wshowWord 
GHC.Show.showLitString 
GHC.Stack.Types.getCallStack 
GHC.Types.$tc'AddrRep1 
GHC.Types.$tcTYPE1 
```

Most of these are `base` definitions so they aren't recompiled with `-fexpose-all-unfoldings`. The RWS one is surprising to me, because I have asked `stack` to recompile `mtl` and `transformers` with the flag.  @christiaanb are you seeing a similar list? 

And as for the `Typeable` stuff, of course if the whole of `memoryMap` is compiled away, there is no HDL-time stuff remaining that uses them. So there shouldn't be a need for blackboxes implementing the `Typeable` methods, I think.

christiaanb, 2020-12-11:
@gergoerdi Yeah, I'm seeing a similar list. And you're correct, we should not have blackboxes for the `Typeable` methods, but we do need their `Core` representations in order to do any compile-time evaluation.

gergoerdi, 2020-12-11:
So `dependent-map` (and specifically, its use of `Typeable`) is the only blocking problem here? I will try to think about a non-`dependent-map`-using solution then.

christiaanb, 2020-12-11:
It's hard to determine whether there is an additional problem on top of `dependent-map`s use of `Typeable`.

gergoerdi, 2020-12-12:
I got rid of using `dependent-map` by using `unsafeCoerce` internally (yuck!). First, it looks promising, because taking the smallest example and pushing it through Clash works:

```
{-# LANGUAGE DerivingStrategies, GeneralizedNewtypeDeriving #-}
{-# LANGUAGE RankNTypes #-}
module Board where

import Clash.Prelude
import Data.Maybe
import Control.Monad
import Control.Monad.RWS

import Unsafe.Coerce
import Data.Map as Map

newtype Component s addr = Component Int
    deriving newtype (Eq, Ord)

newtype FanIn dom a = FanIn{ getFanIn :: Signal dom `Ap` First a }
    deriving newtype (Semigroup, Monoid)

newtype AddrMap s dom = AddrMap{ addrMap :: Map Int (FanIn dom ()) }
    deriving newtype (Monoid)

instance Semigroup (AddrMap s dom) where
    AddrMap map1 <> AddrMap map2 = AddrMap $ unionWithKey (const mappend) map1 map2

newtype Addressing s dom dat addr a = Addressing
    { unAddressing :: RWS
          (FanIn dom addr, AddrMap s dom)
          (FanIn dom (Maybe dat), AddrMap s dom)
          Int
          a
    }
    deriving newtype (Functor, Applicative, Monad)

memoryMap
    :: Signal dom (Maybe addr)
    -> (forall s. Addressing s dom dat addr a)
    -> (Signal dom (Maybe dat), a)
memoryMap addr body = (join <$> firstIn read, x)
  where
    (x, (read, conns)) = evalRWS (unAddressing body) (fanInMaybe addr, conns) 0

readWrite_
    :: (HiddenClockResetEnable dom)
    => (Signal dom (Maybe addr') -> Signal dom (Maybe dat))
    -> Addressing s dom dat addr (Component s addr')
readWrite_ mkComponent = Addressing $ do
    component@(Component i) <- Component <$> get <* modify succ
    (_, addrs) <- ask
    let addr = firstIn . fromMaybe mempty $ Map.lookup i (addrMap addrs)
        read = mkComponent $ unsafeCoerce addr
    tell (fanIn read, mempty)
    return component

ram0
    :: (HiddenClockResetEnable dom, 1 <= n, NFDataX dat, Num dat)
    => SNat n
    -> Addressing s dom dat addr (Component s (Index n))
ram0 size@SNat = readWrite_ $ \addr ->
    fmap Just $ blockRam1 ClearOnReset size 0 (fromMaybe 0 <$> addr) (pure Nothing)

connect
    :: Component s addr
    -> Addressing s dom dat addr ()
connect component@(Component i) = Addressing $ do
    (addr, _) <- ask
    tell (mempty, AddrMap $ Map.singleton i $ unsafeCoerce addr)

firstIn :: FanIn dom a -> Signal dom (Maybe a)
firstIn = fmap getFirst . getAp . getFanIn

fanInMaybe :: Signal dom (Maybe a) -> FanIn dom a
fanInMaybe = FanIn . Ap . fmap First

fanIn :: Signal dom a -> FanIn dom a
fanIn = fanInMaybe . fmap pure

topEntity
    :: Clock System
    -> Reset System
    -> Signal System (Maybe (Index 0x0400))
    -> (Signal System (Maybe (Unsigned 8)), ())
topEntity clk rst addr = withClockResetEnable clk rst enableGen $ memoryMap addr $ do
    ram <- ram0 (SNat @0x0400)
    connect ram
```

But then if I add `matchAddr` back so that `topEntity` can get a bit more complex, we are back to square one, except this time it's `containers`'s `Data.Map.lookup` screwing things up:

```
matchAddr
    :: (addr -> Maybe addr')
    -> Addressing s dom dat addr' a
    -> Addressing s dom dat addr a
matchAddr match body = Addressing $ rws $ \(addr, addrs) s ->
  let addr' = fanInMaybe . fmap (match =<<) . firstIn $ addr
  in runRWS (unAddressing body) (addr', addrs) s

from
    :: forall addr' s dom dat addr a. (Integral addr, Ord addr, Integral addr', Bounded addr')
    => addr
    -> Addressing s dom dat addr' a
    -> Addressing s dom dat addr a
from base = matchAddr $ \addr -> do
    guard $ addr >= base
    let offset = addr - base
    guard $ offset <= lim
    return $ fromIntegral offset
  where
    lim = fromIntegral (maxBound :: addr')

topEntity
    :: Clock System
    -> Reset System
    -> Signal System (Maybe (Index 0x0800))
    -> (Signal System (Maybe (Unsigned 8)), ())
topEntity clk rst addr = withClockResetEnable clk rst enableGen $ memoryMap addr $ do
    ram <- ram0 (SNat @0x0400)
    from 0x0000 $ connect ram
    from 0x0400 $ connect ram
```

```
Clash.Normalize.Transformations(432): InlineNonRep: c$Board.$slookup_go13_wild[12457728] already inlined 200 times in:Board.$slookup_go13[12025728]
Type of the subject is: Data.Map.Internal.Map[8214565720323790163]
  GHC.Types.Int[3674937295934324766]
  (Clash.Signal.Internal.Signal[8214565720323789649]
     "System"
     (GHC.Maybe.Maybe[3674937295934324792]
        GHC.Tuple.()[3746994889972252672]))
```

So it seems this will never quite work out, unless we can get the boot libraries compiled with `-fexpose-all-unfoldings` (and probably also `-fno-worker-wrapper`)... Would [this approach](https://github.com/clash-lang/clash-compiler/issues/1611#issuecomment-741665128) work for that, to decouple Clash from GHC and have Clash compile everything on its own?

christiaanb, 2020-12-12:
I think we want the “fat interface files” described here: https://gitlab.haskell.org/ghc/ghc/-/issues/10871 
Which apparently is something ghcide wants as well.

Also to check whether fat interface files would be a solution is to import `Data.Map.Internal` and copy/paste the definitions of the functions that you’re using from `Data.Map`

gergoerdi, 2020-12-12:
> Also to check whether fat interface files would be a solution is to import `Data.Map.Internal` and copy/paste the definitions of the functions that you’re using from `Data.Map`

Yeah I had the same idea in the meantime, I'm working on this right now.

gergoerdi, 2020-12-12:
Well, it's pretty much the same with an included copy of `Data.Map`, just with different non-unfoldable functions remaining:

```
--
Control.Monad.Trans.RWS.Lazy.$tcRWST1 
Data.Data.$c(,) 
Data.Data.$fData[]_$cgfoldl 
Data.Data.$fData[]_$cgmapM 
Data.Data.$fData[]_$cgmapQ 
Data.Data.$fData[]_$cgmapQi 
Data.Data.$fData[]_$cgmapQl 
Data.Data.$fData[]_$cgmapQr 
Data.Data.$fData[]_$cgmapT 
Data.Data.$fData[]_$cgunfold 
Data.Data.$w$cgmapMo11 
Data.Data.$w$cgmapMp16 
Data.Data.consConstr 
Data.Data.listDataType 
Data.Data.nilConstr 
Data.Functor.Classes.$fEq1NonEmpty_$cliftEq1 
Data.Functor.Classes.$fOrd1NonEmpty_$cliftCompare1 
Data.Functor.Classes.$fRead1(,)_$cliftReadList 
Data.Monoid.$fSemigroupAp_$cstimes 
Data.Monoid.$fSemigroupFirst1 
Data.Monoid.$tcAp1 
Data.OldList.intercalate_$spoly_go 
Data.OldList.prependToAll 
Data.Semigroup.Internal.$wstimesDefault 
Data.Sequence.Internal.$fFunctorSeq_$cfmap 
Data.Sequence.Internal.$fIsListSeq_mkTree 
Data.Sequence.Internal.$w$slookupTree 
Data.Sequence.Internal.$wupdate 
Data.Typeable.Internal.$wmkTrCon 
Data.Typeable.Internal.eqTypeRep 
Data.Typeable.Internal.fpTYPELiftedRep 
Foreign.Storable.$fStorableFingerprint_$s$wpeekW64 
Foreign.Storable.$fStorableFingerprint_$s$wpokeW64 
GHC.Base.++ 
GHC.Base.++_$s++ 
GHC.Base.eqString 
GHC.Base.map 
GHC.CString.unpackAppendCString# 
GHC.Classes.$fEq[]_$c== 
GHC.Classes.$fOrd[]_$ccompare 
GHC.Integer.Type.czeroBigNat 
GHC.Integer.Type.nullBigNat 
GHC.Integer.Type.oneBigNat 
GHC.Integer.Type.zeroBigNat 
GHC.List.$wlenAcc 
GHC.List.reverse1 
GHC.List.splitAt_$s$wsplitAt' 
GHC.Prim.addWordC# 
GHC.Prim.byteArrayContents# 
GHC.Prim.catch# 
GHC.Prim.getSizeofMutableByteArray# 
GHC.Prim.indexWordArray# 
GHC.Prim.newAlignedPinnedByteArray# 
GHC.Prim.newByteArray# 
GHC.Prim.newPinnedByteArray# 
GHC.Prim.plusAddr# 
GHC.Prim.proxy# 
GHC.Prim.raise# 
GHC.Prim.raiseIO# 
GHC.Prim.readWord8OffAddr# 
GHC.Prim.readWordArray# 
GHC.Prim.realWorld# 
GHC.Prim.reallyUnsafePtrEquality# 
GHC.Prim.resizeMutableByteArray# 
GHC.Prim.seq# 
GHC.Prim.setByteArray# 
GHC.Prim.shrinkMutableByteArray# 
GHC.Prim.sizeofByteArray# 
GHC.Prim.timesWord2# 
GHC.Prim.touch# 
GHC.Prim.unsafeFreezeByteArray# 
GHC.Prim.void# 
GHC.Prim.writeWord8OffAddr# 
GHC.Prim.writeWordArray# 
GHC.Read.list 
GHC.Real.$wf1 
GHC.Show.$fShow(,)_$sgo1 
GHC.Show.$fShowInteger_jprintb 
GHC.Show.$witos' 
GHC.Show.$wjblock' 
GHC.Show.$wjsplitf 
GHC.Show.showLitString 
GHC.Stack.Types.getCallStack 
GHC.Types.$tc'AddrRep1 
GHC.Types.$tcTYPE1 
Text.ParserCombinators.ReadP.$fAlternativeP_$c<|> 
Text.ParserCombinators.ReadP.gather_gath 
Text.ParserCombinators.ReadP.run 
Text.ParserCombinators.ReadP.skipSpaces2 
Text.Read.Lex.expect2 
_INTERNAL_.{__pkg_ccall base-4.13.0.0 Addr#
_INTERNAL_.{__pkg_ccall base-4.13.0.0 Int#
_INTERNAL_.{__pkg_ccall integer-gmp-1.0.2.0 ByteArray#
_INTERNAL_.{__pkg_ccall integer-gmp-1.0.2.0 forall s.
```

gergoerdi, 2020-12-12:
Eventualy, the specializer chokes on the (copied, and thus unfoldable) version of `Data.Map.lookup`:

```
    Hit specialisation limit 100 on function `c$Board.topEntity_go[11136]'.
    
    The function `c$Board.topEntity_go[11136] :: GHC.Types.Int[3674937295934324766]
                               -> Containers.Data.Map.Internal.Map[8214565720323896040]
                                    GHC.Types.Int[3674937295934324766]
                                    (Board.FanIn[8214565720324216431]
                                       "System"
                                       GHC.Tuple.()[3746994889972252672])
                               -> Clash.Signal.Internal.Signal[8214565720324039903]
                                    "System"
                                    (Data.Monoid.First[8214565720323804916]
                                       GHC.Tuple.()[3746994889972252672])' is most likely recursive, and looks like it is being indefinitely specialized on a growing argument.
    
    Body of `c$Board.topEntity_go[11136] :: GHC.Types.Int[3674937295934324766]
                               -> Containers.Data.Map.Internal.Map[8214565720323896040]
                                    GHC.Types.Int[3674937295934324766]
                                    (Board.FanIn[8214565720324216431]
                                       "System"
                                       GHC.Tuple.()[3746994889972252672])
                               -> Clash.Signal.Internal.Signal[8214565720324039903]
                                    "System"
                                    (Data.Monoid.First[8214565720323804916]
                                       GHC.Tuple.()[3746994889972252672])':
    λ(ds2[7205759403792923820] :: GHC.Types.Int[3674937295934324766]) ->
    λ(ds3[7205759403792923821] :: Containers.Data.Map.Internal.Map[8214565720323896040]
                                    GHC.Types.Int[3674937295934324766]
                                    (Board.FanIn[8214565720324216431]
                                       "System"
                                       GHC.Tuple.()[3746994889972252672])) ->
    letrec
      ds4[6341068275337788595] :: GHC.Types.Int[3674937295934324766]
      = ds2[7205759403792923820][LocalId]
    in case ds4[6341068275337788595][LocalId] of
         GHC.Types.I#[3891110078048108562]
           (ipv[8286623314362161582] :: GHC.Prim.Int#[3674937295934324764]) ->
           case ds3[7205759403792923821][LocalId] of
             Containers.Data.Map.Internal.Bin[8214565720323896041]
               (ipv1[8286623314361847773] :: GHC.Types.Int[3674937295934324766])
                 (ipv2[8286623314361847774] :: GHC.Types.Int[3674937295934324766])
                 (ipv3[8286623314361847775] :: Board.FanIn[8214565720324216431]
                                                 "System"
                                                 GHC.Tuple.()[3746994889972252672])
                 (ipv4[8286623314361847776] :: Containers.Data.Map.Internal.Map[8214565720323896040]
                                                 GHC.Types.Int[3674937295934324766]
                                                 (Board.FanIn[8214565720324216431]
                                                    "System"
                                                    GHC.Tuple.()[3746994889972252672]))
                 (ipv5[8286623314361847777] :: Containers.Data.Map.Internal.Map[8214565720323896040]
                                                 GHC.Types.Int[3674937295934324766]
                                                 (Board.FanIn[8214565720324216431]
                                                    "System"
                                                    GHC.Tuple.()[3746994889972252672])) ->
               case ipv2[8286623314361847774][LocalId] of
                 GHC.Types.I#[3891110078048108562]
                   (y#[6989586621679066384] :: GHC.Prim.Int#[3674937295934324764]) ->
                   case GHC.Prim.<#
                          ipv[8286623314362161582][LocalId]
                          y#[6989586621679066384][LocalId] of
                     _ ->
                       case GHC.Prim.==#
                              ipv[8286623314362161582][LocalId]
                              y#[6989586621679066384][LocalId] of
                         _ ->
                           c$Board.topEntity_go[11136][GlobalId]
                             ds4[6341068275337788595][LocalId]
                             ipv5[8286623314361847777][LocalId]
                         1 ->
[... 433 lines cut ...]
                        GHC.Types.False[3891110078048108556]  ->
                          GHC.Maybe.Nothing[3891110078048108568]
                            @(Clash.Sized.Internal.Index.Index[8214565720324039936]
                                1024)
                        GHC.Types.True[3891110078048108589]  ->
                          GHC.Maybe.Just[3891110078048108571]
                            @(Clash.Sized.Internal.Index.Index[8214565720324039936]
                                1024)
                            (Clash.Sized.Internal.Index.fromInteger# @1024
                               1024
                               (Clash.Sized.Internal.Index.toInteger# @2048
                                  offset[8286623314362161957][LocalId]))
         in eta[4755801206503243789][LocalId]
    in Board.$fApplicativeAddressing_$sunionWithKey[24274924][GlobalId]
[... cut ...]
         addr1[6989586621679445436][LocalId]
    Run with '-fclash-spec-limit=N' to increase the specialisation limit to N.
    
    The source location of the error is not exact, only indicative, as it is acquired 
    after optimizations. The actual location of the error can be in a function that is 
    inlined. To prevent inlining of those functions, annotate them with a NOINLINE pragma.
    |
103 | topEntity clk rst addr = withClockResetEnable clk rst enableGen $ memoryMap addr $ do
    | ^^^^^^^^^
```

christiaanb, 2020-12-12:
This is somewhat better. Specialization is run last in the “eliminate-non-representable-values” phase, indicating the compiler got a whole lot further.

I’ll see whether I can make sense of the Debug output once I have a chance to sit behind a computer. 

Thanks for all the effort in helping with the debugging so far!

gergoerdi, 2020-12-12:
> Thanks for all the effort in helping with the debugging so far!

Well I am completely blocked on this so I am trying everything I can in frustration...

gergoerdi, 2020-12-12:
My full code including the copied `containers` is at https://github.com/gergoerdi/clash-issue-1536/tree/no-typeable

christiaanb, 2020-12-13:
I've "somewhat" reduced it to:
```haskell
{-# LANGUAGE BangPatterns #-}
{-# LANGUAGE DerivingStrategies, GeneralizedNewtypeDeriving #-}
{-# LANGUAGE RankNTypes #-}
{-# LANGUAGE MagicHash #-}
module Test where

import Clash.Prelude
import Data.Maybe
import Control.Monad
import Control.Monad.RWS

import Unsafe.Coerce
import GHC.Magic (lazy)

import Data.Foldable as Foldable
import GHC.Exts ( reallyUnsafePtrEquality#, isTrue# )

type Size     = Int

data Map k a  = Bin {-# UNPACK #-} !Size !k a !(Map k a) !(Map k a)
              | Tip

data StrictPair a b = !a :*: !b

infixr 1 :*:

-- | Convert a strict pair to a standard pair.
toPair :: StrictPair a b -> (a, b)
toPair (x :*: y) = (x, y)
{-# INLINE toPair #-}

unions :: (Foldable f, Ord k) => f (Map k a) -> Map k a
unions ts
  = Foldable.foldl' union Test.empty ts

singleton :: k -> a -> Map k a
singleton k x = Bin 1 k x Tip Tip
{-# INLINE singleton #-}

lookup :: Ord k => k -> Map k a -> Maybe a
lookup = go
  where
    go !_ Tip = Nothing
    go k (Bin _ kx x l r) = case compare k kx of
      LT -> go k l
      GT -> go k r
      EQ -> Just x
{-# INLINABLE lookup #-}

instance (Ord k) => Monoid (Map k v) where
    mempty  = Test.empty
    mconcat = unions
    mappend = union

instance (Ord k) => Semigroup (Map k v) where
    (<>)    = union

ptrEq :: a -> a -> Bool
-- ptrEq x y = isTrue# (reallyUnsafePtrEquality# x y)
ptrEq _ _ = False

empty :: Map k a
empty = Tip
{-# INLINE empty #-}

insert :: Ord k => k -> a -> Map k a -> Map k a
insert kx0 = go kx0 kx0
  where
[... 274 lines cut ...]
    let offset = addr - base
    guard $ offset <= lim
    return $ fromIntegral offset
  where
    lim = fromIntegral (maxBound :: addr')

topEntity
    :: Maybe (Index 0x0400)
    -> ((Maybe (Index 0x0400)), ())
topEntity addr = memoryMap addr $ do
    ram <- ram0 (SNat @0x0400)
    from 0x0000 $ connect ram
    -- from 0x0400 $ connect ram
```
I've removed the `Ap (Signal dom)` aspect of it, and also removed the `unsafeCoerce`, uncommenting the last `from 0x0400 $ connect ram` seems to throw Clash into an infinite loop.

One thing I also noticed is that sometimes we have:
```haskell
\x -> let y = expr_using_x in expr_not_using_y
```
Which is problamatic because dead code elimination happens at a very late stage because the transformation is deemed expensive: n^2 for n number of let-bindings. But n is in general small before the ANF transform, so perhaps we should run it more often during the constant propagation phase, which is also doing all the inlining. It might be that it's doing all the inlining in the `expr_using_x` from the above example.
One thing I did notice at one point is that sometimes we have:
```
\x -> let y = expr using x in expr not using y
```

christiaanb, 2020-12-21:
I've tried adding dead code elimination inside of the main loop, but it doesn't seem to solve the issue.

We don't really have the resources to investigate further (the debug logs are just too big to analyze in a sensible amount of time). I think we must conclude that Clash currently cannot translate the kind of recursive descriptions given here, or those of #1611 

Hopefully, the partial evaluation work done at https://github.com/clash-lang/clash-compiler/tree/partial-evaluator-primitives-easier will be able to handle these kind of recursive description properly. The problem there is that the work is far from production ready, and due to circumstances, development on that feature will be halted for the coming year (unless it becomes sponsored).

It thus pains me to say that (typed) template haskell is probably the only way forward at this moment in time.

gergoerdi, 2020-12-22:
> We don't really have the resources to investigate further (the debug logs are just too big to analyze in a sensible amount of time).

By this, do you mean they are too big because Clash is doing a lot of work, or because the code to reproduce this issue is still too large? In other words, is there room for me to try to reduce my code further, or is this a lost cause?

christiaanb, 2020-12-22:
Clash is doing too much work, which is both a result of the size of the test case and fact it seems to get stuck in an inlining loop. But the logs are huge (thousands of transformations, order of gigabytes size), and so it's hard to pin down whether one of the transformations is doing something dumb/wrong or where a (slightly) different order/strategy of the transformations would lead to termination.

Anyhow, I've managed to get something which does compile while still being close to the original code in spirit:
```haskell
{-# LANGUAGE DerivingStrategies, GeneralizedNewtypeDeriving #-}
module Test3 where

import Clash.Prelude
import Data.Maybe
import Control.Monad
import Control.Monad.RWS

newtype Component addr = Component Int
    deriving newtype (Eq, Ord)

newtype FanIn a = FanIn{ getFanIn :: First a }
    deriving newtype (Semigroup, Monoid)

newtype AddrMap = AddrMap{ addrMap :: [(Int,(FanIn (Index 0x0400)))] }
    deriving newtype (Monoid)

instance Semigroup (AddrMap) where
    AddrMap map1 <> AddrMap map2 = AddrMap $ unionWithKeyList (const mappend) map1 map2

unionWithKeyList :: Ord k => (k -> a -> a -> a) -> [(k,a)] -> [(k,a)] -> [(k,a)]
unionWithKeyList f l r = foldList (insertWithKeyList f) r l

foldList :: (a -> b -> b) -> b -> [a] -> b
foldList _ z []     = z
foldList f z (x:xs) = foldList f (f x z) xs

insertWithKeyList :: Ord k => (k -> a -> a -> a) -> (k,a) -> [(k,a)] -> [(k,a)]
insertWithKeyList _ k [] = [k]
insertWithKeyList f (k,a) ((k1,b):xs) = case compare k k1 of
  LT -> (k,a):(k1,b):xs
  EQ -> (k,f k a b) : xs
  GT -> (k1,b):insertWithKeyList f (k,a) xs

lookupList :: Eq k => k -> [(k,a)] -> Maybe a
lookupList _ [] = Nothing
lookupList k ((k1,a):xs) = if k == k1 then Just a else lookupList k xs

singletonList :: k -> a -> [(k,a)]
singletonList k a = [(k,a)]

newtype Addressing dat addr a = Addressing
    { unAddressing :: RWS
          (FanIn addr, AddrMap)
          (FanIn (Maybe dat), AddrMap)
          Int
          a
    }
    deriving newtype (Functor, Applicative, Monad)

memoryMap
    :: Maybe addr
    -> (Addressing dat addr a)
    -> (Maybe dat, a)
memoryMap addr body = (join (firstIn read), x)
  where
    (x, (read, conns)) = evalRWS (unAddressing body) (fanInMaybe addr, conns) 0

readWrite_
    :: ((Maybe (Index 1024)) -> (Maybe dat))
    -> Addressing dat addr (Component (Index 1024))
readWrite_ mkComponent = Addressing $ do
    component@(Component i) <- Component <$> get <* modify succ
    (_, addrs) <- ask
    let addr = firstIn . fromMaybe mempty $ lookupList i (addrMap addrs)
        read = mkComponent addr
    tell (fanIn read, mempty)
    return component

ram0
    :: (1 <= n)
    => SNat n
    -> Addressing (Index 1024) addr (Component (Index 1024))
ram0 size@SNat = readWrite_ $ \addr ->
    Just $ negate (fromMaybe 2 addr)

connect
    :: Component (Index 1024)
    -> Addressing dat (Index 1024) ()
connect component@(Component i) = Addressing $ do
    (addr, _) <- ask
    tell (mempty, AddrMap $ singletonList i $ addr)

firstIn :: FanIn a -> Maybe a
firstIn = getFirst . getFanIn

fanInMaybe :: Maybe a -> FanIn a
fanInMaybe = FanIn . First

fanIn :: a -> FanIn a
fanIn = fanInMaybe . pure

matchAddr
    :: (addr -> Maybe addr')
    -> Addressing dat addr' a
    -> Addressing dat addr a
matchAddr match body = Addressing $ rws $ \(addr, addrs) s ->
  let addr' = fanInMaybe . (match =<<) . firstIn $ addr
  in runRWS (unAddressing body) (addr', addrs) s

from
    :: forall addr' dat addr a. (Integral addr, Ord addr, Integral addr', Bounded addr')
    => addr
    -> Addressing dat addr' a
    -> Addressing dat addr a
from base = matchAddr $ \addr -> do
    guard $ addr >= base
    let offset = addr - base
    guard $ offset <= lim
    return $ fromIntegral offset
  where
    lim = fromIntegral (maxBound :: addr')

topEntity
    :: Maybe (Index 0x0400)
    -> ((Maybe (Index 0x0400)), ())
topEntity addr = memoryMap addr $ do
    ram <- ram0 (SNat @0x0400)
    from 0x0000 $ connect ram
    from 0x0200 $ connect ram
```
I think _slowly_ building this code back up to the intended use would be a more fruitful exercise than trying figure out how we could get the original code to compile. I would go towards building up in this order:

1. Add back the RankN type
2. Introduce the `Ap Signal` again
3. Introduce the use of `blockRam` again
4. Introduce the use of `unsafeCoerce` again.

Perhaps there's a step 3.5 between 3 and 4 so that `connect` has a polymorphic type signature again, but perhaps that requires the `unsafeCoerce` of step 4. You'll need to synthesize often to make sure it keeps working, and report back when it stops working. As you can see, when you build back up, you cannot depend on recursive functions from pre-compiled dependencies as they might not have an unfolding. Perhaps that would be a step 5, try and use functions from pre-compiled dependencies.

gergoerdi, 2020-12-23:
> I think _slowly_ building this code back up to the intended use would be a more fruitful exercise than trying figure out how we could get the original code to compile.

This is great stuff!

I managed to get it back to the `unsafeCoerce`-using version (albeit without testing its runtime behaviour yet... but hey, it compiles!) using your alist-for-`Map` idea (see [this version](https://github.com/gergoerdi/clash-issue-1536/tree/d1a53e321c31da3cace9efd52c8d81a5dbdafe0f)). Then, [when I try to apply the same idea to `DMap`](https://github.com/gergoerdi/clash-issue-1536/commit/d2247cb5dd573b6c1c4db9f23c0b15f38bc7a9ea#diff-25a6634263c1b1f6fc4697a04e2b9904ea4b042a89af59dc93ec1f5d44848a26) (i.e. an assoc-list of `DSum k f`s), then I get the following which I think could be the result of an easy-to-fix oversight:

```
src/Board.hs:17:1: error:
    
    Hit specialisation limit 100 on function `c$Board.topEntity_go_go[4436866]'.
    
    The function `c$Board.topEntity_go_go[4436866] :: GHC.Prim.Int#[3674937295934324764]
                                    -> GHC.Ptr.Ptr[3674937295934324890]
                                         GHC.Word.Word8[3674937295934324860]
                                    -> GHC.Prim.Addr#[3674937295934324738]
                                    -> GHC.Types.[][3674937295934324788]
                                         GHC.Word.Word8[3674937295934324860]
                                    -> GHC.Prim.Int#[3674937295934324764]
                                    -> GHC.Prim.State#[3674937295934324836]
                                         GHC.Prim.RealWorld[3674937295934324808]
                                    -> GHC.Prim.(#,#)[3819052484010180612]
                                         (GHC.Types.TupleRep[3891110078048108760]
                                            (GHC.Types.[][3891110078048108577]
                                               GHC.Types.RuntimeRep[3674937295934324926]))
                                         GHC.Types.LiftedRep[3891110078048108766]
                                         (GHC.Prim.State#[3674937295934324836]
                                            GHC.Prim.RealWorld[3674937295934324808])
                                         GHC.Fingerprint.Type.Fingerprint[8214565720323785011]' is most likely recursive, and looks like it is being indefinitely specialized on a growing argument.
    
    Body of `c$Board.topEntity_go_go[4436866] :: GHC.Prim.Int#[3674937295934324764]
                                    -> GHC.Ptr.Ptr[3674937295934324890]
                                         GHC.Word.Word8[3674937295934324860]
                                    -> GHC.Prim.Addr#[3674937295934324738]
                                    -> GHC.Types.[][3674937295934324788]
                                         GHC.Word.Word8[3674937295934324860]
                                    -> GHC.Prim.Int#[3674937295934324764]
                                    -> GHC.Prim.State#[3674937295934324836]
                                         GHC.Prim.RealWorld[3674937295934324808]
                                    -> GHC.Prim.(#,#)[3819052484010180612]
                                         (GHC.Types.TupleRep[3891110078048108760]
                                            (GHC.Types.[][3891110078048108577]
                                               GHC.Types.RuntimeRep[3674937295934324926]))
                                         GHC.Types.LiftedRep[3891110078048108766]
                                         (GHC.Prim.State#[3674937295934324836]
                                            GHC.Prim.RealWorld[3674937295934324808])
                                         GHC.Fingerprint.Type.Fingerprint[8214565720323785011]':
    λ(ww2[8214565720324017915] :: GHC.Prim.Int#[3674937295934324764]) ->
    λ(ptr[8214565720324017917] :: GHC.Ptr.Ptr[3674937295934324890]
                                    GHC.Word.Word8[3674937295934324860]) ->
    λ(a[8214565720324017934] :: GHC.Prim.Addr#[3674937295934324738]) ->
    λ(ds[8214565720324017942] :: GHC.Types.[][3674937295934324788]
                                   GHC.Word.Word8[3674937295934324860]) ->
    λ(ds1[8214565720324017943] :: GHC.Prim.Int#[3674937295934324764]) ->
    λ(eta2[8214565720324017944] :: GHC.Prim.State#[3674937295934324836]
                                     GHC.Prim.RealWorld[3674937295934324808]) ->
    case ds[8214565720324017942][LocalId] of
      GHC.Types.[][3891110078048108577]  ->
        c$Board.topEntity_exit[14273][GlobalId]
          ww2[8214565720324017915][LocalId]
          ptr[8214565720324017917][LocalId]
          eta2[8214565720324017944][LocalId]
      GHC.Types.:[3891110078048108550]
        (val1[8214565720324017949] :: GHC.Word.Word8[3674937295934324860])
          (vals2[8214565720324017950] :: GHC.Types.[][3674937295934324788]
                                           GHC.Word.Word8[3674937295934324860]) ->
        case val1[8214565720324017949][LocalId] of
          GHC.Word.W8#[3891110078048108583]
            (x1[8214565720324017954] :: GHC.Prim.Word#[3674937295934324854]) ->
            letrec
              s1[8214565720324017956] :: GHC.Prim.State#[3674937295934324836]
[... 208 lines cut ...]
                                             1)
                                          1)
                                       1)
                                    1)
                                 1)
[... cut ...]
    The source location of the error is not exact, only indicative, as it is acquired 
    after optimizations. The actual location of the error can be in a function that is 
    inlined. To prevent inlining of those functions, annotate them with a NOINLINE pragma.
   |
17 | topEntity addr wr = memoryMap addr wr $ do
   | ^^^^^^^^^
```

Note, however, that I can probably make do with the `unsafeCoerce`-using variant, so I'll do further testing after the holidays and if this works I will be good to go!

christiaanb, 2020-12-23:
Ah, interesting. I already see one primitive for which Clash definitely doesn't have an implementation: `GHC.Prim.writeWord8OffAddr#`. Regardless, I still think the `Typeable` approach is a bit of a lost cause at the moment, simply because of all those missing unfoldings for definitions in `Data.Typeable.Internal`.

gergoerdi, 2020-12-28:
I have restored all the functionality that I need, and also added an implementation of `RWS`, `First` and `Ap` to make sure everything has unfoldings available. However, as soon as the memory map becomes a bit more complex, I am back to hitting limitations.

[In this commit](https://github.com/gergoerdi/clash-issue-1536/tree/24d6a3f863b74ed8e3c9f2c1994289937d99f13c), I can sythesize (by running the included `build.sh` script). However, if I uncomment [the extra `return ()` in `src/Board.hs`](https://github.com/gergoerdi/clash-issue-1536/blob/24d6a3f863b74ed8e3c9f2c1994289937d99f13c/src/Board.hs#L13), then I am back in a world of pain: Clash seems to be looping with the following two messages printed over and over again:

```
Function c$Board.topEntity_go[1429471] will not reach a normal form, and compilation might fail.
Run with '-fclash-inline-limit=N' to increase the inlining limit to N.
Clash.Normalize.Transformations(432): InlineNonRep: c$Board.topEntity_go[4] already inlined 200 times in:c$Board.topEntity_go[1429471]
Type of the subject is: RetroClash.Memory.AddrMap[8214565720323859492]
  (GHC.Types.Any[3674937295934325098]
     (GHC.Prim.TYPE[3674937295934324912]
        GHC.Types.LiftedRep[3891110078048108766]))
  "System"
```
```
Function c$Board.topEntity_go[1429471] will not reach a normal form, and compilation might fail.
Run with '-fclash-inline-limit=N' to increase the inlining limit to N.
Clash.Normalize.Transformations(432): InlineNonRep: RetroClash.Memory.$fMonoidAddrMap_$sinsertWithKey[8214565720323887811] already inlined 200 times in:c$Board.topEntity_go[1429471]
Type of the subject is: Assoc.Map[8214565720323842939]
  GHC.Types.Int[3674937295934324766]
  (RetroClash.Memory.FanIn[8214565720323859495]
     "System"
     GHC.Tuple.()[3746994889972252672])
Function c$Board.topEntity_go[1429471] will not reach a normal form, and compilation might fail.
Run with '-fclash-inline-limit=N' to increase the inlining limit to N.
```

Note that `AddrMap`'s `Monoid` instance is defined in terms of `Map`, `First` and `Ap`, all of which are defined locally, so they should have unfoldings available.

The list of unknown primitives reported by Clash is the same with and without the extra `return ()`:

```
Data.OldList.intercalate_$spoly_go 
Data.OldList.lines 
Data.OldList.prependToAll 
Data.Semigroup.Internal.$wstimesDefault 
Data.Sequence.Internal.$fFunctorSeq_$cfmap 
Data.Sequence.Internal.$fIsListSeq_mkTree 
Data.Sequence.Internal.$w$slookupTree 
Data.Sequence.Internal.$wupdate 
Data.Singletons.TypeLits.Internal.%^_f 
Data.Typeable.Internal.$wmkTrCon 
Data.Typeable.Internal.eqTypeRep 
Foreign.C.Error.throwErrno1 
Foreign.C.Error.throwErrnoIfMinus1Retry2 
Foreign.C.Error.throwErrnoIfMinus1RetryMayBlock2 
GHC.Base.++ 
GHC.Base.++_$s++ 
GHC.Base.eqString 
GHC.Base.map 
GHC.CString.unpackAppendCString# 
GHC.Classes.$fEqModule_$s$c== 
GHC.Event.Thread.closeFdWith1 
GHC.Event.Thread.threadWaitRead1 
GHC.Foreign.$wpeekCString 
GHC.Foreign.charIsRepresentable3 
GHC.IO.Encoding.Iconv.iconvEncoding6 
GHC.IO.Encoding.Iconv.localeEncodingName 
GHC.IO.Encoding.UTF16.mkUTF3 
GHC.IO.Encoding.UTF16.mkUTF5 
GHC.IO.Encoding.UTF8.mkUTF1 
GHC.IO.Encoding.UTF8.mkUTF4 
GHC.IO.Encoding.getFileSystemEncoding_go 
GHC.IO.Exception.$fExceptionIOException_$ctoException 
GHC.IO.FD.$fBufferedIOFD5 
GHC.IO.FD.$wfdWrite 
GHC.IO.FD.openFile1 
GHC.IO.Handle.Internals.$wdo_operation 
GHC.IO.Handle.Internals.mkDuplexHandle9 
GHC.IO.Handle.Internals.noByteBuffer 
GHC.IO.Handle.Internals.noCharBuffer 
GHC.IO.Handle.Text.hGetContents2 
GHC.Integer.Type.czeroBigNat 
GHC.Integer.Type.nullBigNat 
GHC.Integer.Type.oneBigNat 
GHC.Integer.Type.zeroBigNat 
GHC.List.$wlenAcc 
GHC.List.$wspan 
GHC.List.elem 
GHC.List.reverse1 
GHC.List.splitAt_$s$wsplitAt' 
GHC.Prim.addWordC# 
GHC.Prim.byteArrayContents# 
GHC.Prim.catch# 
GHC.Prim.getMaskingState# 
GHC.Prim.getSizeofMutableByteArray# 
GHC.Prim.indexArray# 
GHC.Prim.indexWordArray# 
GHC.Prim.maskAsyncExceptions# 
GHC.Prim.newAlignedPinnedByteArray# 
GHC.Prim.newArray# 
GHC.Prim.newByteArray# 
GHC.Prim.newMutVar# 
GHC.Prim.newPinnedByteArray# 
GHC.Prim.noDuplicate# 
GHC.Prim.plusAddr# 
GHC.Prim.putMVar# 
GHC.Prim.raise# 
GHC.Prim.raiseIO# 
GHC.Prim.readMutVar# 
GHC.Prim.readWideCharOffAddr# 
GHC.Prim.readWord8OffAddr# 
GHC.Prim.readWordArray# 
GHC.Prim.realWorld# 
GHC.Prim.resizeMutableByteArray# 
GHC.Prim.seq# 
GHC.Prim.setByteArray# 
GHC.Prim.shrinkMutableByteArray# 
GHC.Prim.sizeofByteArray# 
GHC.Prim.takeMVar# 
GHC.Prim.timesWord2# 
GHC.Prim.touch# 
GHC.Prim.unsafeFreezeArray# 
GHC.Prim.unsafeFreezeByteArray# 
GHC.Prim.void# 
GHC.Prim.waitRead# 
GHC.Prim.writeArray# 
GHC.Prim.writeInt8OffAddr# 
GHC.Prim.writeMutVar# 
GHC.Prim.writeWideCharOffAddr# 
GHC.Prim.writeWord8OffAddr# 
GHC.Prim.writeWordArray# 
GHC.Show.$fShowInteger_jprintb 
GHC.Show.$witos' 
GHC.Show.$wjblock' 
GHC.Show.$wjsplitf 
GHC.Show.showLitString 
GHC.Stack.Types.getCallStack 
GHC.Tuple.$tc(,,)1 
System.Posix.Internals.$wfdStat 
System.Posix.Internals.$wtcSetAttr 
Text.ParserCombinators.ReadP.$fAlternativeP_$srun 
```

gergoerdi, 2020-12-28:
Core (after simplifier) for `topEntity` without the extra `return ()`:

```
-- RHS size: {terms: 130, types: 580, coercions: 360, joins: 1/9}
topEntity
  :: HiddenClockResetEnable System =>
     Signal System (Maybe (Unsigned 16))
     -> Signal System (Maybe (Unsigned 8))
     -> Signal System (Maybe (Unsigned 8))
[GblId,
 Arity=3,
 Str=<S,U><L,U><L,U>,
 Unf=Unf{Src=<vanilla>, TopLvl=True, Value=True, ConLike=True,
         WorkFree=True, Expandable=True, Guidance=NEVER}]
topEntity
  = \ ($d(%,,%)_asvI :: HiddenClockResetEnable System)
      (eta_B2 :: Signal System (Maybe (Unsigned 16)))
      (eta1_B1 :: Signal System (Maybe (Unsigned 8))) ->
      let {
        $d(%,%)_stbW [Dmd=<S,U>] :: HiddenClock System
        [LclId]
        $d(%,%)_stbW = GHC.Classes.$p1(%,,%) $d(%,,%)_asvI } in
      let {
        $dKnownDomain_stbV [Dmd=<S,U>] :: KnownDomain System
        [LclId]
        $dKnownDomain_stbV = GHC.Classes.$p2(%,%) $d(%,%)_stbW } in
      case memoryMap
             eta_B2
             eta1_B1
             (\ (@ s_asvP) ->
                let {
                  eta2_stc5 [Dmd=<L,C(C1(U(1*U,1*U,1*U(U,1*U))))>]
                    :: Addressing
                         s_asvP
                         System
                         (Unsigned 8)
                         (Unsigned 16)
                         (Component s_asvP (Index 1024))
                  [LclId]
                  eta2_stc5
                    = ram0
                        ((GHC.Classes.$p1(%,%) $d(%,%)_stbW, $dKnownDomain_stbV),
                         (GHC.Classes.$p1(%,%) (GHC.Classes.$p2(%,,%) $d(%,,%)_asvI),
                          $dKnownDomain_stbV),
                         (GHC.Classes.$p1(%,%) (GHC.Classes.$p3(%,,%) $d(%,,%)_asvI),
                          $dKnownDomain_stbV))
                        ((GHC.Types.Eq# @~ <Co:1>) `cast` <Co:7>)
                        Clash.Sized.Internal.Unsigned.$fNFDataXUnsigned
                        Board.$s$fNumUnsigned
                        (Clash.Promoted.Nat.SNat
                           (Board.$s$fBoundedIndex1 `cast` <Co:7>)) } in
                (\ (eta3_Xl
                      :: (FanIn "System" (Unsigned 16),
                          Signal "System" (Maybe (Unsigned 8)), AddrMap s_asvP "System"))
                   (eta4_XF [OS=OneShot] :: Int) ->
                   let {
                     ds_stce [Dmd=<L,U(1*U,1*U,1*U(U,1*U))>]
                       :: (Component s_asvP (Index 1024), Int,
                           (FanIn "System" (Maybe (Unsigned 8)), AddrMap s_asvP "System"))
                     [LclId]
                     ds_stce = (eta2_stc5 `cast` <Co:40>) eta3_Xl eta4_XF } in
                   let {
                     ram_asvd :: Component s_asvP (Index 1024)
                     [LclId]
                     ram_asvd
                       = case ds_stce of { (x_ailD, s'_ailE, w1_ailF) -> x_ailD } } in
                   let {
                     ds1_stck [Dmd=<L,U(A,1*U,1*U(U,1*U))>]
                       :: ((), Int,
                           (FanIn "System" (Maybe (Unsigned 8)), AddrMap s_asvP "System"))
[... 234 lines cut ...]
                                (ds3_dh8u :: [(Key, FanIn "System" ())])
                         = case ds3_dh8u of {
                             [] -> jump go_sq9Z (b1_apb9 `cast` <Co:3>) (z_agef `cast` <Co:5>);
                             : x2_ageh xs_agei ->
                               jump go1_XqbY
                                 (RetroClash.Memory.$fMonoidAddrMap_$sinsertWithKey
                                    (f1_XqbX `cast` <Co:39>) x2_ageh z_agef)
                                 xs_agei
                           }; } in
                     jump go1_XqbY
                       (GHC.Types.[] `cast` <Co:6>)
                       ((RetroClash.Memory.$fSemigroupAddrMap_$c<> b2_XpcU b'_apbe)
                        `cast` <Co:9>))
                    }
                    }
                    }
                    }
                    }
                    }))
                `cast` <Co:36>)
      of
      { (x_aoYH, ds1_aoYI) ->
      x_aoYH
      }
```

christiaanb, 2020-12-29:
While trying to narrow the issue, I noticed that your `unionWithKey` is differently biassed from my example. I don't know why (and this is somewhat unsatisfactory) but switching the bias makes it go through:

```diff
diff --git a/src/Assoc.hs b/src/Assoc.hs
index d73b5ae..c6c9858 100644
--- a/src/Assoc.hs
+++ b/src/Assoc.hs
@@ -11,10 +11,10 @@ instance (Ord k) => Monoid (Map k a) where
     mempty = MkMap []
 
 unionWithKey :: (Ord k) => (k -> a -> a -> a) -> Map k a -> Map k a -> Map k a
-unionWithKey f l r = fold (insertWithKey f) l (assocs r)
+unionWithKey f l r = fold (insertWithKey f) r (assocs l)
 
 union :: (Ord k) => Map k a -> Map k a -> Map k a
-union = unionWithKey (\ _ old new -> old)
+union = unionWithKey (\ _ new old -> old)
```
I believe that for your use case the bias doesn't matter since the values in the map are a `Monoid`.

gergoerdi, 2020-12-29:
> I believe that for your use case the bias doesn't matter since the values in the map are a `Monoid`.

Just a small note: since my monoids are not commutative, I also needed to change the `Monoid` instance of `AddrMap` to 

```
instance Semigroup (AddrMap s dom) where
    AddrMap map1 <> AddrMap map2 = AddrMap $ unionWithKey (const $ flip (<>)) map1 map2
```

(note the `flip (<>)`)

I have now built up to the full Space Invaders memory map:

```
peripherals
    :: forall dom. HiddenClockResetEnable dom
    => Signal dom (BitVector 8)
    -> Signal dom (Maybe (PortCommand (Index 7) (Unsigned 8)))
    -> Signal dom (Maybe (Unsigned 8))
peripherals dips cmd = pure Nothing

#define ROM 1
#define RAM 1
#define VID 0
#define PORT 0

topEntity
    :: (HiddenClockResetEnable System)
    => Signal System (BitVector 8)
    -> Signal System (Maybe (Unsigned 8))
    -> Signal System (Maybe (Either (Unsigned 8) (Unsigned 16)))
    -> Signal System (Maybe (Unsigned 8))
    -> (Signal System (Maybe (Unsigned 8)), (Signal System (Maybe (Index 7168)), Signal System (Maybe (Unsigned 8))))
topEntity dips vidRead addr wr = memoryMap addr wr $ do
#if ROM
    rom <- romFromFile (SNat @0x2000) "_build/SpaceInvaders.bin"
#endif
#if RAM
    ram <- ram0 (SNat @0x0400)
#endif
#if VID
    (vid, vidAddr, vidWrite) <- conduit vidRead
#else
    let vidAddr = pure Nothing
        vidWrite = pure Nothing
#endif

#if PORT
    io <- port_ $ peripherals dips

    matchLeft $ do
        from 0x00 $ connect io
#endif

    matchRight $ do
#if ROM
        from 0x0000 $ connect rom
#endif
#if RAM
        from 0x2000 $ connect ram
#endif
#if VID
        from 0x2400 $ connect vid
#endif
#if RAM
        from 0x4000 $ connect ram
#endif

    return (vidAddr, vidWrite)
```

Clash can compile this with an inlining limit of 200 (I haven't tested its behaviour yet so maybe it's all bogus, but that's a start). I will now start playing around with turning on `VID` and/or `PORT` and seeing what kind of inlining limit I need with those.

gergoerdi, 2020-12-29:
Unfortunately, turning on `VID` or `PORT` on its own already causes Clash to loop seemingly unending, and increasing the inlining limit (even to something huge like 5000) doesn't seem to help.

christiaanb, 2020-12-30:
Yes, I've tried several variations on the theme. Changing the bias of the `unionWithKey`, changing to a right fold instead of a left fold in the implementation of `fold`. But as soon as I add more than two components and using them with `from X . connect` the compiler keeps asking to raise the inline limit whilst never successfully finishing (either error-ing out because the specialisation limit is hit, or just going on for a very long time after which I kill it).

I'm investigating further, but getting to a solution might take some time.

christiaanb, 2020-12-30:
Currently investigating why compilation is still successful even when the compiler warns that the inlining limit has been exceeded, hopefully that will give me some hints on how to change the strategy/order in which the transformations are applied.

christiaanb, 2020-12-31:
I've been unsuccessful so far in either narrowing down or fixing the problem. 

I tried a rewrite strategy that should require less inlining (https://github.com/clash-lang/clash-compiler/commit/c33c8ea4b5211cdc96e872b235dbeeb1dc188678) but even then an inline limit of 65000 isn't sufficient and clash errors out after 79m letting me know the specialisation limit is hit. There's no point in raising the spec limit because I already get warnings that the inlining limit is exceeded (one should only raise the spec limit if there are no warnings that the inline limit is exceeded).

gergoerdi, 2021-01-23:
One thing I just now noticed, and it might be a hint of what is happening here, is that even when it does finish compilation (with lots of complaining about the inline limit reached), the resulting Verilog looks awful for what it's supposed to do. 

For example, with the [latest version of the minimized code](https://github.com/gergoerdi/clash-issue-1536/tree/0ef6a4f47105f5218965aaf5e75610d7ca02268b), the included `build.sh` script finishes after 2.5 minutes, emits 810474 lines of warnings, and then produces a 20370-line Verilog file.

gergoerdi, 2021-01-24:
Another interesting observation is that once the levee breaks, there is no coming back. What I mean by that, is that [as of this version](https://github.com/gergoerdi/clash-issue-1536/tree/d11fd3725de5b5244314803d73e5d571036246b2), the following `topEntity` compiles, **with no warnings at all, with an inliner limit of 30** (and 30 is an exact bound, it fails with 29):

```
topEntity
    :: (HiddenClockResetEnable System)
    => Signal System (Maybe (Either (Unsigned 8) (Unsigned 16)))
    -> Signal System (Maybe (Unsigned 8))
    -> Signal System (Maybe (Unsigned 8))
topEntity addr wr = memoryMap_ addr wr $ do
    rom <- romFromFile (SNat @0x2000) "_build/SpaceInvaders.bin"
    ram <- ram0 (SNat @0x0400)

    matchRight $ do
        from 0x0000 $ connect rom
        from 0x2000 $ connect ram
        from 0x4000 $ connect ram
```

But then, if I add a single `return ()` to the end, it doesn't work **even with an inliner limit of 200**. So there is something qualitatively breaking, not just quantitatively, since there is no way the one extra bind (with one extra `(<> mempty)` on the writer) would really make any recursion that much more deep.

Could it be that GHC is inlining things differently based on some treshold, where it decides that "oops this recursion may be unbound, so I better not inline at all"?

[21 more replies]

commit c7d2e5c8 (2020-10-14): Make sure elemExistentials only eliminates existentinals and not other tyvars

Fixes #1536
commit e96c8e4c (2020-10-15): Make sure elemExistentials only eliminates existentinals and not other tyvars

Fixes #1536
commit 56350abf (2020-10-16): Make sure elemExistentials only eliminates existentinals and not other tyvars

Fixes #1536
PR #1551 (2020-10-16): Make sure elemExistentials only eliminates existentinals and not other tyvars
In #1536 clash got something like:
```
(ΛboolTv ->
  case pre'[LocalId] of
    INothing (ipv :: boolTv ~ False) ->
      ...
    IJust (co1 :: boolTv ~ True) .. ->
      ... SOMETHING_IMPOSSIBLE
) @False
```

Normally `appPropFast` should propagate the applied `False` type, so that `caseElemNonReachable` can eliminate the `IJust` alternative from the case because of its absurd constraint (`False ~ True`).  
Only before they could do that, `elemExistentials` came along and tried "solved" the type equalities locally an [... cut]