clash-lang/clash-compiler #1537: TypeInType no longer turned on by default?

gergoerdi, 2020-09-25:
I see in
https://github.com/clash-lang/clash-compiler/blob/master/docs/developing-hardware/language.rst that `TypeInType` is still listed as a language extension turned on by default. However, after upgrading Clash from 5b65cf4d515dcc6cf5e9b3c5d6784415a5275ad7 to b0a239e1a6b0e0d0f7cf63a0c85f2c1448a0a13e, I now have some modules that only typecheck if I explicitly turn on `TypeInType` in a language pragma.

Is this a bug or an intended change? If the latter, the documentation should be changed.

alex-mckenna, 2020-09-25:
Hmm, I don't immediately see any changes between these which would change this. `-XTypeInType` does only appear to be turned on by default in GHC < 8.6, if you've also upgraded your GHC this may have something to do with it.

It is worth noting that since GHC 8.6, `-XTypeInType` is deprecated and is just a synonym for `-XDataKinds` and `-XPolyKinds`. Perhaps on GHC > 8.6 we should turn on `-XPolyKinds` by default as well? Regardless, the documentation should be changed to mention how GHC version affects language pragmas.

gergoerdi, 2020-09-25:
Yeah, no, you're right -- I got confused because I had `TypeInType` in my `package.yaml` because the docs made me think I should have it there; so the change was not between the the two Clash versions, but rather my build environment.

alex-mckenna, 2020-09-25:
Good to know, at least it isn't some horrifically subtle bug then. I'll put a PR in to improve the docs when I have some time :+1:

gergoerdi, 2020-09-25:
I feel like all my bug reports fall into these two buckets:
* "Horrifically subtle bug"
* "I got myself confused"

leonschoorl, 2020-09-25:
In GHC 8.6 `TypeInType` stopped existing as an actual extension, instead it became just shorthand for `PolyKinds, DataKinds, KindSignatures`:
https://downloads.haskell.org/~ghc/8.6.1/docs/html/users_guide/glasgow_exts.html#extension-TypeInType

And it was deprecated in GHC 8.8: https://downloads.haskell.org/~ghc/8.8.1/docs/html/users_guide/glasgow_exts.html#extension-TypeInType

gergoerdi, 2020-09-27:
> In GHC 8.6 `TypeInType` stopped existing as an actual extension, instead it became just shorthand for `PolyKinds, DataKinds, KindSignatures`

Yes, in my particular case, `PolyKinds` was the missing one.

PR #1541 (2020-09-28): Make language extension documentation clearer
The language extension documentation now makes it clear which
extensions are only enabled until / since certain GHC versions.
Notes are provided below which provide information about these
extensions.

Closes #1537
closed 2020-09-28 by alex-mckenna