clash-lang/clash-compiler #1540: Idea: digit grouping in `bitPattern`

gergoerdi, 2020-09-26:
I've just now realized that I would sometimes like to group the digits of a `bitPattern`, similar to [the `NumericUnderscores` language extension](https://ghc.gitlab.haskell.org/ghc/doc/users_guide/exts/numeric_underscores.html); for example, I'd like to rewrite this:

```
    $(bitPattern "01110110") -> HLT
    $(bitPattern "01......") -> MOV dest src
    $(bitPattern "00...110") -> MVI dest
    $(bitPattern "00..0001") -> LXI rr
```

into this:

```
    $(bitPattern "01_110_110") -> HLT
    $(bitPattern "01_..._...") -> MOV dest src
    $(bitPattern "00_..._110") -> MVI dest
    $(bitPattern "00_..0_001") -> LXI rr
```

I am of course volunteering to implement this, if the consensus is that it's a good idea. Let me know what you think.

gergoerdi, 2020-09-26:
I also wonder how batshit crazy it would be to allow `$(bitPattern "01dddsss")` to bind `ddd` and `sss` of type `BitVector 3`. And yes, if you instead wanted to get a `BitVector 4` and a `BitVector 2`, you would have to write `$(bitPattern "01ddddss")`; the length is encoded in the variable name.

basile-henry, 2020-09-26:
That seems like a great idea! I didn't know about `bitPattern`. :smile: 

As you say you can already use the `NumericUnderscores` language extension with`BinaryLiterals` to pattern match on for example this `0b1001_1101`. I guess the main benefit of `bitPattern` is to be able to have _don't care_ bits.

I expect changing `bs = parse <$> s` to `bs = parse <$> filter (/= '_') s` should be enough?

https://github.com/clash-lang/clash-compiler/blob/1ccadf371bf5c3c19e16327b21370a56294edd11/clash-prelude/src/Clash/Sized/Internal/BitVector.hs#L1106-L1117

martijnbastiaan, 2020-09-26:
Allowing `_` seems like a good idea! I think you're right @basile-henry, that's the only thing that'd need changing. 

>  also wonder how batshit crazy it would be to allow $(bitPattern "01dddsss") to bind ddd and sss of type BitVector 3.

What would you do with: `01dd01dd`? Would it be an idea to bind any number of `d`s to the single-letter variable `d` instead?

basile-henry, 2020-09-26:
That's interesting, I'm not sure how often you'd want to bind a single bit though. Also I'm worried that would be incompatible with numeric underscored as the underscore would probably be seen as not binding a bit (like `'.'` does at the moment)

martijnbastiaan, 2020-09-26:
Perhaps I'm misreading your comment but my proposal would work on multiple bits. E.g.,

```
$(bitPattern "01ddssdd") 
```

would bind two variables: `d :: BitVector 4` and `s :: Bitvector 2`. Does that make sense? :)

> Also I'm worried that would be incompatible with numeric underscored as the underscore would probably be seen as not binding a bit

Good point. That would make `_` ambiguous. Although I guess you could opt to only bind `[a-z]`..

basile-henry, 2020-09-26:
Oh right, I misunderstood that using the same character multiple times was binding different bits to the same variable. That does sound pretty cool!
Maybe the underscore ambiguity is not too bad. This is a fairly specialised pattern match setup anyway.

gergoerdi, 2020-09-27:
> Perhaps I'm misreading your comment but my proposal would work on multiple bits. E.g.,
> 
> ```
> $(bitPattern "01ddssdd") 
> ```
> 
> would bind two variables: `d :: BitVector 4` and `s :: Bitvector 2`. Does that make sense? :)
>
> Edit: perhaps it would be more obvious to bind to `dddd` and `ss` still.

JFTR, my 2c would be to bind `dddd` and `ss`, yes.

martijnbastiaan, 2020-09-28:
@gergoerdi If it wasn't obvious yet, I'd be happy to accept a PR doing this ^ :)

basile-henry, 2020-10-03:
I'd like to implement this if no one's already done it! :smile:

PR #1545 (2020-10-10): Grouping and binding in bitPattern
This PR extends the functionality of the `bitPattern` TH pattern function.

It adds the possibility to use `_` as a spacer in a similar fashion to the `NumericUnderscores` language extension. There is no restriction as to where this spacer can be used (which is a bit looser than the language extension which doesn't allow the spacer on either end of a number `_0011` and `64_` for example).

It also adds the possibility to bind arbitrary bits to a variable of type `BitVector n` where `n` is the number of bits bound using the same character. Examples of this mec [... cut]
closed 2020-10-10 by martijnbastiaan