clash-lang/clash-compiler #1547: minimum numbers' literal of the `SFixed` data type

IchiroKawashima, 2020-10-09:
Hello,

When I try using the `SFixed` data type in the `clashi` prompt, I found a wired phenomenon about it's literal.

The code I tried was as follows:

```haskell
>>> minBound :: SFixed 1 0
-1.0

>>> -1.0 :: SFixed 1 0
0.0

>>> minBound :: SFixed 0 1
-0.5

>>> -0.5 :: SFixed 0 1
0.0
```

The code demonstrates that the minimum number's literals are not recognized properly.
`-1.0 :: SFixed 1 0` should become `-1.0` because `SFixed 1 0` can represent `-1.0` and `0.0`.
I saw the same problem with other fixed-point numbers that have different precisions.
The data type translates the minimum number's literals as the number just above the minimum number.

Thank you.

basile-henry, 2020-10-09:
The issue is that `-1.0` isn't a literal for `SFixed 1 0`. The `-` is applied after using the `1.0` literal. and `1.0` doesn't exist as an `SFixed 1 0`. So first `1.0` is used as an `SFixed 1 0` and since the closest valid value is `0.0` it uses that before negating it.

Here as an example with `SFixed 1 1` (so we can reach negative numbers with literals):

```
> minBound :: SFixed 1 1
-1.0
> maxBound :: SFixed 1 1
0.5
> -1.0 :: SFixed 1 1
-0.5
> (-0.5 - 0.5) :: SFixed 1 1
-1.0
```

One thing you can do to treat the `-` as part of the number definition is to use `fromRational`:

```
> fromRational (-1.0) :: SFixed 1 0
-1.0
```

This works because `1.0` is a valid Rational even though it's not a valid `SFixed 1 0`.

I hope that makes sense. I'm not too sure there is much we can do in clash/clash-prelude itself to make this easier.

christiaanb, 2020-10-09:
Perhaps this needs `-XNegativeLiterals`

basile-henry, 2020-10-09:
Woah, yeah `-XNegativeLiterals` does work. I had not idea that was a thing.

```
> :set -XNegativeLiterals
> -1.0 :: SFixed 1 0
-1.0
```

IchiroKawashima, 2020-10-09:
@christiaanb, @basile-henry Thank you. I could fix the problem.