clash-lang/clash-compiler #1557: Bad time complexity for VHDL, not Verilog

DigitalBrains1, 2020-10-22:
The file below exposes bad time complexity in the normalization phase of generating VHDL, whereas Verilog is near-instant. With a vector of length 25 as below, the numbers are:
VDHL: `Clash: Normalisation took 3.052s`
Verilog: `Clash: Normalisation took 0.027s`
More lengths:
- 30:
  VHDL: `Clash: Normalisation took 6.543s`
  Verilog: `Clash: Normalisation took 0.031s`
- 35:
  VHDL: `Clash: Normalisation took 12.721s`
  Verilog: `Clash: Normalisation took 0.034s`
- 40:
  VHDL: `Clash: Normalisation took 22.088s`
  Verilog: `Clash: Normalisation took 0.037s`

This is with CλaSH `master`. Using `1.2.4` seems to exhibit the same problem but less pronounced.

```Haskell
module LongVecCompl where

import Clash.Prelude

topEntity
  :: SystemClockResetEnable
  => Signal System (Bool, Unsigned 8)
{-# NOINLINE topEntity #-}
topEntity = f (packetVecToStreamVec $(listToVecTH [1 :: Unsigned 8 .. 25]))

f :: forall n
   . (SystemClockResetEnable, KnownNat n, 1 <= n)
  => Vec n (Bool, Unsigned 8)
  -> Signal System (Bool, Unsigned 8)
f is = (is!!) <$> sel
 where
  sel :: Signal System (Index n)
  sel = register 0 $ satSucc SatWrap <$> sel

packetVecToStreamVec pkt =    map (\e -> (True, e)) (init pkt)
                           :< (False, last pkt)
```

DigitalBrains1, 2020-10-23:
The code can be further reduced while keeping the problematic behaviour:
```haskell
module LongVecCompl where

import Clash.Prelude

topEntity
  :: SystemClockResetEnable
  => Signal System (Unsigned 8)
  -> Signal System (Bool, Unsigned 8)
{-# NOINLINE topEntity #-}
topEntity = fmap (packetVecToStreamVec (replicate d25 0) !!)

packetVecToStreamVec pkt =    map (\e -> (True, e)) (init pkt)
                           :< (False, last pkt)
```

leonschoorl, 2020-10-26:
For some reason when compiling to Verilog the `map` isn't evaluated.
When compiling to VHDL or SystemVerilog it is evaluated.

commit 10aca06e (2021-06-04): Stop forcing index_int arguments

Clash was incorrectly trying to force the evaluator to evaluate
the vector argument of the indexing operator in order to get
the special primitive rule for indexing in verilog to fire,
while it only needs the index argument to be evaluated to a
literal when possible.

Fixes #1557
PR #1842 (2021-06-04): Fix1557
Fixes #1557

## Still TODO:

  - [x] Write a changelog entry (see changelog/README.md)
  - [x] Check copyright notices are up to date in edited files
  - [x] Commit message on https://github.com/clash-lang/clash-compiler/pull/1842/commits/5c3e7c63cc23c1ca6b702925e791dcf6f22801e6

[comment]: # (Depending on the change, some of these points may not be necessary. If so, leave the boxes unchecked so it is easier for a reviewer to see what has been omitted.)
commit 52742382 (2021-06-04): Stop forcing index_int arguments

Clash was incorrectly trying to force the evaluator to evaluate
the vector argument of the indexing operator in order to get
the special primitive rule for indexing in verilog to fire,
while it only needs the index argument to be evaluated to a
literal when possible.

Fixes #1557
commit 06158479 (2021-08-02): Stop forcing index_int arguments

Clash was incorrectly trying to force the evaluator to evaluate
the vector argument of the indexing operator in order to get
the special primitive rule for indexing in verilog to fire,
while it only needs the index argument to be evaluated to a
literal when possible.

Fixes #1557

(cherry picked from commit 5274238207d372f0066b48287eab18bdc112a39f)
closed 2021-06-04 by christiaanb