clash-lang/clash-compiler #1570: clash.cabal getDirectoryContents:openDirStream permission denied

rongcuid, 2020-11-06:
I tried to setup a project:

```
$ clash.cabal init
Warning: The package list for 'hackage.haskell.org' does not exist. Run 'cabal
update' to download it.
/var/lib/snapd/void: getDirectoryContents:openDirStream: permission denied (Permission denied)
```

Attempting to perform update also fails:

```
/var/lib/snapd/void: getDirectoryContents:openDirStream: permission denied (Permission denied)
```

I am on Pop OS (Ubuntu) 20.04, using the snap version of Clash.

martijnbastiaan, 2020-11-09:
I can't reproduce this one unfortunately. What does running `snap connections clash` give you? It looks like:

```
Interface  Plug           Slot      Notes
home       clash:home     :home     -
network    clash:network  :network  -
```

for me. As a workaround, use the [Clash Starter project](https://github.com/clash-lang/clash-compiler/tree/master/clash-starters/clash-example-project) instead. You can take the Stack route by installing it with apt:

```
sudo apt install haskell-stack
```

If you want to go the cabal route, I'd recommend [ghcup](https://www.haskell.org/ghcup/) to install a recent GHC.

rongcuid, 2020-11-09:
I get this:

```
$ snap connections clash
Interface  Plug           Slot      Notes
home       clash:home     :home     -
network    clash:network  :network  -
```

I already have Stack installed. Right now my project is set up just laying everything flat in a directory, but I can try the starter template.

martijnbastiaan, 2021-03-19:
We now recommend using the [Clash starter project](https://github.com/clash-lang/clash-starters) to handle external dependencies. I don't think we can make snaps work well with external dependencies, so I'll be closing this issue.

closed 2021-03-19 by martijnbastiaan