clash-lang/clash-compiler #1573: Logical right shift missing

rongcuid, 2020-11-08:
I know Haskell does not expose the logical right shift operator, but this is an operator used in hardware. Should this be provided by Clash, or do I need to do `uncheckedIShiftRL#`?

rowanG077, 2020-11-09:
`shiftR` is exposed under the [`Bits`](https://hackage.haskell.org/package/base-4.14.0.0/docs/Data-Bits.html#v:shiftR) type class. Is this not what you need?

martijnbastiaan, 2020-11-09:
I partly blame this on our documentation. Given that bit manipulation is such an integral part of hardware design, perhaps we should just re-export `Data.Bits` in clash-prelude such that it [shows up under "Models"](https://hackage.haskell.org/package/clash-prelude-1.2.4)?

rowanG077, 2020-11-09:
What would be most logical to me would be to roll `BitIndex` and `BitReduction` into a single page and there re-expose the `Data.Bits` functions on that page as well. That would keep all bit level stuff together. But i'm not sure if that's possible with haddock.

martijnbastiaan, 2020-11-09:
`Clash.Bits`!

rongcuid, 2020-11-09:
I asked on Slack channel, and found out I misunderstood the docs. Haskell `Data.Bits` says `shiftR` performs arithmetic right shift, but it turns out Unsigned and BitVector types actually perform logical right shift. I would say this is a documentation problem.

Speaking of which, I find the docs of Clash quite hard to navigate, because the Prelude docs don't show many methods, and I have to guess their names and look inside Index or other modules.

martijnbastiaan, 2020-11-09:
Thanks for the feedback @rongcuid. If you've got any specific suggestions on how to improve our documentation (e.g., where to show certain functionality), we'd be happy to hear it!

rongcuid, 2020-11-09:
I would like a page describing the most important data types of ClaSH, and a list of operations. If that sounds too vague, maybe it is good to provide a quick reference and comparison between, say, Verilog operators. Currently the docs are very much scattered around.

alex-mckenna, 2020-11-09:
We have [this page](https://clash-lang.readthedocs.io/en/latest/developing-hardware/prelude.html) in our readthedocs, but it only covers the main types in `Clash.Prelude`, not their full APIs. It may help give you an idea which types / modules to then look at in the haddock documentation though.

rongcuid, 2020-11-09:
Ah, I didn't even see this site. The website Documentation page directs straight to the Hackage documents. Is there a contributing guide or something? I might consider doing so later.

alex-mckenna, 2020-11-09:
If we had a contribution guide I feel it would just say _we encourage users to submit PRs / issues_. This issue seems like a good place to at least get some ideas for how to improve documentation

christiaanb, 2020-11-09:
We should link clash-lang.readthedocs.io from the main clash-lang.org website...

martijnbastiaan, 2021-04-16:
@leonschoorl and I were just casually chatting about this: given that it's all functions on `BitPack a`, we could just export them from `Clash.Class.BitPack`.

martijnbastiaan, 2021-04-16:
On the general topic of "the documentation is hard to browse": perhaps we should swap `Bit` and `BitVector` in `Clash.Sized.BitVector`.