clash-lang/clash-compiler #1574: doctests failure

jchia, 2020-11-09:
When I run `stack test clash-prelude` under `resolver: nightly-2020-11-06`, I get the following doctests failure. I use a newer resolver because I normally don't use LTS-15 and the failure may still repro there.

```
/data/jchia/gh/clash-compiler/clash-prelude/src/Clash/Explicit/Reset.hs:160: failure in expression `sampleResetN 12 (resetGlitchFilter d2 systemClockGen rst)'
expected: [True,True,True,True,False,False,False,False,False,True,True,False]
 but got: [True,True,True,True,False,False,False,False,False,True,True,*** Exception: X: Clash.Sized.BitVector.== called with (partially) undefined arguments: 0. == 11
                                                                       ^
          CallStack (from HasCallStack):
            undefErrorI, called at /data/jchia/gh/clash-compiler/clash-prelude/src/Clash/Sized/Internal/BitVector.hs:452:15 in main:Clash.Sized.Internal.BitVector

/data/jchia/gh/clash-compiler/clash-prelude/src/Clash/Explicit/Reset.hs:206: failure in expression `sampleWithReset (holdReset @System clockGen enableGen (SNat @2) (resetGenN (SNat @3)))'
expected: [True,True,True,True,True,False,False,False]
 but got: 
          ^
          <interactive>:3329:18: error:
              • Couldn't match expected type ‘Enable dom0
                                              -> SNat 2 -> Reset dom3 -> Reset dom1’
                            with actual type ‘Reset System’
              • The expression ‘holdReset @System’ is applied to four arguments,
                but its type ‘SNat m0 -> Reset System’ has only one
                In the first argument of ‘sampleWithReset’, namely
                  ‘(holdReset
                      @System clockGen enableGen (SNat @2) (resetGenN (SNat @3)))’
                In the expression:
                  sampleWithReset
                    (holdReset
                       @System clockGen enableGen (SNat @2) (resetGenN (SNat @3)))
          
          <interactive>:3329:36: error:
              • Couldn't match expected type ‘SNat m0’
                            with actual type ‘Clock dom2’
              • In the second argument of ‘holdReset’, namely ‘clockGen’
                In the first argument of ‘sampleWithReset’, namely
                  ‘(holdReset
                      @System clockGen enableGen (SNat @2) (resetGenN (SNat @3)))’
                In the expression:
                  sampleWithReset
                    (holdReset
                       @System clockGen enableGen (SNat @2) (resetGenN (SNat @3)))

/data/jchia/gh/clash-compiler/clash-prelude/src/Clash/Sized/Vector.hs:1893: failure in expression `Vec.unsafeFromList [1,2,3,4,5] :: Vec 10 Int'
expected: <1,2,3,4,5,*** Exception: Clash.Sized.Vector.unsafeFromList: vector larger than list
 but got: <1,2,3,4,5,*** Exception: Clash.Sized.Vector.unsafeFromList: vector larger than list
          CallStack (from HasCallStack):
          ^
            error, called at /data/jchia/gh/clash-compiler/clash-prelude/src/Clash/Sized/Vector.hs:1905:16 in main:Clash.Sized.Vector

Examples: 817  Tried: 815  Errors: 0  Failures: 3

clash-prelude          > Test suite doctests failed
```

christiaanb, 2020-11-09:
Confirmed locally, somewhat worrying that our CI hasn't caught this.

martijnbastiaan, 2020-11-09:
Thanks @jchia. Looks like

https://github.com/clash-lang/clash-compiler/blob/24ace78665d651cab8541221b544826e48ec7555/.ci/test.sh#L2

should include the `e` flag too.

PR #1579 (2020-11-09): Fix failing doctests
Fixes #1574
commit 34398aac (2020-11-09): Fix failing doctests

Fixes #1574
closed 2020-11-09 by martijnbastiaan